openbmc/openbmc #1349: REST server: Add ability to merge two copies of a single dbus object into a single REST object

anoo1, 2017-03-20:


geissonator, 2017-03-23:
Lots of discussion on this one in task review, seems like this mostly works(?).  Need to test it out and come back to team with observations to figure out what we need to change.  Could be 1 point of doing this validation and minor changes, could be 2 points if we find issues and need to make bigger changes.

dkodihal, 2017-03-24:
- [ ] Write test app (python mostly) which can create d-bus objects. App takes as input object name and interface(s) to implement.
- [ ] Run two instances of above app which creates the same d-bus object implementing the same interfaces and verify behavior.
- [ ] Run two instances of above app which creates the same d-bus object implementing different interfaces and verify behavior.
- [ ] Fix issues found in rest server.

dkodihal, 2017-03-27:
1. In case of the same d-bus object implementing the same interfaces across two services, the rest server reported the object and properties as expected (no dups).
2. In case of the same d-bus object implementing different interfaces across two services, the rest server reported the object and properties as expected, the properties were 'merged' from the two interfaces.

So I think there's nothing to fix here.

closed 2017-04-09 by dkodihal