openbmc/openbmc #1350: Update Manager: Do not use patch path when BMC is in Field Mode

anoo1, 2017-03-20:


mtritz, 2017-07-26:
- [ ] Extend the PNOR software updater with a new dbus property for field mode
- [ ] Add a condition to the update manager - when the property is set, ignore the PNOR patch path

# **STATUS**
7/26: Wrote task list.

amboar, 2017-07-27:
What's driving this requirement? Why differentiate between 'Fielded' and 'Non-Fielded' states. It sounds very Enterprise.

mtritz, 2017-07-28:
Per @williamspatrick - "It is not uncommon for a OEM / ODM (including IBM) to want to disable function once the machine ships.  For example, you might want to allow voltage tweaking in manufacturing to make sure the machine has some stability, but you don't want your customer doing that.  PNOR patches are another item that we're planning to disable so that customers only install fully formed, proper firmware images."

amboar, 2017-07-31:
It's an open source project - anyone can literally change the source, build it and toast the machine with a fully formed, proper firmware image.

rfrandse, 2017-08-11:
<https://gerrit.openbmc-project.xyz/5939> Add field mode to BMC updater
Resolves  #1350 Update Manager: Do not use patch path when BMC is in Field Mode

closed 2017-08-21 by williamspatrick