openbmc/openbmc #1357: Inventory FRU and Core is set "Functional" but not "Present"  [witherspoon]

gkeishin, 2017-03-21:
`There is a mismatch is the "Functional"  vs "Present"  which contradicts the property of an inventory items`

Example:

```
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu0/core5": {
        "Functional": 1,
        "Present": 1,
        "PrettyName": ""
    },
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu0/core6": {
        "Functional": 1,
        "Present": 0,    <------ Functional but not present 
        "PrettyName": ""
    },
```

`Full inventory data`
```
Executing: GET  /xyz/openbmc_project/inventory/enumerate

/xyz/openbmc_project/inventory/enumerate
{
    "/xyz/openbmc_project/inventory/createme3": {
        "ExampleProperty1": "foo"
    },
    "/xyz/openbmc_project/inventory/system": {
        "BuildDate": "",
        "Cached": 0,
        "FieldReplaceable": 0,
        "Manufacturer": "",
        "Model": "2",
        "PartNumber": "0000000000000000",
        "Present": 1,
        "PrettyName": "",
        "SerialNumber": "0000000000000000"
    },
    "/xyz/openbmc_project/inventory/system/chassis/motherboard": {
        "BuildDate": "",
        "Manufacturer": "0000000000000000",
        "Model": "",
        "PartNumber": "0000000",
        "Present": 0,
        "PrettyName": "SYSTEM PLANAR   ",
        "SerialNumber": "000000000000"
    },
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu0": {
        "BuildDate": "",
        "Cached": 0,
        "FieldReplaceable": 1,
        "Functional": 1,
        "Manufacturer": "IBM",
        "Model": "",
        "PartNumber": "01HL322",
        "Present": 1,
        "PrettyName": "PROCESSOR MODULE",
        "SerialNumber": "YA3933741574",
        "Version": "10"
    },
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu0/core0": {
        "Functional": 1,
        "Present": 0,
        "PrettyName": ""
    },
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu0/core1": {
        "Functional": 1,
        "Present": 0,
        "PrettyName": ""
    },
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu0/core10": {
[... 363 lines cut ...]
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/dimm7": {
        "BuildDate": "",
        "Cached": 0,
        "FieldReplaceable": 1,
        "Functional": 1,
        "Manufacturer": "0x2c80",
        "Model": "18ASF1G72PZ-2G3B1   ",
        "PartNumber": "",
        "Present": 1,
        "PrettyName": "0x0c",
        "SerialNumber": "0x13afb2f2",
        "Version": "0x31"
    },
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/dimm8": {
        "Functional": 1,
        "Present": 0,
        "PrettyName": ""
    },
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/dimm9": {
        "Functional": 1,
        "Present": 0,
        "PrettyName": ""
    }
}
```

gkeishin, 2017-03-21:
`Driver info`
- [x] SYSTEM     : W42 ( Witherspoon - 2 Sockets - 16 DIMMs ) 
- [x] BUILD_ID   : https://openpower.xyz/job/openbmc-build/641/
- [x] VERSION_ID : v1.99.3-118-g0032233
- [x] Kernel     : Linux witherspoon 4.7.10-c72ced5a19ceae19c7d792ae3e982bee5607dab4 #1 Tue Mar 21 00:22:45 UTC 2017 armv6l GNU/Linux
- [x] PNOR VERSION:
IBM-witherspoon-ibm-OP9_v1.15_1.3
        op-build-v1.15-15-gce29449
        buildroot-2017.02-4-gc28573e
        skiboot-28e96ff
        hostboot-86bae1c
        linux-4.10.1-p976d063
        petitboot-v1.4.1-p3ec3d6e
        witherspoon-xml-065aaa2
        occ-4141b5f
        hostboot-binaries-4b44090
        capp-ucode-1bb7503
```
- [x] Journal log 
[journald_v1.99.3-118-g0032233_inventory.txt](https://github.com/openbmc/openbmc/files/857996/journald_v1.99.3-118-g0032233_inventory.txt)

ratagupt, 2017-03-21:
Seems to me host is not sending the data as it is expected.
There are two issues.
   1) why the host is sending the sensor info for the fru(eg dimm) where there is no VPD for that.
             it might happen that there was VPD read failure @host side so the host is not sending the fru data but sending the sensor info.

   2) The sensor data is not expected as per the spec.
         Memory
          bit 4 : true : NonFunctional
                  : false : Functional
          bit 6 : true : Present
                  : false : Not Present
         Proc/Core
          bit 8 : true : NonFunctional
                  : false : Functional
          bit 7 : true : Present
                  : false : Not Present

ratagupt, 2017-03-21:
IPMI Incoming: Seq 0x12, NetFn 0x04, CMD: 0x30 
0x000000: b8 a8 00 00 00 50 00 00 00 00

the above packet is for sensor (dimm) where we can see that deassertion bits are on for 4th and 6th bit.
which means non functional but present.

ratagupt, 2017-03-21:
sensor-id :- b8  which is for dimm9

williamspatrick, 2017-03-21:
@ratagupt : If ```deassertion bits are on for 4th and 6th bit which means non functional but present.``` why ```    "/xyz/openbmc_project/inventory/system/chassis/motherboard/dimm9": {
        "Functional": 1,
        "Present": 0,
        "PrettyName": ""
    }```

I'm not understanding your last two comments.

tomjoseph83, 2017-03-22:
@williamspatrick 

For Sensor Type Memory Bit 4 corresponds to " Memory Device Disabled". An assertion event on this bit means the DIMM is not functional and deassertion event on the bit indicate DIMM is functional.

For Sensor Type Memory Bit 6 corresponds to " Presence Detected". An assertion event on this bit means the DIMM is present and deassertion event on the bit indicate DIMM is not present.

In the IPMI packet analysed,bits 4 & 6 are deasserted, which means DIMM is not present and DIMM is functional.

ratagupt, 2017-03-22:
@williamspatrick : I had to type "Functional but not present".
Tom corrected me.
IPMI Incoming: Seq 0x12, NetFn 0x04, CMD: 0x30
0x000000: b8 a8 00 00 00 50 00 00 00 00

here 6th and 7th byte for deassertion. 
50                   00
01010000 
76543210 : bit numbers
on 6th byte we can see that 4th and 6th bits are on which means "not present but functional".

williamspatrick, 2017-03-22:
So we think Hostboot is sending the wrong thing?  I wonder why AMI never complained. @wghoffa - need your help here.

ratagupt, 2017-03-22:
Yes,we are thinking like that.

wghoffa, 2017-03-22:
Which sensors do you think are wrong? The title says "Inventory FRU and Core". What is an 'Inventory FRU'? Are you indeed talking about the core? Most of these comments are related to dimms. Is it all of the above?

ratagupt, 2017-03-23:
@wghoffa : you can take anything core /dimm, both have the same issue.
more precisely dimm8-dimm15 and most of the cores except(core14,15,18,19 on processor1 and core12,13 on processor2) are having this issue.
You can take a look at the data which @gkeishin pasted above

sivassrr, 2017-06-30:
We are seeing issues for DIMM as well. /xyz/openbmc_project/inventory/system/chassis/motherboard/dimm10 is functional but not present.

ratagupt, 2017-07-24:
Agreed design on this..

dhruvibm, 2017-07-25:
@ratagupt @williamspatrick @tomjoseph83 
To implement with above yaml, there will be a considerable rework in the current logic, because
1) The yaml is not according to the existing structure
2) Present or functional are properties which again needs to be translated, we may need to mention according to assertion or deassertion of bits at offsets.

My suggestion is to put something like this
xyz.openbmc_project.State.Decorator.OperationalStatus:
      Functional:
        0x04:
          assert: false
          type: bool
          deassert: true
              validate:
                    0x6: assert
This can be added easily in the current logic, by looking at the assertion or deassertion of a bit based on the given offset and validate the condition.

dhruvibm, 2017-07-27:
- [ ] Changing the YAML for the fix.

- [ ] Change in ipmid to accept new configuration.

dhruvibm, 2017-07-31:
Updated yaml after discussion, capturing following requirement 
Processor sockets => must do 'notify' as non-present.
Processor cores => must NOT do a 'notify' as non-present.
example socket:
0x5a:
  sensorType: 0x07
  path: /system/chassis/motherboard/cpu0
  sensorReadingType: 0x6F
  serviceInterface: xyz.openbmc_project.Inventory.Manager
  readingType: assertion
  interfaces:
    xyz.openbmc_project.State.Decorator.OperationalStatus:
      Functional:
          0x08:
            assert: false
            deassert: true
                 validate:
                       0x07: assert
            type: bool
    xyz.openbmc_project.Inventory.Item:
      Present:
          0x07:
            assert: true
            deassert: false
            type: bool

example core:

0x12:
  sensorType: 0x07
  path: /system/chassis/motherboard/cpu0/core0
  sensorReadingType: 0x6F
  serviceInterface: xyz.openbmc_project.Inventory.Manager
  readingType: assertion
  interfaces:
    xyz.openbmc_project.State.Decorator.OperationalStatus:
      Functional:
          0x08:
            assert: false
            deassert: true
            type: bool
    xyz.openbmc_project.Inventory.Item:
      Present:
          0x07:
            assert: true
            deassert: false
            type: bool
    noaction:
            0x07: deassert

dhruvibm, 2017-08-01:
# **STATUS**
08/01: Design discussion in progress

rfrandse, 2017-08-09:
<https://gerrit.openbmc-project.xyz/5822> config yaml changes to mark non present as non functional
Resolves  #1357 Inventory FRU and Core is set "Functional" but not "Present"  [witherspoon]

rfrandse, 2017-08-30:
<https://gerrit.openbmc-project.xyz/6164> IPMI changes to mark non present as non functional
Resolves  #1357 Inventory FRU and Core is set "Functional" but not "Present"  [witherspoon]

dkodihal, 2017-10-13:
All the code is merged - https://gerrit.openbmc-project.xyz/#/q/status:merged+AND+topic:1357.

closed 2017-10-13 by dkodihal