openbmc/openbmc #1360: CI dependency for phosphor-logging.

williamspatrick, 2017-03-21:
There are, as of now, two different `*-dbus-interfaces` repositories.  These repositories install YAML files into `/usr/share`.  The phosphor-logging repository has a script that runs on these YAML files to create header files for exception metadata logging.

An individual code repository `phosphor-code-example` may depend on `{ phosphor-dbus-interfaces, phosphor-logging }`.  Another code repository `openpower-code-example` may depend on `{ phosphor-dbus-interfaces, openpower-dbus-interfaces, phosphor-logging }`.  `phosphor-logging` itself does not need to depend on any `*-dbus-interfaces` repositories and it cannot depend on all of them.  In fact, some of them may be in conflict (ex. `power-dbus-interfaces`, `x86-dbus-interfaces` may conflict).

When we use bitbake we will have a 'packagegroup' that the needed `*-dbus-interfaces` repositories are added to for that system.  `phosphor-logging` will depend on `packagegroup-dbus-interfaces` and therefore all of the `*-dbus-interfaces` recipes are ordered before `phosphor-logging`.

Since the CI testing simply parses autoconf files to determine dependencies and `phosphor-logging` does not / cannot directly depend on the `*-dbus-interfaces` repositories, we need an enhancement to the CI script to deal with this.

The proposal:
1. After the dependency graph is created by the CI script, if both `phosphor-logging` and any `*-dbus-interfaces` are in the dependency graph, manually insert a dependency from `phosphor-logging` to `*-dbus-interfaces`.
2. Verify that `*-dbus-interfaces` 'make install' is correctly installing the YAML to a location like `/usr/share`.
3. Ensure that the location is correctly passed as a `./configure` option to phosphor-logging so the appropriate processing takes place.

lgon, 2017-03-23:
**# Tasklist for 1360**
- [ ] After the dependency graph is created by the CI script, if both phosphor-logging and any *-dbus-interfaces are in the dependency graph, manually insert a dependency from phosphor-logging to *-dbus-interfaces.
- [ ] Verify that *-dbus-interfaces 'make install' is correctly installing the YAML to a location like /usr/share.
- [ ] Ensure that the location is correctly passed as a ./configure option to phosphor-logging so the appropriate processing takes place.

rfrandse, 2017-05-11:
https://gerrit.openbmc-project.xyz/3597 Add dbus interfaces dependency to phosphor logging
Fixes #1360 CI dependency for phosphor-logging.

closed 2017-06-06 by williamspatrick