openbmc/openbmc #1362: Fix GNU_HASH warnings in iicmaster

saqibkh, 2017-03-21:
While creating a build using yocto2.2 we found the following error:
Error: iicmaster-r0 do_package_qa: QA Issue: No GNU_HASH in the elf binary

As a work around we have added LDFLAGS to TARGET_CC_ARCH in the iicmaster.bb file to fix the warning.
To resolve the issue, upstream code needs to fix their build. (githublink: https://github.com/eddiejames/iicmaster)

Once completed please remove the "TODO" from openbmc/openbmc/meta-openbmc-machines/meta-openpower/common/recipes-bsp/iicmaster/iicmaster.bb

williamspatrick, 2017-03-21:
Please open issue for @eddiejames and reference that issue here.  Otherwise he isn't going to know to do it.

gtmills, 2018-06-01:
@eddiejames We still need a code change in iicmaster for this?