openbmc/openbmc #1363: Request OCC reset when fail detected

geissonator, 2017-03-21:
Per the OCC interface specification (we'll need to see what's applicable to open bmc with this story)

3.4.2	BMC-OCC Communication Failure
A communication failure is defined as one of the following:
•	Response Checksum failure
•	Sequence number mismatch after command timeout has been reached.
•	OCC Return Status still “In Progress” after command timeout has been reached.  See “Command Summary Table” section for recommended timeout by command.
•	Any non-successful Return Code
•	I2C Failure sending command or reading response

When any of the above communication failures occur the BMC should first verify that the “OCC Active” sensor is TRUE.  If the OCCs are not active the error should be ignored and communication with the OCC should not be retired until the “OCC Active” sensor is TRUE.  If the “OCC Active” sensor is TRUE the command should be retried twice.  If the command still fails after two retries and the “OCC Active” sensor is still “TRUE” and there is no checkstop the error is valid and a request to reset the OCCs should be sent.

geissonator, 2017-04-17:
Probably should break into at least 2 stories
- Add support to ipmid host interfaces to request the OCC reset
-- https://github.com/openbmc/phosphor-dbus-interfaces/blob/master/xyz/openbmc_project/Control/Host.interface.yaml
- Request reset when above conditions occur

geissonator, 2017-06-07:
Need to involve fan control team in this design, here's a comment from @spinler :
```
my concern is that if fan control runs into hwmon errors enough then it will die and the watchdog will permanently kick in.  However, if fan control can tell the OCC is unhappy, then it can gracefully just set fans to full speed until the OCC is reset and everything comes back to life again
```

geissonator, 2017-06-16:
Per the design meeting, we have 3 parts to this function (each getting 1 pt)
- #1802 
- #1698
- #1363 (this one)

geissonator, 2017-06-16:
From the interface, here's the occ reset flow:
```
Any request for an OCC reset will be resetting the whole power management complex in the system.  When BMC determines that it needs to request a reset the following must be done:

1.  BMC generates an error log with the reason for reset to aid in debug.
2.  BMC updates “OCC Active” sensor to FALSE for all OCCs
3.  BMC sends SMS Attention for OCC reset to OPAL with sensor ID of failing OCC
4.  OPAL reads IPMI command buffer and translates sensor ID to a chip ID and calls HBRT Process OCC Error interface with the chip ID
5.  HBRT calls HTMGT OCC error handling function with the chip ID
6.  HTMGT runs OCC Reset Procedure.
```

geissonator, 2017-06-16:
Some things to note with this story
- The occ reset goes through the SMS host ipmi path, but since this is openpower function, it will need a new interface into that function
- We need to pass the the sensor ID of the failing OCC up to the host, which means we'll need to enhance the host interface to take a parameter

geissonator, 2017-06-29:
Needs some implementation discussion tomorrow

geissonator, 2017-06-30:
Discussed technical details in interlock today, Vishwa ok with committing.

vishwabmc, 2017-07-13:
OCC error detection code has been up for review since some days.
Interface yaml for open_power host interface is up for review for few days.
Queue Manager is now implemented and I plugged the old host command interface to use this new Manager and its working fine so far

rfrandse, 2017-07-25:
<https://gerrit.openbmc-project.xyz/5475> Reset OCC on error condition
Fixes  #1363 Request OCC reset when fail detected

dkodihal, 2017-08-03:
This commit chain is yet to be merged - https://gerrit.openbmc-project.xyz/#/c/5331/.

vishwabmc, 2017-08-18:
@dkodihal This can now be closed since all the commits have been merged

closed 2017-08-18 by dkodihal