openbmc/openbmc #1367: REST login intermittent fails

geissonator, 2017-03-22:
We've been seeing this with our automated jobs that check the health of the system after someone releases it.  I caught a journal of it happening on the openbmc side.

The system was at BMC_READY and running fine.

@gkeishin can provide some logs from the client side (?) but seems like we're just exceeding the 10s timeout allowed here?

```
[barreleye]# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.3-118"
VERSION_ID="v1.99.3-118-g0032233"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.3-118"
BUILD_ID="v1.99.3"

Mar 21 09:25:59 barreleye systemd[1]: Started SSH Per-Connection Server (9.41.165.103:34572).
Mar 21 09:25:59 barreleye dropbear[979]: Failed loading /etc/dropbear/dropbear_dss_host_key
Mar 21 09:25:59 barreleye dropbear[979]: Failed loading /etc/dropbear/dropbear_ecdsa_host_key
Mar 21 09:25:59 barreleye dropbear[979]: Child connection from ::ffff:9.41.165.103:34572
Mar 21 09:26:00 barreleye dropbear[979]: PAM password auth succeeded for 'root' from ::ffff:9.41.165.103:34572
Mar 21 09:26:01 barreleye systemd[1]: Started Phosphor Webserver.
Mar 21 09:26:03 barreleye hwmon.py[802]: Cannot read attributes: /sys/class/hwmon/hwmon3/fan5_input
Mar 21 09:26:11 barreleye systemd[1]: Started SSH Per-Connection Server (9.41.165.103:34576).
Mar 21 09:26:11 barreleye dropbear[983]: Failed loading /etc/dropbear/dropbear_dss_host_key
Mar 21 09:26:11 barreleye dropbear[983]: Failed loading /etc/dropbear/dropbear_ecdsa_host_key
Mar 21 09:26:11 barreleye dropbear[983]: Child connection from ::ffff:9.41.165.103:34576
Mar 21 09:26:12 barreleye phosphor-hwmon-readd[974]: Remote I/O error
Mar 21 09:26:12 barreleye systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x401c0-nct7904\x402e.service: Main process exited, code=exited, status=1/FAILURE
Mar 21 09:26:13 barreleye systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x401c0-nct7904\x402e.service: Unit entered failed state.
Mar 21 09:26:13 barreleye systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x401c0-nct7904\x402e.service: Failed with result 'exit-code'.
Mar 21 09:26:13 barreleye systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x401c0-nct7904\x402e.service: Service hold-off time over, scheduling restart.
Mar 21 09:26:13 barreleye systemd[1]: Stopped Phosphor Hwmon Poller.
Mar 21 09:26:13 barreleye systemd[1]: Started Phosphor Hwmon Poller.
Mar 21 09:26:16 barreleye dropbear[983]: PAM password auth succeeded for 'root' from ::ffff:9.41.165.103:34576
```

gkeishin, 2017-03-22:
One case of failure here
`Power on REST call failure`

`Client side log`

```
Power On Test :: Power off and on.                                    .FURI:https://9.3.185.224/xyz/openbmc_project/state/host0//attr/CurrentHostState, method:Get, args:{}
Response code:200, Content:{
  "data": "xyz.openbmc_project.State.Host.HostState.Off", 
  "message": "200 OK", 
  "status": "ok"
}
Power On Test :: Power off and on.                                    | FAIL |
ReadTimeout: HTTPSConnectionPool(host='9.3.185.224', port=443): Read timed out. (read timeout=10.0): FAIL != PASS
```

```
RequestsLibrary.RequestsKeywords . Post Request openbmc, /login, data=${data}, headers=${headers}
Documentation: 	
Send a POST request on the session object found using the
Start / End / Elapsed: 	20170322 00:26:47.476 / 20170322 00:26:57.764 / 00:00:10.288
00:26:57.764 	FAIL 	ReadTimeout: HTTPSConnectionPool(host='9.3.185.224', port=443): Read timed out. (read timeout=10.0) 	
00:26:57.765 	INFO 	${status} = FAIL 	
00:26:57.765 	INFO 	${resp} = ReadTimeout: HTTPSConnectionPool(host='9.3.185.224', port=443): Read timed out. (read timeout=10.0)
```

`Server side log`

```
Mar 20 22:09:16 witherspoon systemd[1]: Created slice system-obmc\x2dconsole\x2dssh.slice.
Mar 20 22:09:16 witherspoon systemd[1]: Started Phosphor Host Console SSH Per-Connection (9.193.103.55:51718).
Mar 20 22:09:16 witherspoon systemd[1]: Started Phosphor Webserver.
Mar 20 22:09:16 witherspoon dropbear[905]: Failed loading /etc/dropbear/dropbear_dss_host_key
Mar 20 22:09:16 witherspoon dropbear[905]: Failed loading /etc/dropbear/dropbear_ecdsa_host_key
Mar 20 22:09:16 witherspoon dropbear[905]: Child connection from ::ffff:9.193.103.55:51718
Mar 20 22:09:19 witherspoon dropbear[905]: PAM password auth succeeded for 'root' from ::ffff:9.193.103.55:51718
Mar 20 22:09:29 witherspoon phosphor-gevent[906]: Traceback (most recent call last):
Mar 20 22:09:29 witherspoon phosphor-gevent[906]:   File "/usr/lib/python2.7/site-packages/gevent/greenlet.py", line 327, in run
Mar 20 22:09:29 witherspoon phosphor-gevent[906]:     result = self._run(*self.args, **self.kwargs)
Mar 20 22:09:29 witherspoon phosphor-gevent[906]:   File "/usr/lib/python2.7/site-packages/gevent/server.py", line 102, in wrap_socket_and_handle
Mar 20 22:09:29 witherspoon phosphor-gevent[906]:     ssl_socket = self.wrap_socket(client_socket, **self.ssl_args)
Mar 20 22:09:29 witherspoon phosphor-gevent[906]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 404, in wrap_socket
Mar 20 22:09:29 witherspoon phosphor-gevent[906]:     ciphers=ciphers)
Mar 20 22:09:29 witherspoon phosphor-gevent[906]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 113, in __init__
Mar 20 22:09:29 witherspoon phosphor-gevent[906]:     self.do_handshake()
Mar 20 22:09:29 witherspoon phosphor-gevent[906]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 324, in do_handshake
Mar 20 22:09:29 witherspoon phosphor-gevent[906]:     return self._sslobj.do_handshake()
Mar 20 22:09:29 witherspoon phosphor-gevent[906]: SSLEOFError: EOF occurred in violation of protocol (_ssl.c:590)
Mar 20 22:09:29 witherspoon phosphor-gevent[906]: <Greenlet at 0x76484260: <bound method WSGIServer.wrap_socket_and_handle of <WSGIServer at 0x76465550 fileno=5 address=('::80ac:9448:444c:a576%1984378352', 443, 0, 1984378352)>>(<socket at 0x7649aa10 fileno=9>, ('::c0d3:a376:8e91:f402%1217630632', 51719, 0, 121)> failed with SSLEOFError
Mar 20 22:09:30 witherspoon phosphor-gevent[906]: ::c0d3:a376:8e91:f402%1217630632 - - [2017-03-20 22:09:30] "POST /login HTTP/1.1" 200 320 0.048332
Mar 20 22:09:31 witherspoon phosphor-gevent[906]: ::c0d3:a376:8e91:f402%1217630632 - - [2017-03-20 22:09:31] "GET /xyz/openbmc_project/state/host0//attr/CurrentHostState HTTP/1.1" 200 212 0.059510
Mar 20 22:09:35 witherspoon systemd[1]: Created slice system-dropbear.slice.
Mar 20 22:09:35 witherspoon systemd[1]: Started SSH Per-Connection Server (9.193.103.55:51724).
Mar 20 22:09:35 witherspoon dropbear[910]: Failed loading /etc/dropbear/dropbear_dss_host_key
Mar 20 22:09:35 witherspoon dropbear[910]: Failed loading /etc/dropbear/dropbear_ecdsa_host_key
Mar 20 22:09:35 witherspoon dropbear[910]: Child connection from ::ffff:9.193.103.55:51724
Mar 20 22:09:37 witherspoon dropbear[910]: PAM password auth succeeded for 'root' from ::ffff:9.193.103.55:51724
Mar 20 22:09:40 witherspoon systemd[1]: Started SSH Per-Connection Server (9.193.103.55:51725).
Mar 20 22:09:40 witherspoon dropbear[916]: Failed loading /etc/dropbear/dropbear_dss_host_key
Mar 20 22:09:40 witherspoon dropbear[916]: Failed loading /etc/dropbear/dropbear_ecdsa_host_key
Mar 20 22:09:40 witherspoon dropbear[916]: Child connection from ::ffff:9.193.103.55:51725
Mar 20 22:09:42 witherspoon dropbear[916]: PAM password auth succeeded for 'root' from ::ffff:9.193.103.55:51725
Mar 20 22:09:53 witherspoon systemd[1]: Started SSH Per-Connection Server (9.193.103.55:51726).
Mar 20 22:09:53 witherspoon dropbear[925]: Failed loading /etc/dropbear/dropbear_dss_host_key
Mar 20 22:09:53 witherspoon dropbear[925]: Failed loading /etc/dropbear/dropbear_ecdsa_host_key
```

- [x] Journal log 
[journald_v1.99.3-124-gbbe2cbc_REST_Connection_error.txt](https://github.com/openbmc/openbmc/files/860722/journald_v1.99.3-124-gbbe2cbc_REST_Connection_error.txt)

williamspatrick, 2017-03-22:
This looks like a bad client behavior in the SSL protocol. We had this happen before on the IBM firmware pool machines running an old RHEL when they did an upgrade to openssl. Are we back to seeing that?  We have not changed anything recently on our end for SSL.

geissonator, 2017-03-22:
This is running in a docker container, seems fairly recent:
/$ ssh -V
OpenSSH_7.2p2 Ubuntu-4ubuntu2.1, OpenSSL 1.0.2g  1 Mar 2016

gkeishin, 2017-03-29:
For reference https://github.com/openbmc/openbmc/issues/612

lgon, 2017-03-31:
I have not been able to reproduce the failure on the system in the log file, or on other systems. I have noticed that the Jenkins parameter SystemType is set to "barreleye" instead of "witherspoon" Also, the extended/standby_bmc.robot test checks for the PDU_IP, PDU_SLOT_NO, PDU_USERNAME, PDU_TYPE, PDU_PASSWORD environment variables, but those aren't set in the  jenkins environmental parameters, so the test doesn't reboot via PDU. Is that intentional?  http://tulpok01p4.aus.stglabs.ibm.com:8080/jenkins/job/OpenBMC/job/automated-test/job/helper-jobs/job/AES-Standby/1752/injectedEnvVars/

gkeishin, 2017-04-03:
@lgon Here are couple of info
1. The PDU doesnt work on eaton switch management , so those system managing witherspoon it doesnt work PDU cycle from the test code. Need to find script or way to do so

2.  The system type doesn't matter in this case since the test code doesnt do anything with it.. it just a parameter defaulted to.

3. The best way to recreate is just do REST Get operation request and you should be able to see it from time to time.. first time not guaranteed.

lgon, 2017-04-03:
@gkeishin    The gevent server can use an SSL Context configured with "supress_ragged_eofs". The problem might be that the client fails to send an SSL shutdown, but just closes the socket when communication is complete, and setting this option would keep the exception from being raised.

geissonator, 2017-04-04:
Seems like the issue usually hits the first time we hit the REST server?  I know the REST server doesn't actually start until it's accessed so maybe there's an issue in the timing on that first startup?

Here was my fail today that shows the initial fail then the subsequent logins working
```
Mar 30 06:01:25 barreleye systemd[1]: Started Phosphor Webserver.
Mar 30 06:01:41 barreleye phosphor-gevent[1229]: Traceback (most recent call last):
Mar 30 06:01:41 barreleye phosphor-gevent[1229]:   File "/usr/lib/python2.7/site-packages/gevent/greenlet.py", line 327, in run
Mar 30 06:01:41 barreleye phosphor-gevent[1229]:     result = self._run(*self.args, **self.kwargs)
Mar 30 06:01:41 barreleye phosphor-gevent[1229]:   File "/usr/lib/python2.7/site-packages/gevent/server.py", line 102, in wrap_socket_and_handle
Mar 30 06:01:42 barreleye phosphor-gevent[1229]:     ssl_socket = self.wrap_socket(client_socket, **self.ssl_args)
Mar 30 06:01:42 barreleye phosphor-gevent[1229]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 404, in wrap_socket
Mar 30 06:01:42 barreleye phosphor-gevent[1229]:     ciphers=ciphers)
Mar 30 06:01:42 barreleye phosphor-gevent[1229]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 113, in __init__
Mar 30 06:01:42 barreleye phosphor-gevent[1229]:     self.do_handshake()
Mar 30 06:01:42 barreleye phosphor-gevent[1229]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 324, in do_handshake
Mar 30 06:01:42 barreleye phosphor-gevent[1229]:     return self._sslobj.do_handshake()
Mar 30 06:01:42 barreleye phosphor-gevent[1229]: SSLEOFError: EOF occurred in violation of protocol (_ssl.c:590)
Mar 30 06:01:42 barreleye phosphor-gevent[1229]: <Greenlet at 0xb657d2b0L: <bound method WSGIServer.wrap_socket_and_handle of <WSGIServer at 0xb655f270L fileno=5 address=('::14bb:a941:60fd:b8b6%3065576784', 443, 0, 3065576784L)>>(<socket at 0xb6593850L fileno=9>, ('::6009:b7b6:8e91:f402%1101571068', 54562, 0, 110)> failed with SSLEOFError
Mar 30 06:01:42 barreleye phosphor-gevent[1229]: ::6009:b7b6:8e91:f402%1101571068 - - [2017-03-30 06:01:42] "POST /login HTTP/1.1" 200 320 0.063845
Mar 30 06:01:42 barreleye phosphor-gevent[1229]: ::6009:b7b6:8e91:f402%1101571068 - - [2017-03-30 06:01:42] "GET /xyz/openbmc_project/state/host0//attr/CurrentHostState HTTP/1.1" 200 212 0.074204
Mar 30 06:01:45 barreleye systemd[1]: Started SSH Per-Connection Server (9.41.165.103:58330).
Mar 30 06:01:45 barreleye dropbear[1231]: Failed loading /etc/dropbear/dropbear_dss_host_key
Mar 30 06:01:45 barreleye dropbear[1231]: Failed loading /etc/dropbear/dropbear_ecdsa_host_key
Mar 30 06:01:45 barreleye dropbear[1231]: Child connection from ::ffff:9.41.165.103:58330
Mar 30 06:01:47 barreleye dropbear[1231]: PAM password auth succeeded for 'root' from ::ffff:9.41.165.103:58330
Mar 30 06:01:48 barreleye systemd[1]: Started SSH Per-Connection Server (9.41.165.103:58332).
Mar 30 06:01:48 barreleye dropbear[1238]: Failed loading /etc/dropbear/dropbear_dss_host_key
Mar 30 06:01:48 barreleye dropbear[1238]: Failed loading /etc/dropbear/dropbear_ecdsa_host_key
Mar 30 06:01:48 barreleye dropbear[1238]: Child connection from ::ffff:9.41.165.103:58332
Mar 30 06:01:50 barreleye dropbear[1238]: PAM password auth succeeded for 'root' from ::ffff:9.41.165.103:58332
Mar 30 06:01:54 barreleye systemd[1]: Started SSH Per-Connection Server (9.41.165.103:58334).
Mar 30 06:01:54 barreleye dropbear[1248]: Failed loading /etc/dropbear/dropbear_dss_host_key
Mar 30 06:01:54 barreleye dropbear[1248]: Failed loading /etc/dropbear/dropbear_ecdsa_host_key
Mar 30 06:01:54 barreleye dropbear[1248]: Child connection from ::ffff:9.41.165.103:58334
Mar 30 06:01:56 barreleye dropbear[1248]: PAM password auth succeeded for 'root' from ::ffff:9.41.165.103:58334
Mar 30 06:01:57 barreleye phosphor-gevent[1229]: ::6009:b7b6:8e91:f402%1101571068 - - [2017-03-30 06:01:57] "POST /login HTTP/1.1" 200 320 0.072758
Mar 30 06:01:59 barreleye phosphor-gevent[1229]: ::6009:b7b6:8e91:f402%1101571068 - - [2017-03-30 06:01:59] "GET /xyz/openbmc_project/logging/entry/enumerate HTTP/1.1" 200 18081 2.120437
Mar 30 06:02:00 barreleye phosphor-gevent[1229]: ::6009:b7b6:8e91:f402%1101571068 - - [2017-03-30 06:02:00] "POST /login HTTP/1.1" 200 320 0.042991
```

williamspatrick, 2017-04-04:
I suspect this is due to the RNG not having enough entropy and blocking SSL. The client times out waiting for the TLS handshake and ends the connection?  Perhaps a driver for the hardware RNG would solve some aspect of this.

geissonator, 2017-04-04:
Some thoughts from our meeting:
- Talk with Brad about supress_ragged_eofs option
- Look into having the web server service check to verify there is enough entropy before starting it (so users will just get an instant reject until we can start the server)
- RNG driver coming which could be ultimate fix

lgon, 2017-04-04:
One temporary option could be to use haveged to increase entropy. 
http://www.issihosts.com/haveged/

> How does the havege algorithm generate random numbers?
>     In effect, the havege algorithm gathers the random timing variations encountered in benchmarking a PRNG calculation. The calculation is constructed to be sensitive to the effects of hardware interrupts and the output continually reseeded by XOR differences of successive iterations of the benchmark!

kc8apf, 2017-04-04:
I've sent @shenki DT changes for the Aspeed's HW RNG.  The driver in dev-4.7 will work but is limited to 4  bytes of random per 10ms.  I've sent a rewrite of the driver to mainline (http://www.mail-archive.com/linux-crypto@vger.kernel.org/msg24461.html) that gets to the full 4 bytes per 1us the hardware supports.

sivassrr, 2017-04-13:
Observing more HTTPSConnectionPool(host=‘wsbmc005’, port=443): Read timed out. (read timeout=10.0): this kind of error.

lgon, 2017-05-05:
After talking with Patrick, I'm changing the approach to fix this problem, and instead of checking entropy and blocking until threshold is reached, going to use rngd.
**Tasklist**
- [ ] Add rngd package to image.
- [ ] .bb append to remove init script, and add custom systemd script.

rfrandse, 2017-05-17:
https://gerrit.openbmc-project.xyz/3580 Add phosphor-gevent-watchdog.sh to test entropy
Resolves #1367 REST login intermittent fails

closed 2017-06-23 by williamspatrick