openbmc/openbmc #1368: New OCC sensors - performance mode and memory power down

geissonator, 2017-03-22:
New OCC sensors being defined.

- Maximum Performance Mode vs Nominal Performance Mode
-- Just a 0/1 type value, default is 0 (nominal), Maximum is 1
- Memory Power Down settings
-- Just a 0x00 type value, TBD on what the values mean

vishwabmc, 2017-09-07:
# **TASK LIST**

vishwabmc, 2017-09-08:
# **STATUS**

- 9/8:
   - Decided to write 2 YAML files and 2 implementations based on the discussion. Have not started writing yaml

vishwabmc, 2017-09-11:
I looked at the Power Thermal document and it defines following Modes.

 -- Nominal
 -- Nominal dynamic performance
 -- Maximum dynamic performance
 -- Dynamic Power - Favor Performance ( not sure what this is )
 -- Fixed max frequency

Will talk to OCC folks and see what is really needed from this doc.

vishwabmc, 2017-09-12:
Discussed with Todd on this and here is what it is : 

- Only Nominal and Maximum are needed since the rest are only for PowerVM
- Boolean 0/1 for the Performance Mode
- No need of the Memory Power Down sensor
```
Todd Rosedahl [3:06 AM] 
Hi.  So I checked around and since Witherspoon will default this to 0x00, there is no need to implement anything here actually.  So that saves you some work.

[3:07] 
We do not need to override it to other values.  And no need for an OpenBMC sensor for this at all```

mdmillerii, 2017-09-12:
Using a single boolean to control power / performance trade-offs seems like a tatical decision for witherspoon versus a design for many open power (or even other power9) systems.

@bradbishop @geissonator Agree?

vishwabmc, 2017-09-19:
@geissonator please cancel this issue per the email I forwarded last week.

Todd indicated that we do not do any work on this for now.

geissonator, 2017-09-20:
Closing out per discussion with Todd

closed 2017-09-20 by geissonator