openbmc/openbmc #1372: phosphor logging fails to compile

amboar, 2017-03-23:
Building from openbmc/openbmc@29c30bd005a7fff96764ba86a8570ca90ee261dc with some other unrelated commits on top:

```
In file included from ../git/elog_meta.cpp:2:0:
../git/elog_meta.hpp:110:12: error: 'xyz' was not declared in this scope
 void build<xyz::openbmc_project::Common::
            ^
../git/elog_meta.hpp:110:12: note: suggested alternatives:
In file included from ../git/elog_entry.hpp:5:0,
                 from ../git/elog_meta.hpp:9,
                 from ../git/elog_meta.cpp:2:
/home/andrew/src/openbmc/openbmc/build/tmp/sysroots/zaius/usr/include/xyz/openbmc_project/Logging/Entry/server.hpp:9:1: note:   'sdbusplus::xyz'
 {
 ^
In file included from ../git/elog_meta.hpp:8:0,
                 from ../git/elog_meta.cpp:2:
./phosphor-logging/elog-errors.hpp:22:1: note:   'phosphor::logging::example::xyz'
 {
 ^
In file included from ../git/elog_meta.cpp:2:0:
../git/elog_meta.hpp:110:6: error: parse error in template argument list
 void build<xyz::openbmc_project::Common::
      ^
../git/elog_meta.hpp:110:6: error: template-id 'build<<expression error> >' for 'void phosphor::logging::metadata::associations::build(const string&, const std::vector<std::__cxx11::basic_string<char> >&, phosphor::logging::AssociationList&)' does not match any template declaration
../git/elog_meta.cpp:16:12: error: 'xyz' was not declared in this scope
 void build<xyz::openbmc_project::Common::
            ^
../git/elog_meta.cpp:16:12: note: suggested alternatives:
In file included from ../git/elog_entry.hpp:5:0,
                 from ../git/elog_meta.hpp:9,
                 from ../git/elog_meta.cpp:2:
/home/andrew/src/openbmc/openbmc/build/tmp/sysroots/zaius/usr/include/xyz/openbmc_project/Logging/Entry/server.hpp:9:1: note:   'sdbusplus::xyz'
 {
 ^
In file included from ../git/elog_meta.hpp:8:0,
                 from ../git/elog_meta.cpp:2:
./phosphor-logging/elog-errors.hpp:22:1: note:   'phosphor::logging::example::xyz'
 {
 ^
../git/elog_meta.cpp:16:6: error: parse error in template argument list
 void build<xyz::openbmc_project::Common::
      ^
../git/elog_meta.cpp:16:6: error: template-id 'build<<expression error> >' for 'void phosphor::logging::metadata::associations::build(const string&, const std::vector<std::__cxx11::basic_string<char> >&, phosphor::logging::AssociationList&)' does not match any template declaration
Makefile:825: recipe for target 'libphosphor_logging_la-elog_meta.lo' failed
make[2]: *** [libphosphor_logging_la-elog_meta.lo] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from ../git/log_manager.cpp:16:0:
../git/elog_meta.hpp:110:12: error: 'xyz' was not declared in this scope
 void build<xyz::openbmc_project::Common::
            ^
../git/elog_meta.hpp:110:12: note: suggested alternatives:
In file included from ../git/elog_entry.hpp:5:0,
                 from ../git/log_manager.cpp:13:
/home/andrew/src/openbmc/openbmc/build/tmp/sysroots/zaius/usr/include/xyz/openbmc_project/Logging/Entry/server.hpp:9:1: note:   'sdbusplus::xyz'
 {
 ^
In file included from ../git/elog_meta.hpp:8:0,
                 from ../git/log_manager.cpp:16:
./phosphor-logging/elog-errors.hpp:22:1: note:   'phosphor::logging::example::xyz'
 {
 ^
In file included from ../git/log_manager.cpp:16:0:
../git/elog_meta.hpp:110:6: error: parse error in template argument list
 void build<xyz::openbmc_project::Common::
      ^
../git/elog_meta.hpp:110:6: error: template-id 'build<<expression error> >' for 'void phosphor::logging::metadata::associations::build(const string&, const std::vector<std::__cxx11::basic_string<char> >&, phosphor::logging::AssociationList&)' does not match any template declaration
Makefile:874: recipe for target 'phosphor_log_manager-log_manager.o' failed
make[2]: *** [phosphor_log_manager-log_manager.o] Error 1
make[2]: Leaving directory '/home/andrew/src/openbmc/openbmc/build/tmp/work/armv6-openbmc-linux-gnueabi/phosphor-logging/1.0-r1/build'
Makefile:1052: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory '/home/andrew/src/openbmc/openbmc/build/tmp/work/armv6-openbmc-linux-gnueabi/phosphor-logging/1.0-r1/build'
Makefile:548: recipe for target 'all' failed
make: *** [all] Error 2
ERROR: oe_runmake failed
ERROR: Function failed: do_compile (log file is located at /home/andrew/src/openbmc/openbmc/build/tmp/work/armv6-openbmc-linux-gnueabi/phosphor-logging/1.0-r1/temp/log.do_compile.27034)
```

Any thoughts @williamspatrick @anoo1 ?

williamspatrick, 2017-03-23:
Likely a missing dependency?

msbarth, 2017-03-23:
Similar errors occur when compiling within the unit test docker image for a package attempting to log an error with a callout. Same code compiles within SDK, possibly also missing configure flags within unit test script(unit-test.py) for phosphor-logging?

For change -> https://gerrit.openbmc-project.xyz/#/c/2948/
```22:02:04 sysfs.cpp: In function 'uint64_t writeSysfsWithCallout(const uint64_t&, const string&, const string&)':
22:02:04 sysfs.cpp:124:18: error: 'xyz' was not declared in this scope
22:02:04              elog<xyz::openbmc_project::Control::Device::WriteFailure>(
22:02:04                   ^
22:02:04 sysfs.cpp:124:13: error: parse error in template argument list
22:02:04              elog<xyz::openbmc_project::Control::Device::WriteFailure>(
22:02:04              ^
22:02:04 sysfs.cpp:125:28: error: the value of 'xyz' is not usable in a constant expression
22:02:04                  prev_entry<xyz::openbmc_project::Control::
22:02:04                             ^
22:02:04 sysfs.cpp:124:18: note: 'xyz' was not declared 'constexpr'
22:02:04              elog<xyz::openbmc_project::Control::Device::WriteFailure>(
22:02:04                   ^
22:02:04 sysfs.cpp:126:63: error: type/value mismatch at argument 1 in template parameter list for 'template<class T> struct phosphor::logging::prev_entry'
22:02:04                             Device::WriteFailure::CALLOUT_ERRNO>(),
22:02:04                                                                ^
22:02:04 sysfs.cpp:126:63: note:   expected a type, got 'xyz'
22:02:04 sysfs.cpp:127:28: error: the value of 'xyz' is not usable in a constant expression
22:02:04                  prev_entry<xyz::openbmc_project::Control::
22:02:04                             ^
22:02:04 sysfs.cpp:124:18: note: 'xyz' was not declared 'constexpr'
22:02:04              elog<xyz::openbmc_project::Control::Device::WriteFailure>(
22:02:04                   ^
22:02:04 sysfs.cpp:128:69: error: type/value mismatch at argument 1 in template parameter list for 'template<class T> struct phosphor::logging::prev_entry'
22:02:04                             Device::WriteFailure::CALLOUT_DEVICE_PATH>());
22:02:04                                                                      ^
22:02:04 sysfs.cpp:128:69: note:   expected a type, got 'xyz'
22:02:04 sysfs.cpp:130:16: error: 'elogException' does not name a type
22:02:04          catch (elogException<xyz::openbmc_project::Control::
22:02:04                 ^
22:02:04 sysfs.cpp:130:29: error: expected ')' before '<' token
22:02:04          catch (elogException<xyz::openbmc_project::Control::
22:02:04                              ^
22:02:04 sysfs.cpp:130:29: error: expected '{' before '<' token
22:02:04 sysfs.cpp:130:29: error: expected primary-expression before '<' token
22:02:04 sysfs.cpp:130:30: error: 'xyz' has not been declared
22:02:04          catch (elogException<xyz::openbmc_project::Control::
22:02:04                               ^
22:02:04 Makefile:602: recipe for target 'libhwmon_la-sysfs.lo' failed```

amboar, 2017-03-23:
If someone wants to bisect, 7f0646c2cb1a3ac7758462646c191113e5321637 on master builds fine.

dkodihal, 2017-03-23:
Indeed a missing dependency. Pushed a fix here: https://gerrit.openbmc-project.xyz/#/c/3219.

@msbarth could you check if this helps you? If not, we may need an update to the CI/unit test script like we have been discussing.

closed 2017-03-24 by williamspatrick