openbmc/openbmc #1374: out of band IPMI doesn't work

stewart-ibm, 2017-03-23:
It working with @shenki to get op-test-framework to run on a Witherspoon (OpenBMC based), I enabled IPMI for OpenBMC in op-test-framework.

Unfortunately, we didn't get very far, as the `chassis power off` command failed like so:
```
Feb 22 07:00:49 w34 netipmid[860]: << RAKP12
Feb 22 07:00:49 w34 netipmid[860]: I> Channel::Read : DataIn Fd[3] Req[17] Recv[17]
Feb 22 07:00:49 w34 netipmid[860]: >> GetChannelCapabilities
Feb 22 07:00:49 w34 netipmid[860]: << GetChannelCapabilities
Feb 22 07:00:49 w34 netipmid[860]: I> Channel::Read : DataIn Fd[3] Req[30] Recv[30]
Feb 22 07:00:49 w34 netipmid[860]: >> openSession
Feb 22 07:00:49 w34 netipmid[860]: << openSession
Feb 22 07:00:50 w34 netipmid[860]: I> Channel::Read : DataIn Fd[3] Req[30] Recv[30]
Feb 22 07:00:50 w34 netipmid[860]: >> RAKP12
Feb 22 07:00:50 w34 netipmid[860]: Failed to get /org/openbmc/control/chassis0 bus name: Connection timed out
```

This was with `VERSION_ID="v1.99.3-110-g4a03f88` and https://github.com/open-power/op-test-framework as of https://github.com/open-power/op-test-framework/commit/b19d231096ac27fa137eaf07b22c6deb61032e24

sivassrr, 2017-03-23:
@rahulmah - Is it working for us?

williamspatrick, 2017-03-23:
I'm not sure if this is expected to work right now. You really should be using REST APIs as the primary method for interacting with a openBMC. We do not have full IPMI 2.0 support yet and it will always have less functionality than REST. 

Rather than reimplement all of this support that is already in openbmc-test-automation, is it possible for you to use those robot/Python routines?

stewart-ibm, 2017-03-23:
Patrick Williams <notifications@github.com> writes:
> I'm not sure if this is expected to work right now. You really should
> be using REST APIs as the primary method for interacting with a
> openBMC. We do not have full IPMI 2.0 support yet and it will [...]

Maybe? Are there simple python routines we could grab for doing things
like power on/off, set boot device, get sensors ?

The REST API itself seems sparsely documented when it comes to
functionality like powering the host on and off.

-- 
Stewart Smith
OPAL Architect, IBM.

williamspatrick, 2017-03-23:
> The REST API itself seems sparsely documented when it comes to
> functionality like powering the host on and off.

Yeah, that is by design.  If you read the documentation for the REST API it tells you the REST API is just an automatic introspection of the dbus objects.  The dbus API is where the bulk of the documentation is.

https://github.com/openbmc/phosphor-dbus-interfaces

stewart-ibm, 2017-03-24:
We may have to go that route, but I was under the impression that at least for basics (like turning the host on and off) that IPMI was meant to work now.

stewart-ibm, 2017-03-31:
Well, IPMI power off seems to function, so this is further than I got before, but immediately hit https://github.com/openbmc/openbmc/issues/1416

closed 2017-03-31 by stewart-ibm