openbmc/openbmc #1375: [REST]  REST entry shows "associations" as EMPTY list

gkeishin, 2017-03-23:
`The REST interface shows "associations" as EMPTY list..`

`Driver Info`
```
- [x] SYSTEM     : W42 ( Witherspoon - 2 Sockets - 16 DIMMs ) 
- [x] BUILD_ID   : https://openpower.xyz/job/openbmc-build/647/
- [x] VERSION_ID : v1.99.3-126-g29c30bd
- [x] Kernel     : Linux witherspoon 4.7.10-c72ced5a19ceae19c7d792ae3e982bee5607dab4 #1 Wed Mar 22 18:22:25 UTC 2017 armv6l GNU/Linux
- [x] PNOR VERSION:
IBM-witherspoon-ibm-OP9_v1.15_1.3
        op-build-v1.15-15-gce29449
        buildroot-2017.02-4-gc28573e
        skiboot-28e96ff
        hostboot-86bae1c
        linux-4.10.1-p976d063
        petitboot-v1.4.1-p3ec3d6e
        witherspoon-xml-065aaa2
        occ-4141b5f
        hostboot-binaries-4b44090
        capp-ucode-1bb7503
```
```
Executing: GET  /xyz/openbmc_project/logging/enumerate

/xyz/openbmc_project/logging/enumerate
{
    "/xyz/openbmc_project/logging/entry/1": {
        "AdditionalData": [],
        "Id": 1,
        "Message": "org.open_power.Error.Host.Event.Event",
        "Severity": "xyz.openbmc_project.Logging.Entry.Level.Informational",
        "Timestamp": 1490047788527,
        "associations": []
    },
    "/xyz/openbmc_project/logging/entry/2": {
        "AdditionalData": [
            "ESEL=00 00 df 00 00 00 00 20 00 04 12 01 6f aa 00 00 50 48 00 30 01 00 17 00 00 00 00 04 31 71 56 4c 00 00 00 04 31 73 df b2 42 00 00 07 00 00 00 00 00 00 00 00 00 00 00 00 90 00 00 28 90 00 00 28 55 48 00 18 01 00 17 00 8a 03 40 00 00 00 00 00 ff ff 00 00 00 00 00 00 50 53 00 50 01 01 00 00 02 00 00 09 17 0e 00 48 00 00 00 e0 00 00 0a 00 00 00 00 00 00 20 00 00 00 05 00 01 00 2d 00 04 00 00 00 00 00 00 00 00 42 43 38 41 31 37 30 45 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 55 44 00 1c 01 06 01 00 02 54 41 4b 00 00 00 06 00 00 00 55 00 03 b6 94 00 00 00 00 55 44 00 4c 01 02 01 00 ee ee ee ee 54 61 72 67 65 74 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 05 00 01 00 00 00 03 00 00 00 05 00 00 00 14 0f 7a bb 7c 23 01 00 02 00 05 01 0b 5a fc d7 13 01 00 02 00 05 01 00 00 55 44 00 30 01 03 01 00 00 00 00 00 00 03 f3 54 00 00 00 00 00 03 b6 94 00 00 00 00 40 03 dd 68 00 00 00 00 40 03 ba 48 00 00 00 00 00 00 25 80 55 44 00 38 01 01 01 00 48 6f 73 74 62 6f 6f 74 20 42 75 69 6c 64 20 49 44 3a 20 68 6f 73 74 62 6f 6f 74 2d 38 36 62 61 65 31 63 2f 68 62 69 63 6f 72 65 2e 62 69 6e 00 "
        ],
        "Id": 2,
        "Message": "org.open_power.Error.Host.Event.Event",
        "Severity": "xyz.openbmc_project.Logging.Entry.Level.Informational",
        "Timestamp": 1490047791799,
        "associations": []
    },
    "/xyz/openbmc_project/logging/internal/manager": {}
}
```

```
root@witherspoon:~# busctl introspect xyz.openbmc_project.Logging /xyz/openbmc_project/logging/entry/1 | cat
NAME                                TYPE      SIGNATURE RESULT/VALUE                             FLAGS
org.freedesktop.DBus.Introspectable interface -         -                                        -
.Introspect                         method    -         s                                        -
org.freedesktop.DBus.Peer           interface -         -                                        -
.GetMachineId                       method    -         s                                        -
.Ping                               method    -         -                                        -
org.freedesktop.DBus.Properties     interface -         -                                        -
.Get                                method    ss        v                                        -
.GetAll                             method    s         a{sv}                                    -
.Set                                method    ssv       -                                        -
.PropertiesChanged                  signal    sa{sv}as  -                                        -
org.openbmc.Associations            interface -         -                                        -
.associations                       property  a(sss)    0                                        emits-change writable
xyz.openbmc_project.Logging.Entry   interface -         -                                        -
.AdditionalData                     property  as        0                                        emits-change writable
.Id                                 property  u         1                                        emits-change writable
.Message                            property  s         "org.open_power.Error.Host.Event.Event"  emits-change writable
.Severity                           property  s         "xyz.openbmc_project.Logging.Entry.Le... emits-change writable
.Timestamp                          property  t         1490047788527                            emits-change writable
root@witherspoon:~# busctl introspect xyz.openbmc_project.Logging /xyz/openbmc_project/logging/entry/2 | cat
NAME                                TYPE      SIGNATURE RESULT/VALUE                             FLAGS
org.freedesktop.DBus.Introspectable interface -         -                                        -
.Introspect                         method    -         s                                        -
org.freedesktop.DBus.Peer           interface -         -                                        -
.GetMachineId                       method    -         s                                        -
.Ping                               method    -         -                                        -
org.freedesktop.DBus.Properties     interface -         -                                        -
.Get                                method    ss        v                                        -
.GetAll                             method    s         a{sv}                                    -
.Set                                method    ssv       -                                        -
.PropertiesChanged                  signal    sa{sv}as  -                                        -
org.openbmc.Associations            interface -         -                                        -
.associations                       property  a(sss)    0                                        emits-change writable
xyz.openbmc_project.Logging.Entry   interface -         -                                        -
.AdditionalData                     property  as        1 "ESEL=00 00 df 00 00 00 00 20 00 04... emits-change writable
.Id                                 property  u         2                                        emits-change writable
.Message                            property  s         "org.open_power.Error.Host.Event.Event"  emits-change writable
.Severity                           property  s         "xyz.openbmc_project.Logging.Entry.Le... emits-change writable
.Timestamp                          property  t         1490047791799                            emits-change writable
root@witherspoon:~#

````

anoo1, 2017-03-23:
I believe Andrew mentioned that ESELs are not expected to have associations, but @dkodihal can comment on this.

dkodihal, 2017-03-23:
I think @bradbishop suggested make a change in the rest server to not show the associations object when it's empty.

williamspatrick, 2017-04-03:
@bradbishop - I think we need a mechanism to hide interfaces and/or properties from rest. "Associations" is an example of a property that is deceptive. Since associations can be made outside of the interface with the association property, it is misleading to give this out to users.

gkeishin, 2017-09-23:
@anoo1 We can close this one up ?