openbmc/openbmc #1383: Host ipmi dbus errors [ Witherspoon ]

gkeishin, 2017-03-24:
Looks like the dbus-send to send  a message bus to the host interfaces are seems to have not working on witherspoon, the same works good on non MRW systems.

logs and the command are in the in the note bellow
@ojayanth  is looking into it and he suggested to open this issue,

```
Feb 22 06:16:38 witherspoon ipmid[875]: Failed to parse signal message: No such device or address
Feb 22 06:18:05 witherspoon ipmid[875]: Failed to parse signal message: No such device or address
Feb 22 06:18:11 witherspoon ipmid[875]: Failed to call the method: No route to host
Feb 22 06:18:11 witherspoon ipmid[875]: Dest: :1.153, Path: /org/openbmc/HostIpmi/1
Feb 22 06:18:18 witherspoon ipmid[875]: Failed to get  busname: No such file or directory
Feb 22 06:18:18 witherspoon ipmid[875]: Failed to call the method: No route to host
Feb 22 06:18:18 witherspoon ipmid[875]: Dest: :1.155, Path: /org/openbmc/HostIpmi/1
Feb 22 06:18:26 witherspoon ipmid[875]: Failed to get  busname: No such file or directory
Feb 22 06:18:26 witherspoon ipmid[875]: Failed to call the method: No route to host
Feb 22 06:18:26 witherspoon ipmid[875]: Dest: :1.156, Path: /org/openbmc/HostIpmi/1
Feb 22 06:18:34 witherspoon ipmid[875]: Failed to get  busname: No such file or directory
Feb 22 06:18:34 witherspoon ipmid[875]: Failed to call the method: No route to host
Feb 22 06:18:34 witherspoon ipmid[875]: Dest: :1.157, Path: /org/openbmc/HostIpmi/1
Feb 22 06:18:42 witherspoon ipmid[875]: Failed to get  busname: No such file or directory
Feb 22 06:18:43 witherspoon ipmid[875]: Failed to call the method: No route to host
Feb 22 06:18:43 witherspoon ipmid[875]: Dest: :1.158, Path: /org/openbmc/HostIpmi/1
Feb 22 06:18:51 witherspoon ipmid[875]: Failed to get  busname: No such file or directory
Feb 22 06:18:51 witherspoon ipmid[875]: Failed to call the method: No route to host
Feb 22 06:18:51 witherspoon ipmid[875]: Dest: :1.159, Path: /org/openbmc/HostIpmi/1
Feb 22 06:18:59 witherspoon ipmid[875]: Failed to get  busname: No such file or directory
Feb 22 06:18:59 witherspoon ipmid[875]: Handling Netfn:[0x6], Cmd:[0x36]
Feb 22 06:18:59 witherspoon ipmid[875]: Handling Set-SEL-Time:[0xA], Cmd:[0x49]
Feb 22 06:18:59 witherspoon ipmid[875]: Data: 0x58D4CF38]
Feb 22 06:18:59 witherspoon ipmid[875]: Error [Permission denied] setting time
Feb 22 06:18:59 witherspoon ipmid[875]: Dest: :1.160, Path: /org/openbmc/HostIpmi/1
Feb 22 06:19:06 witherspoon ipmid[875]: Failed to get /org/openbmc/inventory/system/powerlimit busname: No such file or directory
Feb 22 06:19:06 witherspoon ipmid[875]: Failed to call the method: No route to host
Feb 22 06:19:06 witherspoon ipmid[875]: Dest: :1.161, Path: /org/openbmc/HostIpmi/1
Feb 22 06:19:14 witherspoon ipmid[875]: Failed to get /org/openbmc/inventory/system/powerlimit busname: No such file or directory
Feb 22 06:19:14 witherspoon ipmid[875]: Failed to call the method: Connection timed out
Feb 22 06:19:14 witherspoon ipmid[875]: Dest: :1.162, Path: /org/openbmc/HostIpmi/1

```

The test uses the dbus-send to send message to host 

**ipmi capabilities test**

`dbus-send --system /org/openbmc/HostIpmi/1 org.openbmc.HostIpmi.ReceivedMessage byte:0x00 byte:0x06 byte:0x00 byte:0x36`

**Set Sensor Boot Count**

`dbus-send --system /org/openbmc/HostIpmi/1 org.openbmc.HostIpmi.ReceivedMessage byte:0x00 byte:0x04 byte:0x00 byte:0x30 array:byte:95,0x01,0x00,0x35,0x00,0x00,0x00,0x00,0x00,0x00`

**Set Sensor Boot Progress**
`dbus-send --system /org/openbmc/HostIpmi/1 org.openbmc.HostIpmi.ReceivedMessage byte:0x00 byte:0x04 byte:0x00 byte:0x30 array:byte:5,0xa9,0x00,0x04,0x00,0x00,0x00,0x00,0x14,0x00`

**OS Status Sensor Progress**
`dbus-send --system /org/openbmc/HostIpmi/1 org.openbmc.HostIpmi.ReceivedMessage byte:0x00 byte:0x04 byte:0x00 byte:0x30 array:byte:181,0x00,0x00,0x04,0x00,0x00,0x00,0x00,0x20,0x00`

 **OCC Active Sensor On Enabled**
`dbus-send --system /org/openbmc/HostIpmi/1 org.openbmc.HostIpmi.ReceivedMessage byte:0x00 byte:0x04 byte:0x00 byte:0x30 array:byte:8,0x00,0x00,0x02,0x00,0x00,0x00,0x00,0x20,0x00`

- [x] Journald log 
[journald_v1.99.3-128-ga0040f7_host_ipmid.txt](https://github.com/openbmc/openbmc/files/867200/journald_v1.99.3-128-ga0040f7_host_ipmid.txt)

vishwabmc, 2017-03-24:
Feb 22 06:19:06 witherspoon ipmid[875]: Failed to get /org/openbmc/inventory/system/powerlimit busname: No such file or directory

Are you using old path ?

gkeishin, 2017-03-24:
we explicitly don't use that path, probably the dbus-send system command message finds it way out to that org path.. perhaps ?

@tomjoseph83 Any more you can add on this ?

vishwabmc, 2017-03-24:
dbus-send does not know anything .. All the input are to be given to it by the user.. dbus-send just send the command over to the path / service / interface provided by the user @gkeishin

tomjoseph83, 2017-04-04:
@gkeishin The sensor numbers are different for Witherspoon and Barreleye. The test cases is probably hardcoded based on the Barreleye Sensor numbers. Please take a look at the Witherspoon numbers and confirm the behaviour.

/openbmc/skeleton/configs/Witherspoon.py

gkeishin, 2017-04-19:
Closing as the code is now fixed in automation.. https://gerrit.openbmc-project.xyz/#/c/3540/

closed 2017-04-19 by gkeishin