openbmc/openbmc #1384: Can't ssh ast2500-evb

ppakxe, 2017-03-24:
I built ast2500-evb image and ran on qemu with bridged network.

Ping works but ssh doesn't.
(Both of host->qemu and qemu->host)

Traceroute works when host to qemu only, not opposite.

I think ICMP network layer works, but above doesn't.

How can I fix this?

(10.0.2.28 is ubuntu host, 10.0.4.186 is ast2500 on qemu)

[image]

[image]

williamspatrick, 2017-03-24:
Can you give the parameters you used to start qemu?  I don't think most of our developers use anything except SLIRP.

ppakxe, 2017-03-26:
./qemu-system-arm -M ast2500-evb -nographic -drive file=flash-evb-ast2500,if=mtd,format=raw -net nic -net bridge,br=br0

SSH runs perfectly on palmetto machine with same option above.

P.S what is the SLIRP? Thanks to reply.

williamspatrick, 2017-03-27:
This sounds like a qemu problem then. @amboar, who can look into this?

ppakxe, 2017-03-27:
So I tried other qemu in official website :
http://download.qemu-project.org/qemu-2.8.0.tar.xz

but it doesn't works with errorcode 0x00000000 dump error. 
I don't know what is different between openbmc/qemu and above qemu
I noticed that palmetto uses latter linux version; I can't remember correctly, there is difference 0.05 version.

When using nmap from host to qemu, port 22 seem to be filtered. (I'll add pics tomorrow)
but not to palmetto.

amboar, 2017-03-28:
@ozbenh has re-written the ftgmac driver and so @cdrx nees to make some adjustments to the qemu model, might be worth having @cdrx look at this as well given he'll be working in the area.

gtmills, 2018-07-05:
@ppakxe Still an issue or can this be closed?

gtmills, 2018-10-12:
@amboar Can we close?