openbmc/openbmc #1388: Witherspoon LED groups do not conform to dbus path standards.

williamspatrick, 2017-03-24:
```
      └─/xyz/openbmc_project/led/groups
        ├─/xyz/openbmc_project/led/groups/EnclosureFault
        ├─/xyz/openbmc_project/led/groups/EnclosureIdentify
        ├─/xyz/openbmc_project/led/groups/Fan0Fault
        ├─/xyz/openbmc_project/led/groups/Fan0Identify
        ├─/xyz/openbmc_project/led/groups/Fan1Fault
        ├─/xyz/openbmc_project/led/groups/Fan1Identify
        ├─/xyz/openbmc_project/led/groups/Fan2Fault
        ├─/xyz/openbmc_project/led/groups/Fan2Identify
        ├─/xyz/openbmc_project/led/groups/Fan3Fault
        ├─/xyz/openbmc_project/led/groups/Fan3Identify
        ├─/xyz/openbmc_project/led/groups/PowerOff
        ├─/xyz/openbmc_project/led/groups/PowerOn
        ├─/xyz/openbmc_project/led/groups/bmcFault
        ├─/xyz/openbmc_project/led/groups/boxelderFault
```

vishwabmc, 2017-03-27:
@williamspatrick Is it not same as #1312 ?

vishwabmc, 2017-03-27:
https://gerrit.openbmc-project.xyz/#/c/3294/

williamspatrick, 2017-03-28:
Yep.  Duplicate of #1312.  I forgot I already opened an issue for this.

closed 2017-03-28 by williamspatrick