openbmc/openbmc #1389: WSGIServer - IOError: [Errno 24] Too many open files  [Witherspoon]

gkeishin, 2017-03-25:
`100 boot test.  Seen Poweron breaks at 51th boot. On debugging we see that the WSGIServer seems to have given up.. `

`Driver Info`
- [x] Build  :  https://openpower.xyz/job/openbmc-build/655/
- [x] Driver : v1.99.3-137-g1ddd805
- [x] Linux Version : Linux witherspoon 4.7.10-382872843bb8e7755fddeaaa284a00c640bfc732 #1 Sat Mar 25 03:21:44 UTC 2017 armv6l GNU/Linux
- [x] PNOR VERSION:
IBM-witherspoon-ibm-OP9_v1.15_1.3
        op-build-v1.15-15-gce29449
        buildroot-2017.02-4-gc28573e
        skiboot-28e96ff
        hostboot-86bae1c
        linux-4.10.1-p976d063
        petitboot-v1.4.1-p3ec3d6e
        witherspoon-xml-065aaa2
        occ-4141b5f
        hostboot-binaries-4b44090
        capp-ucode-1bb7503
```
`Boot Test`

```
16:48:08.806 	INFO 	Repeating keyword, round 49/100. 	
16:51:29.516 	INFO 	Repeating keyword, round 50/100. 	
16:54:23.425 	INFO 	Repeating keyword, round 51/100.  <-- Failed on 51 st
```

`This the Host is booted`

```
`root@witherspoon:~# uptime`
 09:16:18 up  3:11,  load average: 1.09, 1.08, 1.27
root@witherspoon:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.On
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Running

Executing: GET  /org/openbmc/sensors/host/BootProgress

/org/openbmc/sensors/host/BootProgress
{
    "error": 0,
    "units": "",
    "value": "FW Progress, Starting OS"
}

```

- [x] Journal log 
[journald_v1.99.3-137-g1ddd805_IPL.txt](https://github.com/openbmc/openbmc/files/869987/journald_v1.99.3-137-g1ddd805_IPL.txt)

```

FAIL 	ConnectionError: ('Connection aborted.', error(104, 'Connection reset by peer')): FAIL != PASS

`WSGIServer log`
```
Feb 22 08:48:11 witherspoon phosphor-gevent[1176]: Traceback (most recent call last):
Feb 22 08:48:11 witherspoon phosphor-gevent[1176]:   File "/usr/lib/python2.7/site-packages/gevent/baseserver.py", line 140, in _do_read
Feb 22 08:48:11 witherspoon phosphor-gevent[1176]:   File "/usr/lib/python2.7/site-packages/gevent/server.py", line 93, in do_read
Feb 22 08:48:11 witherspoon phosphor-gevent[1176]:     client_socket, address = self.socket.accept()
Feb 22 08:48:11 witherspoon phosphor-gevent[1176]: error: [Errno 24] Too many open files
Feb 22 08:48:11 witherspoon phosphor-gevent[1176]: <WSGIServer at 0x7655a570 fileno=5 address=('::80ac:6447:445c:bd76%1985390064', 443, 0, 1985390064)> failed with error
Feb 22 08:48:11 witherspoon phosphor-gevent[1176]: Traceback (most recent call last):
Feb 22 08:48:11 witherspoon phosphor-gevent[1176]:   File "/usr/lib/python2.7/site-packages/gevent/greenlet.py", line 327, in run
Feb 22 08:48:11 witherspoon phosphor-gevent[1176]:     result = self._run(*self.args, **self.kwargs)
Feb 22 08:48:11 witherspoon phosphor-gevent[1176]:   File "/usr/lib/python2.7/site-packages/gevent/server.py", line 102, in wrap_socket_and_handle
Feb 22 08:48:11 witherspoon phosphor-gevent[1176]:     ssl_socket = self.wrap_socket(client_socket, **self.ssl_args)
Feb 22 08:48:11 witherspoon phosphor-gevent[1176]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 404, in wrap_socket
Feb 22 08:48:11 witherspoon phosphor-gevent[1176]:     ciphers=ciphers)
Feb 22 08:48:11 witherspoon phosphor-gevent[1176]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 106, in __init__
Feb 22 08:48:11 witherspoon phosphor-gevent[1176]:     cert_reqs, ssl_version, ca_certs)
Feb 22 08:48:11 witherspoon phosphor-gevent[1176]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 31, in new_sslwrap
Feb 22 08:48:11 witherspoon phosphor-gevent[1176]:     context.load_cert_chain(certfile, keyfile)
Feb 22 08:48:11 witherspoon phosphor-gevent[1176]: IOError: [Errno 24] Too many open files
Feb 22 08:48:11 witherspoon phosphor-gevent[1176]: <Greenlet at 0x74e71c60: <bound method WSGIServer.wrap_socket_and_handle of <WSGIServer at 0x7655a570 fileno=5 address=('::80ac:6447:445c:bd76%1985390064', 443, 0, 1985390064)>>(<socket at 0x74e33c30 fileno=1023>, ('::e0e3:bb76:8e91:f402%1197707688', 53372, 0, 119)> failed with IOError
```

```
root@witherspoon:~# cat /proc/sys/net/ipv4/tcp_max_orphans
2048
root@witherspoon:~#

Feb 22 09:00:13 witherspoon kernel[777]: TCP: too many orphaned sockets
Feb 22 09:00:17 witherspoon kernel: TCP: too many orphaned sockets
Feb 22 09:00:17 witherspoon kernel: TCP: too many orphaned sockets
Feb 22 09:00:17 witherspoon kernel: TCP: too many orphaned sockets
Feb 22 09:00:17 witherspoon kernel[777]: TCP: too many orphaned sockets
Feb 22 09:00:17 witherspoon kernel[777]: TCP: too many orphaned sockets
Feb 22 09:00:17 witherspoon kernel[777]: TCP: too many orphaned sockets
Feb 22 09:00:20 witherspoon kernel: TCP: too many orphaned sockets
Feb 22 09:00:20 witherspoon kernel[777]: TCP: too many orphaned sockets
Feb 22 09:00:21 witherspoon kernel: TCP: too many orphaned sockets
Feb 22 09:00:21 witherspoon kernel[777]: TCP: too many orphaned sockets
Feb 22 09:00:21 witherspoon kernel: TCP: too many orphaned sockets
Feb 22 09:00:21 witherspoon kernel[777]: TCP: too many orphaned socket
```

gkeishin, 2017-03-25:
FYI: Test will also need cleanup to remove active sessions per power on .. https://gerrit.openbmc-project.xyz/#/c/3270/

gkeishin, 2017-03-27:
This seems to be same as https://github.com/openbmc/openbmc/issues/875

gkeishin, 2017-05-11:
In one of the CT run
```
- [x] BUILD_ID   : https://openpower.xyz/job/openbmc-build/736/
- [x] VERSION_ID : v1.99.5-26-geb2af39
- [x] Kernel     : Linux witherspoon 4.7.10-15c30d4deb0a01f398230395dc0d9af4be92d856 #1 Thu May 11 04:21:21 UTC 2017 armv6l GNU/Linux
- [x] PNOR VERSION:
IBM-witherspoon-ibm-OP9_v1.16_1.3
        op-build-v1.16-35-g2bb0a83-dirty
        buildroot-2017.02-4-gc28573e
        skiboot-5.5.0
        hostboot-c9bd42f
        linux-4.10.7-openpower1-p61306aa
        petitboot-v1.4.2-p58d71a5
        witherspoon-xml-3a7d05c
        occ-e1889e5
        hostboot-binaries-8cbbc67
        capp-ucode-1bb7503
        sbe-9b98de3-p554c410
```

```
Apr 20 17:32:08 witherspoon kernel: TCP: too many orphaned sockets
Apr 20 17:32:08 witherspoon kernel: TCP: too many orphaned sockets
Apr 20 17:32:08 witherspoon kernel: TCP: too many orphaned sockets
Apr 20 17:32:08 witherspoon kernel[767]: TCP: too many orphaned sockets
Apr 20 17:32:08 witherspoon kernel[767]: TCP: too many orphaned sockets
Apr 20 17:32:08 witherspoon kernel[767]: TCP: too many orphaned sockets
```
[journald_v1.99.5-26-geb2af39_orphaned_sockets.txt](https://github.com/openbmc/openbmc/files/992354/journald_v1.99.5-26-geb2af39_orphaned_sockets.txt)

sivassrr, 2017-05-31:
Observed it in Driver 771 
 	ConnectionError: ('Connection aborted.', error(104, 'Connection reset by peer')): FAIL != PASS
 	
root@witherspoon:~# cat /proc/sys/net/ipv4/tcp_max_orphans
2048
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.6-21"
VERSION_ID="v1.99.6-21-g3de8405"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.6-21"
BUILD_ID="v1.99.6"
root@witherspoon:~# cat /etc/timestamp
20170531023047
root@witherspoon:~#

rahulmah, 2017-07-14:
@gkeishin : As discussed, I am seeing this problem again with today's todays driver 883/v1.99.8-8-g87342b9. 

```
Jun 29 04:43:40 witherspoon phosphor-gevent[905]: error: [Errno 24] Too many open files
Jun 29 04:43:40 witherspoon phosphor-gevent[905]: <WSGIServer at 0x765a58b0 fileno=5 address=('::2cc2:f848:b08d:b176%1991347616', 443, 0, 1991347616)> failed with error
Jun 29 04:43:40 witherspoon phosphor-gevent[905]: Traceback (most recent call last):
Jun 29 04:43:40 witherspoon phosphor-gevent[905]:   File "/usr/lib/python2.7/site-packages/gevent/greenlet.py", line 327, in run
Jun 29 04:43:40 witherspoon phosphor-gevent[905]:     result = self._run(*self.args, **self.kwargs)
Jun 29 04:43:38 witherspoon dropbear[12913]: Exit (root): Error reading: Connection reset by peer
Jun 29 04:43:41 witherspoon phosphor-gevent[905]:   File "/usr/lib/python2.7/site-packages/gevent/server.py", line 102, in wrap_socket_and_handle
Jun 29 04:43:41 witherspoon phosphor-gevent[905]:     ssl_socket = self.wrap_socket(client_socket, **self.ssl_args)
Jun 29 04:43:41 witherspoon phosphor-gevent[905]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 404, in wrap_socket
Jun 29 04:43:41 witherspoon phosphor-gevent[905]:     ciphers=ciphers)
Jun 29 04:43:41 witherspoon phosphor-gevent[905]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 106, in __init__
Jun 29 04:43:41 witherspoon phosphor-gevent[905]:     cert_reqs, ssl_version, ca_certs)
Jun 29 04:43:41 witherspoon phosphor-gevent[905]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 31, in new_sslwrap
Jun 29 04:43:41 witherspoon phosphor-gevent[905]:     context.load_cert_chain(certfile, keyfile)
Jun 29 04:43:41 witherspoon phosphor-gevent[905]: IOError: [Errno 24] Too many open files
Jun 29 04:43:41 witherspoon phosphor-gevent[905]: <Greenlet at 0x7591d1c0: <bound method WSGIServer.wrap_socket_and_handle of <WSGIServer at 0x765a58b0 fileno=5 address=('::2cc2:f848:b08d:b176%1991347616', 443, 0, 1991347616)>>(<socket at 0x759249b0 fileno=1023>, ('::6099:af76:8e91:f402%1224190740', 32964, 0, 122)> failed with IOError
```
Due to this REST login is also failing.
```
bash-4.1$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -X POST https://<BMC_IP>/login -d "{\"data\": [ \"root\", \"0penBmc\" ] }"
curl: (35) SSL connect error
bash-4.1$
```

geissonator, 2017-07-14:
Stephanie just opened a defect on this, hit on one of her systems as well.  Please do a "ls /proc/*/fd/", look for any suspicious responses (a PID with a lot of fd's), and then check which process has that PID.

gkeishin, 2017-07-18:
- [x] proc fd list 
[proc_fd_list_v1.99.8-8-g87342b9.txt](https://github.com/openbmc/openbmc/files/1155575/proc_fd_list_v1.99.8-8-g87342b9.txt)

We have a couple of system in failed state as well.

geissonator, 2017-07-19:
The process with the fd's was the REST server, so either a leak within it to the sdbus or a leak between the client and the server.

gkeishin, 2017-07-19:
I 'll fix this one up from test code.. from the test looks like we leak a lots of sessions open.. post the fix if we see it again we will open a new one..

gkeishin, 2017-07-19:
https://gerrit.openbmc-project.xyz/#/c/5386/

closed 2017-07-19 by williamspatrick