openbmc/openbmc #1390: IPMI Setting Boot Device to Default(No override) fail [ Witherspoon ]

gkeishin, 2017-03-27:
`Driver info`
```
- [x] BUILD_ID   : https://openpower.xyz/job/openbmc-build/655/
- [x] VERSION_ID : v1.99.3-137-g1ddd805
- [x] Kernel     : Linux witherspoon 4.7.10-382872843bb8e7755fddeaaa284a00c640bfc732 #1 Sat Mar 25 03:21:44 UTC 2017 armv6l GNU/Linux
```

```
$ ipmitool -I lanplus -C 3 -P 0penBmc -H 9.3.185.224 raw 0x0 0x8 0x05 0x80 0x00 0x00 0x00 0x00
Unable to send RAW command (channel=0x0 netfn=0x0 lun=0x0 cmd=0x8 rsp=0xff): Unspecified error
```
```
Feb 22 07:20:41 witherspoon netipmid[902]: Failed to issue method call: Traceback (most recent call last):
Feb 22 07:20:41 witherspoon netipmid[902]:   File "/usr/lib/python2.7/site-packages/dbus/service.py", line 707, in _message_cb
Feb 22 07:20:41 witherspoon netipmid[902]:     retval = candidate_method(self, *args, **keywords)
Feb 22 07:20:41 witherspoon netipmid[902]:   File "/usr/lib/python2.7/site-packages/obmc/dbuslib/bindings.py", line 82, in Set
Feb 22 07:20:41 witherspoon netipmid[902]:     self.validator(interface_name, property_name, new_value)
Feb 22 07:20:41 witherspoon netipmid[902]:   File "/usr/sbin/settings_manager.py", line 236, in input_validator
Feb 22 07:20:41 witherspoon netipmid[902]:     self.validate_list(shk['allowed'], value)
Feb 22 07:20:41 witherspoon netipmid[902]:   File "/usr/sbin/settings_manager.py", line 176, in validate_list
Feb 22 07:20:41 witherspoon netipmid[902]:     raise ValueError("Invalid input. Data not in allowed values")
Feb 22 07:20:41 witherspoon netipmid[902]: ValueError: Invalid input. Data not in allowed values
Feb 22 07:20:41 witherspoon netipmid[902]: Dbus set property(boot_flags) failed for set_sys_boot_options.
```

The same worked for other boot_flag settings but this setting to "Default" seems to be problem.

gkeishin, 2017-03-27:
- [x] Journal log 
[journald_v1.99.3-137-g1ddd805_IPMITOOL.txt](https://github.com/openbmc/openbmc/files/871569/journald_v1.99.3-137-g1ddd805_IPMITOOL.txt)

ratagupt, 2017-04-04:
https://gerrit.openbmc-project.xyz/#/c/3478/
This fix Needs a code update with clear persistent files option.

closed 2017-04-04 by williamspatrick