openbmc/openbmc #1393: Barreleye going into quiesced state after HOST power on

rahulmah, 2017-03-28:
Barreleye system going into quiesced state after powering on the host. This issue is seen with recent OpenBMC fw.

Step to recreate:
1. Power on the host from off state.
```
bash-4.1$ curl -c cjar -b cjar -k https://xx.xx.xx.xx/xyz/openbmc_project/state/host0
{
  "data": {
    "CurrentHostState": "xyz.openbmc_project.State.Host.HostState.Off",
    "RequestedHostTransition": "xyz.openbmc_project.State.Host.Transition.Off"
  },
  "message": "200 OK",
  "status": "ok"
}bash-4.1$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -d "{\"data\": \"xyz.openbmc_project.State.Host.Transition.On\"}" -X PUT  https://9.3.2332/xyz/openbmc_project/state/host0/attr/RequestedHostTransition
{
  "data": null,
  "message": "200 OK",
  "status": "ok"
}
```
2. After some time, we will see that HOST will do a reboot an move into Quiesced state.

```
bash-4.1$ curl -c cjar -b cjar -khttps://xx.xx.xx.xx/xyz/openbmc_project/state/host0                                                                            {
  "data": {
    "CurrentHostState": "xyz.openbmc_project.State.Host.HostState.Off",
    "RequestedHostTransition": "xyz.openbmc_project.State.Host.Transition.On"
  },
  "message": "200 OK",
  "status": "ok"
}bash-4.1$ curl -c cjar -b cjar -k https://xx.xx.xx.xx/xyz/openbmc_project/state/host0
{
  "data": {
    "CurrentHostState": "xyz.openbmc_project.State.Host.HostState.Running",
    "RequestedHostTransition": "xyz.openbmc_project.State.Host.Transition.On"
  },
  "message": "200 OK",
  "status": "ok"
}bash-4.1$ curl -c cjar -b cjar -k https://xx.xx.xx.xx/xyz/openbmc_project/state/host0
{
  "data": {
    "CurrentHostState": "xyz.openbmc_project.State.Host.HostState.Running",
    "RequestedHostTransition": "xyz.openbmc_project.State.Host.Transition.Reboot"
  },
  "message": "200 OK",
  "status": "ok"
}bash-4.1$ curl -c cjar -b cjar -k https://xx.xx.xx.xx/xyz/openbmc_project/state/host0
{
  "data": {
    "CurrentHostState": "xyz.openbmc_project.State.Host.HostState.Quiesced",
    "RequestedHostTransition": "xyz.openbmc_project.State.Host.Transition.Reboot"
  },
  "message": "200 OK",
  "status": "ok"
}
```
Journal logs from failure point.
```
root@barreleye:~# journalctl -f
-- Logs begin at Wed 2017-02-01 02:35:05 UTC. --
Feb 01 02:43:41 barreleye dropbear[1059]: Child connection from ::ffff:9.202.30.165:57510
Feb 01 02:43:44 barreleye host_watchdog.exe[709]: Polling watchdog: 0
Feb 01 02:43:44 barreleye chassis_control.py[683]: Watchdog Error, Going to quiesce
Feb 01 02:43:44 barreleye systemd[1]: Reached target Quiesce Target.
Feb 01 02:43:44 barreleye phosphor-host-state-manager[845]: Auto reboot disabled. Maintaining quiesce.
Feb 01 02:43:44 barreleye phosphor-host-state-manager[845]: Change to Host State
Feb 01 02:43:47 barreleye dropbear[1059]: pam_unix(dropbear:account): account root has password changed in future
Feb 01 02:43:47 barreleye dropbear[1059]: PAM password auth succeeded for 'root' from ::ffff:9.202.30.165:57510
Feb 01 02:43:47 barreleye dropbear[1060]: lastlog_perform_login: Couldn't stat /var/log/lastlog: No such file or directory
Feb 01 02:43:47 barreleye dropbear[1060]: lastlog_openseek: /var/log/lastlog is not a file or directory!
Feb 01 02:44:14 barreleye phosphor-hwmon-readd[1034]: Remote I/O error
Feb 01 02:44:14 barreleye systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x401c0-nct7904\x402e.service: Main process exited, code=exited, status=1/FAILURE
Feb 01 02:44:14 barreleye systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x401c0-nct7904\x402e.service: Unit entered failed state.
Feb 01 02:44:14 barreleye systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x401c0-nct7904\x402e.service: Failed with result 'exit-code'.
Feb 01 02:44:14 barreleye systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x401c0-nct7904\x402e.service: Service hold-off time over, scheduling restart.
Feb 01 02:44:14 barreleye systemd[1]: Stopped Phosphor Hwmon Poller.
Feb 01 02:44:14 barreleye systemd[1]: Started Phosphor Hwmon Poller.
Feb 01 02:44:18 barreleye phosphor-gevent[872]: ::c073:b9b6:8e91:f402%1185712456 - - [2017-02-01 02:44:18] "GET /xyz/openbmc_project/state/host0 HTTP/1.1" 200 331 1.151997
Feb 01 02:44:48 barreleye phosphor-gevent[872]: ::c073:b9b6:8e91:f402%1185712456 - - [2017-02-01 02:44:48] "GET /org/openbmc/settings/host0 HTTP/1.1" 200 538 0.150553
```
  
BMC FW:
```  
root@barreleye:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.3-147"
VERSION_ID="v1.99.3-147-g49fe17d"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.3-147"
BUILD_ID="v1.99.3"

root@barreleye:~# cat /etc/timestamp
20170327212853
root@barreleye:~#
```

FFDC : 
[20170328035303945134_BMC_journalctl.txt](https://github.com/openbmc/openbmc/files/875081/20170328035303945134_BMC_journalctl.txt)

swe12345, 2017-03-28:
Observing the same issue in Palmetto after HOST power on.
Driver Details:
```
root@palmetto:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.3-147"
VERSION_ID="v1.99.3-147-g49fe17d"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.3-147"
BUILD_ID="v1.99.3"

```

```
root@palmetto:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.On
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Quiesced
```

williamspatrick, 2017-03-28:
The journal output indicates the watchdog fired. Do you have the Host console output?

geissonator, 2017-03-28:
Did you reflash PNOR to ensure it was in a good state?  In my recent testing on barreleye, I'm still seeing 2 things, either hostboot is garding out all the cores and prevents a boot at istep 6.3 or the PNOR is corrupted to the point where it wont boot at all.  A reflash of PNOR fixes both issues.

rahulmah, 2017-03-30:
System was having latest PNOR(#335) so I did not reflash.  Today we found that we have a new PNOR build 339. We flash this new PNOR on Barreleye and not seeing the issue. We will try to test with this new PNOR image on palmetto. If no issue is seen then will close down this issue.

closed 2017-03-30 by geissonator