openbmc/openbmc #1394: Factory reset through dbus IPMI not working

prkatti1, 2017-03-28:
Trying to do factory reset using following dbus command, it is not resetting the system.
Run Dbus IPMI Raw Command  0x32 0xBA 00 00
Run Dbus IPMI Raw Command  0x32 0x66

williamspatrick, 2017-03-28:
Is this an openpower ipmi command or something in ipmi 2.0?  If this is openpower specific we do not plan to support.

prkatti1, 2017-03-28:
@williamspatrick Thanks for the confirmation, could you Pls provide the command which works for openbmc. I tried this also
Run Dbus IPMI Raw Command  0x3c 0x40

williamspatrick, 2017-03-28:
I don't know what these ipmi commands are. We do not currently have any support for a factory reset. That will come after the code update work being done by @anoo.

williamspatrick, 2017-03-28:
These netfun are in the IPMI OEM range. There does not appear to be anything in IPMI 2.0 for factory reset, so we have no plans to support over IPMI. There will be a REST API when function is complete.

closed 2017-03-28 by williamspatrick