openbmc/openbmc #1399: witherspoon: power button LED should blink at standby

cdeadmin, 2017-03-28:


williamspatrick, 2017-03-28:
I believe this requires a new feature in the LED group manager to allow, in some cases, 'on' to take precedence over 'blink'.

vishwabmc, 2017-03-30:
I am hoping this could all be managed by putting appropriate groups into certain systemd targets instead of putting a system specific thing onto the code ?

vishwabmc, 2017-04-24:
This is my current thought.
```
* Have BmcStandby group that is specific for Witherspoon and assert that group ( update bbappend ) on way to bmc_ready 

* When the Power is applied, same LED gets to Solid-On as part of `PowerOn` group

* As part of power-off, `PowerOn` group gets de-asserted and hence it results in LEDs to behave as part of `BmcStandby`
```

williamspatrick, 2017-04-25:
@vishwabmc - As discussed earlier today, the big gap I see here is that this blink vs on is opposite our current policy.  (Blink always takes priority).  Need to understand how you are going to solve that.  Otherwise, what you described is effectively what is already implemented.

vishwabmc, 2017-06-28:
Oh.. Please help close this one. I had put the `Fixes` on a open-power pull request that I ended up cancelling and hence this is open.

@williamspatrick Please ?.

closed 2017-06-30 by williamspatrick