openbmc/openbmc #1403: Stop boot at istep

GHF, 2017-03-28:
One of the hardware testing procedures we want to perform on our P9 modules (see slides "P9 IOTK Testing") requires that we stop at an istep: "IPL by step to fabric_io_dccal"

We can do this on Cronus by hacking at the BMC boot sequence to not write the SBE start bit (0x2801 IBM bit 0?) and let Cronus drive isteps.

Is there a better way to do this? Can we switch to debug mode *and* not start SBE without having to hack at `openpower-proc-control`? Or better yet, can the BMC drive isteps up to a certain istep somehow?

spinler, 2017-03-28:
The architected way to do this to start the `obmc-power-chassis-on@0.target` target to power on the chassis and then let cronus take over from there.  May not work on zaius yet since vcs-on is still in the wrong place.

williamspatrick, 2017-03-29:
The IBM team is not going to be able to contain an "ipl-by-steps" implementation on the BMC at this time.  
I think doing it right would need to have much more intricate cooperation between which istep is executing and some of the monitoring applications that get started.  Would require some thought into all the pieces that would need to be changed.  (There is probably a shortcut to put it into a mode where auto-recovery does not happen.)

Matt gave information here and sent to the mailing list how to stop the boot after power supplies have been enabled and then switch to Cronus driven boot.  There are a few issues open that Matt mentioned in the mailing list (and mentioned one here too).  The vcs move is the most important one for Zaius because the VCS workarounds are not currently done right when just powering on the chassis.

GHF, 2017-03-30:
Thanks Matt, Patrick for the information. I should add that in addition to the vcs-on for Zaius, we also need to switch the FSI clocks to synchronous mode before using Cronus, which is already covered in #1397. We can do this currently by uploading pdbg from the debug tarball and poking the right CFAM.

gtmills, 2018-05-31:
@GHF No update in a year. Can this be closed or still a problem?

closed 2018-06-01 by GHF

hanetzer, 2019-08-31:
For those of you from the future like me `obmc-power-chassis-on` has become
`obmc-chassis-poweron`.