openbmc/openbmc #1404: witherspoon: need VDDR lowered to 1.2V

austinwrx, 2017-03-29:
Currently we see that VDDR voltage is coming up at 1.35V on witherspoon systems. This is before starting an IPL. Need it lowered to 1.2V

/usr/bin/vrm-control.sh vddra=1.2 vddrb=1.2

geissonator, 2017-03-29:
https://github.com/openbmc/openbmc/issues/1285 was where it was requested we removed the 1.35 values.  I guess they're the defaults in the power on logic anyway?

geissonator, 2017-03-29:
https://gerrit.openbmc-project.xyz/#/c/3378/
Uma is going to help verify it.

austinwrx, 2017-03-29:
It looks like once the new BMC code is on there, VRMs might still come up with 1.35 until the next power cycle. Please check with Paul Robertson.

paulgrobertson, 2017-03-29:
Yes. VRM will remain programmed to  1.35V until you a/c power cycle, or manually override it.   Looks like the vddra/b VRM default is 1.211V.

geissonator, 2017-03-31:
ok, closing this issue then based on above info.  The previous overrides stay on the system until you power cycle it.

closed 2017-03-31 by geissonator