openbmc/openbmc #1407: Deprecate support for RHEL6.

williamspatrick, 2017-03-29:
Yocto 2.2 deprecated support for CentOS6, which is pretty similar to RHEL6.  RHEL6 is still used heavily inside IBM so we are putting a few patches into the tree to accommodate this usage.  IBM needs to come up with a better solution for their developers so these patches can be removed.  Looking forward, doing development of a modern Linux distribution on top of a distribution that is now 6.5 years old is not workable.

@geissonator - I'm assigning this to you as the mediator to the build / test organization.  A move to RHEL7 might work in the short term but we can expect some near-future Yocto 2.x to also deprecate that.

I'm going to reference this issue with a number of TODO entries in the tree to remove workarounds in the future.

gtmills, 2018-05-23:
Looks like a few places in the code that need to be removed. 
https://github.com/search?q=org%3Aopenbmc+1407&type=Code
We no longer support RHEL6

gtmills, 2018-06-13:
https://github.com/openbmc/openbmc/commit/afc77225f4881a26fe3705251da3c156a6d21a34#diff-353a70eee1aa1e3a99b4419bb001ab4e 
https://github.com/openbmc/openbmc/commit/50b1e0882813f56beaab7c90f2ce00d8fa3f1c83#diff-0aeba766c5b8f3cfb0588e3d8e2d568e
https://github.com/openbmc/openbmc/commit/dfe681a7bf34ddc64f2723665ea676c6f81f3985#diff-0aeba766c5b8f3cfb0588e3d8e2d568e
Did this. Closing this issue.

closed 2018-06-13 by gtmills