openbmc/openbmc #1408: softPowerOff isn't working after IPL test

johnhcwang, 2017-03-30:
We ran the chassis power on and soft power off cycle overnight and then got the failure that we cannot soft power off the system anymore.
I tried to soft power off by busctl and got the error as below.
```
root@zaius:~# busctl call org.openbmc.control.Chassis /org/openbmc/control/chassis0 org.openbmc.control.Chassis softPowerOff
The name org.openbmc.HostServices was not provided by any .service files
```
Confirmed the HostServices is still running.
```
root@zaius:~# systemctl status --no-pager org.openbmc.HostServices -l
● org.openbmc.HostServices.service - Phosphor Inband IPMI
   Loaded: loaded (/lib/systemd/system/org.openbmc.HostServices.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2017-03-24 06:31:56 UTC; 9h ago
 Main PID: 13275 (ipmid)
   CGroup: /system.slice/org.openbmc.HostServices.service
           └─13275 /usr/sbin/ipmid

Mar 24 15:37:28 zaius ipmid[13275]: _append_to_dict: VPD Key = [Custom Field 3] : Type Code=[ASCII+Latin] : Len = [17] : Val = [        hostboot-a0437b2]
Mar 24 15:37:28 zaius ipmid[13275]: _append_to_dict: VPD Key = [Custom Field 4] : Type Code=[ASCII+Latin] : Len = [22] : Val = [        linux-4.10.1-p5af9063]
Mar 24 15:37:28 zaius ipmid[13275]: _append_to_dict: VPD Key = [Custom Field 5] : Type Code=[ASCII+Latin] : Len = [26] : Val = [        petitboot-v1.4.1-p35b5615]
Mar 24 15:37:28 zaius ipmid[13275]: _append_to_dict: VPD Key = [Custom Field 6] : Type Code=[ASCII+Latin] : Len = [18] : Val = [        zaius-xml-abfe0a3]
Mar 24 15:37:28 zaius ipmid[13275]: _append_to_dict: VPD Key = [Custom Field 7] : Type Code=[ASCII+Latin] : Len = [12] : Val = [        occ-4141b5f]
Mar 24 15:37:28 zaius ipmid[13275]: _append_to_dict: VPD Key = [Custom Field 8] : Type Code=[ASCII+Latin] : Len = [26] : Val = [        hostboot-binaries-4b44090]
Mar 24 15:37:28 zaius ipmid[13275]: ERROR Unable to get the fru info for FRU=0
Mar 24 15:37:28 zaius ipmid[13275]: Error updating inventory
Mar 24 15:37:30 zaius ipmid[13275]: Received Host Event
Mar 24 15:37:30 zaius ipmid[13275]: A host system event was received
```
The only failure point from journal log.
```
Division by zero in kernel.
CPU: 0 PID: 850 Comm: python Not tainted 4.7.10-76120549ed8832760de2bacfd42cb0690964b31f #1
Hardware name: ASpeed SoC
[<801072f8>] (unwind_backtrace) from [<80105558>] (show_stack+0x10/0x14)
[<80105558>] (show_stack) from [<8029f778>] (Ldiv0+0x8/0x10)
[<8029f778>] (Ldiv0) from [<803378d8>] (show_fan+0x48/0x68)
[<803378d8>] (show_fan) from [<802e4428>] (dev_attr_show+0x20/0x44)
[<802e4428>] (dev_attr_show) from [<801f823c>] (sysfs_kf_seq_show+0x84/0xe8)
[<801f823c>] (sysfs_kf_seq_show) from [<801c0640>] (seq_read+0x1bc/0x42c)
[<801c0640>] (seq_read) from [<801a30ac>] (__vfs_read+0x28/0x128)
[<801a30ac>] (__vfs_read) from [<801a323c>] (vfs_read+0x90/0xfc)
[<801a323c>] (vfs_read) from [<801a3628>] (SyS_read+0x54/0xac)
[<801a3628>] (SyS_read) from [<801021c0>] (ret_fast_syscall+0x0/0x3c)
```
The test version is based on openbmc#217b5ac and linux#6786cab.
```
root@zaius:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.3-156"
VERSION_ID="v1.99.3-156-g900493f-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.3-156"
BUILD_ID="v1.99.3"
```

williamspatrick, 2017-03-30:
I do not believe this is the right method to do a soft power off anymore. @vishwabmc, @geissonator?

williamspatrick, 2017-03-30:
@GHF - looks like there might also be an issue with Zaius' sensor number for the host firmware version. This is unrelated to the original problem. 

@GHF, @shenki - I don't know what driver is responsible for this kernel backtrace but there also appears to be a driver bug.

geissonator, 2017-03-30:
This will be changing but for now it's the way to do it still.  I wonder if you hit #875?

mdmillerii, 2017-03-30:
@shenki the log above has a divide by 0 in show_fan (attribute read).  Is there a fix available?

GHF, 2017-03-30:
I don't think we have any drivers enabled that have a `show_fan` function to handle a sysfs attribute. @johnhcwang Are you using any drivers that isn't normally enabled in OpenBMC?

johnhcwang, 2017-03-30:
There's nct7904 and lm75 enabled in this test firmware.

GHF, 2017-03-30:
I guess the kernel error is a buggy `nct7904` driver.

https://github.com/openbmc/linux/blob/1252be9ce0ab4f622b8692b648894d09c0df71ce/drivers/hwmon/nct7904.c#L172

Like @williamspatrick mentioned, the kernel backtrace is probably not related to your soft power off bug.

@williamspatrick I'll open an issue for the FRU look up error.

johnhcwang, 2017-04-05:
Hi @geissonator , I ran the soft power cycle test again and got the error messages.
```
Mar 26 19:22:14 zaius phosphor-gevent[938]: Traceback (most recent call last):
Mar 26 19:22:14 zaius phosphor-gevent[938]:   File "/usr/lib/python2.7/site-packages/bottle.py", line 862, in _handle
Mar 26 19:22:14 zaius phosphor-gevent[938]:     return route.call(**args)
Mar 26 19:22:14 zaius phosphor-gevent[938]:   File "/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py", line 550, in __call__
Mar 26 19:22:14 zaius phosphor-gevent[938]:     return self.callback(*a, **kw)
Mar 26 19:22:14 zaius phosphor-gevent[938]:   File "/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py", line 607, in wrap
Mar 26 19:22:14 zaius phosphor-gevent[938]:     return callback(*a, **kw)
Mar 26 19:22:14 zaius phosphor-gevent[938]:   File "/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py", line 748, in wrap
Mar 26 19:22:14 zaius phosphor-gevent[938]:     return self.to_jsonp(callback(*a, **kw))
Mar 26 19:22:14 zaius phosphor-gevent[938]:   File "/usr/lib/python2.7/site-packages/bottle.py", line 1734, in wrapper
Mar 26 19:22:14 zaius phosphor-gevent[938]:     rv = callback(*a, **ka)
Mar 26 19:22:14 zaius phosphor-gevent[938]:   File "/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py", line 713, in wrap
Mar 26 19:22:14 zaius phosphor-gevent[938]:     data = callback(*a, **kw)
Mar 26 19:22:14 zaius phosphor-gevent[938]:   File "/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py", line 653, in wrap
Mar 26 19:22:14 zaius phosphor-gevent[938]:     return callback(*a, **kw)
Mar 26 19:22:14 zaius phosphor-gevent[938]:   File "/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py", line 678, in wrap
Mar 26 19:22:14 zaius phosphor-gevent[938]:     return callback(*a, **kw)
Mar 26 19:22:14 zaius phosphor-gevent[938]:   File "/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py", line 101, in __call__
Mar 26 19:22:14 zaius phosphor-gevent[938]:     return getattr(self, 'do_' + request.method.lower())(**kw)
Mar 26 19:22:14 zaius phosphor-gevent[938]:   File "/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py", line 237, in do_post
Mar 26 19:22:14 zaius phosphor-gevent[938]:     return request.route_data['method']()
Mar 26 19:22:14 zaius phosphor-gevent[938]:   File "/usr/lib/python2.7/site-packages/dbus/proxies.py", line 145, in __call__
Mar 26 19:22:14 zaius phosphor-gevent[938]:     **keywords)
Mar 26 19:22:14 zaius phosphor-gevent[938]:   File "/usr/lib/python2.7/site-packages/dbus/connection.py", line 651, in call_blocking
Mar 26 19:22:14 zaius phosphor-gevent[938]:     message, timeout)
Mar 26 19:22:14 zaius phosphor-gevent[938]: DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name org.openbmc.HostServices was not provided by any .service files
```
Am I hit #875 that you mentioned because the error seems different?

williamspatrick, 2017-04-05:
No, this is different. How are you attempting to initiate a soft power off?

johnhcwang, 2017-04-05:
Run the script over the weekend.
```
:LOOP
echo Start DC Power Cycle Stress
curl -c cjar -b cjar -k -H "Content-Type: application/json" -X POST -d "{\"data\": []}"  https://192.168.1.122/org/openbmc/control/chassis0/action/powerOn
TIMEOUT /T 420 /NOBREAK
curl -c cjar -b cjar -k -H "Content-Type: application/json" -X POST -d "{\"data\": []}"  https://192.168.1.122/org/openbmc/control/chassis0/action/softPowerOff
TIMEOUT /T 60 /NOBREAK
GOTO LOOP
```

vishwabmc, 2017-04-05:
curl -c cjar -b cjar -k -H "Content-Type: application/json" -X POST -d "{\"data\": []}"  https://192.168.1.122/org/openbmc/control/chassis0/action/softPowerOff

>> That should not be used. I should have removed `SoftPowerOff` from chassis control. 

Please move over to using `xyz.openbmc_project.Ipmi.Internal.SoftPowerOff.service`. 

Please note that SoftPowerOff is IPMI internal only and hence the real way would be to use xyz.openbmc_project.State.Host and currently State.Host does not have it and there is a work under progress to call this soft off from State.

That also reminds me of supporting command : 5 ( soft power off ) in ipmi_chassis. Can you use ipmitool or you need the REST way only ?

vishwabmc, 2017-04-05:
@geissonator What is the plan on incorporating soft off into State.Host.PowerOff ?

williamspatrick, 2017-04-05:
Soft power off is supported now by using the "host off" operation and hard power off is supported by using the "chassis off" operation.

williamspatrick, 2017-04-05:
@johnhcwang - as @vishwabmc said, the REST API for power on / power off has changed. 

Please read through this and the corresponding dbus definition yamls. https://github.com/openbmc/phosphor-dbus-interfaces/blob/master/xyz/openbmc_project/State/README.md

closed 2017-04-05 by williamspatrick

rlippert, 2017-04-05:
FYI I think I've seen this particular error before when using obmcutil poweroff/poweron:

DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name org.openbmc.HostServices was not provided by any .service files

when the ObjectMapper died (issue #893).