openbmc/openbmc #1410: Sensor Service fails to start and doesn't boot to standby [ Witherspoon ]

gkeishin, 2017-03-30:
```
- [x] BUILD_ID   : https://openpower.xyz/job/openbmc-build/669/ 
                 : v1.99.3-176-gb54c3b2                      
- [x] Kernel     : Linux witherspoon 4.7.10-6786cab8778ba96d84d89c7aec34374b5f4f1968 #1 Thu Mar 30 03:20:35 UTC 2017 armv6l GNU/Linux
```
```
● org.openbmc.Sensors.service - Phosphor Sensor Manager
   Loaded: loaded (/lib/systemd/system/org.openbmc.Sensors.service; enabled; vendor preset: enabled)
   Active: activating (start) since Wed 2017-03-29 18:26:04 UTC; 12s ago
 Main PID: 847 (python)
   CGroup: /system.slice/org.openbmc.Sensors.service

Mar 29 18:26:04 witherspoon systemd[1]: Stopped Phosphor Sensor Manager.
Mar 29 18:26:04 witherspoon systemd[1]: Starting Phosphor Sensor Manager...
Mar 29 18:26:16 witherspoon sensor_manager2.py[847]: Traceback (most recent call last):
Mar 29 18:26:16 witherspoon sensor_manager2.py[847]:   File "/usr/sbin/sensor_manager2.py", line 58, in <module>
Mar 29 18:26:16 witherspoon sensor_manager2.py[847]:     sensor_obj = getattr(obmc.sensors, sensor_class)(bus, obj_path)
Mar 29 18:26:16 witherspoon sensor_manager2.py[847]: AttributeError: 'module' object has no attribute 'PowerSupplyDeratingSensor'
Mar 29 18:26:16 witherspoon systemd[1]: org.openbmc.Sensors.service: Main process exited, code=exited, status=1/FAILURE
Mar 29 18:26:16 witherspoon systemd[1]: Failed to start Phosphor Sensor Manager.
Mar 29 18:26:16 witherspoon systemd[1]: org.openbmc.Sensors.service: Unit entered failed state.
Mar 29 18:26:16 witherspoon systemd[1]: org.openbmc.Sensors.service: Failed with result 'exit-code'

root@witherspoon:~# uptime
 18:32:22 up 7 min,  load average: 0.97, 1.28, 0.82
root@witherspoon:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.NotReady
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.Off
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Off
root@witherspoon:~#
```

- [x] Journald  
[journald_v1.99.3-176-gb54c3b2_Sensors_DOA.txt](https://github.com/openbmc/openbmc/files/881149/journald_v1.99.3-176-gb54c3b2_Sensors_DOA.txt)

dkodihal, 2017-03-30:
Doesn't look like a code bug. This is happening because a skeleton bump happened without a pyphosphor dump. This commit from pyphosphor should fix this : 857e4b834f4bcece8d1006a1fbaef7c2847681d4.

gkeishin, 2017-03-30:
@dkodihal That's good to know..  

@williamspatrick @geissonator   So when a new build with this dependencies fix let me know.. If worked I'll close this one out.. Thanks

geissonator, 2017-03-30:
I see a pyphosphor update after this commin in https://github.com/openbmc/openbmc/commits/master  We good to go now with the latest?

gkeishin, 2017-03-30:
This new build https://openpower.xyz/job/openbmc-build/670/  havent seen this failure.. Closing it as fixed.
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.3-178"
VERSION_ID="v1.99.3-178-g5f115c0"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.3-178"
BUILD_ID="v1.99.3"
root@witherspoon:~# uname -a
Linux witherspoon 4.7.10-6786cab8778ba96d84d89c7aec34374b5f4f1968 #1 Thu Mar 30 11:36:52 UTC 2017 armv6l GNU/Linux
root@witherspoon:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.Off
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Off
root@witherspoon:~# uptime
 18:26:59 up 2 min,  load average: 1.41, 1.50, 0.65
root@witherspoon:~#
````

closed 2017-03-30 by gkeishin