openbmc/openbmc #1414: [Zaius] Unable to get the fru info for FRU=0

GHF, 2017-03-30:
Getting an error from ipmid about not being able to lookup FRU ID 0:

```
Mar 22 02:36:59 zaius ipmid[926]: LOOKATMEIncomplete Fru file.. Size:[112]
Mar 22 02:36:59 zaius ipmid[926]: Populating FRU areas failed for:[0]
Mar 22 02:37:01 zaius ipmid[926]: Incomplete Fru file.. Size:[168]
Mar 22 02:37:01 zaius ipmid[926]: Populating FRU areas failed for:[0]
Mar 22 02:37:01 zaius ipmid[926]: Incomplete Fru file.. Size:[224]
Mar 22 02:37:01 zaius ipmid[926]: Populating FRU areas failed for:[0]
Mar 22 02:37:01 zaius ipmid[926]: : Sensor should not be reported:  Type 0x00, Offset 0x01
Mar 22 02:37:01 zaius ipmid[926]: fru data size:[112], area offset:[8], area_size:[248]
Mar 22 02:37:01 zaius ipmid[926]: fru data size:[168], area offset:[8], area_size:[248]
Mar 22 02:37:01 zaius ipmid[926]: fru data size:[224], area offset:[8], area_size:[248]
Mar 22 02:37:01 zaius ipmid[926]: fru data size:[256], area offset:[8], area_size:[248]
Mar 22 02:37:01 zaius ipmid[926]: Successfully verified area checksum. offset:[8]
Mar 22 02:37:01 zaius ipmid[926]: SUCCESS: Populated FRU areas for:[/tmp/ipmifru00]
Mar 22 02:37:01 zaius ipmid[926]: _append_to_dict: VPD Key = [Manufacturer] : Type Code = [BINARY] : Len = [0] : Val = []
Mar 22 02:37:01 zaius ipmid[926]: _append_to_dict: VPD Key = [Name] : Type Code=[ASCII+Latin] : Len = [18] : Val = [OpenPOWER Firmware]
Mar 22 02:37:01 zaius ipmid[926]: _append_to_dict: VPD Key = [Model Number] : Type Code = [BINARY] : Len = [0] : Val = []
Mar 22 02:37:01 zaius ipmid[926]: _append_to_dict: VPD Key = [Version] : Type Code=[ASCII+Latin] : Len = [42] : Val = [open-power-zaius-0.9.8-43-g455603c9e-dirty]
Mar 22 02:37:01 zaius ipmid[926]: _append_to_dict: VPD Key = [Serial Number] : Type Code = [BINARY] : Len = [0] : Val = []
Mar 22 02:37:01 zaius ipmid[926]: _append_to_dict: VPD Key = [Asset Tag] : Type Code = [BINARY] : Len = [0] : Val = []
Mar 22 02:37:01 zaius ipmid[926]: _append_to_dict: VPD Key = [FRU File ID] : Type Code = [BINARY] : Len = [0] : Val = []
Mar 22 02:37:01 zaius ipmid[926]: _append_to_dict: VPD Key = [Custom Field 1] : Type Code=[ASCII+Latin] : Len = [36] : Val = [        buildroot-0.9.8-43-g455603c9e-dirty]
Mar 22 02:37:01 zaius ipmid[926]: _append_to_dict: VPD Key = [Custom Field 2] : Type Code=[ASCII+Latin] : Len = [6] : Val = [custom]
Mar 22 02:37:01 zaius ipmid[926]: _append_to_dict: VPD Key = [Custom Field 3] : Type Code=[ASCII+Latin] : Len = [6] : Val = [custom]
Mar 22 02:37:01 zaius ipmid[926]: _append_to_dict: VPD Key = [Custom Field 4] : Type Code=[ASCII+Latin] : Len = [6] : Val = [custom]
Mar 22 02:37:01 zaius ipmid[926]: _append_to_dict: VPD Key = [Custom Field 5] : Type Code=[ASCII+Latin] : Len = [6] : Val = [9a0eae]
Mar 22 02:37:01 zaius ipmid[926]: _append_to_dict: VPD Key = [Custom Field 6] : Type Code=[ASCII+Latin] : Len = [40] : Val = [81439cd58d8c51a3ad48247e71fb9475f3f37c51]
Mar 22 02:37:01 zaius ipmid[926]: _append_to_dict: VPD Key = [Custom Field 7] : Type Code=[ASCII+Latin] : Len = [6] : Val = [custom]
Mar 22 02:37:01 zaius ipmid[926]: _append_to_dict: VPD Key = [Custom Field 8] : Type Code=[ASCII+Latin] : Len = [40] : Val = [4b440901a7331507f56ca19a96c9203f7e570309]
Mar 22 02:37:01 zaius ipmid[926]: ERROR Unable to get the fru info for FRU=0
Mar 22 02:37:01 zaius ipmid[926]: Error updating inventory
```

in #1408, @williamspatrick mentions that it's the "sensor number for the host firmware version."

I'm not sure how to proceed for this. Is sensor 0 a reserved error number or does it implicitly serve some data? I don't see a definition for it in Witherspoon.py.

williamspatrick, 2017-03-31:
Sensor numbers would be assigned to hostboot in Zaius-MRW. We currently manually add those to system.py files as appropriate. If 0 is the sensor number being used for host firmware version it means either that is what is in the Zaius-MRW or perhaps the default is 0. Hopefully you don't have a bunch of sensors in the MRW with 0s.

gtmills, 2018-05-31:
@GHF No update in a year. Can this be closed or still a problem?

closed 2018-06-01 by GHF