openbmc/openbmc #1416: IPMI: No way to tell if host has booted

stewart-ibm, 2017-03-31:
The way to tell if a system has booted has been (on OpenPOWER P8 systems) to look at the 'Host Status' like so:
```
ipmitool -H 9.3.181.64 -I lanplus -P 0penBmc sdr elist |grep 'Host Status'
```

But on OpenBMC:
```
$ ipmitool -H 9.3.181.64 -I lanplus -P 0penBmc sdr elist
Err in cmd get sensor sdr info
Unable to open SDR for reading
```
```
root@witherspoon:~# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.3-179"
VERSION_ID="v1.99.3-179-g206f712"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.3-179"
BUILD_ID="v1.99.3"
root@witherspoon:~# 
```

This is how op-test-framework has been doing host booted detection on OpenPOWER POWER8, and this breaks with OpenBMC, making it impossible to tell if the machine has booted unless you run expect scripts over the console (which is a bit icky)

geissonator, 2017-03-31:
I wonder if there's an easy way to tie into the openbmc test framework?  All the things you need (power on / off, check if OS booted, grab errors, ...) are in there.  When it comes to openbmc, you really want to be using the REST api.  Here's an internal sensor that tracks the host boot status that you can grab via REST
https://github.com/openbmc/openbmc-test-automation/blob/master/lib/utils.robot#L73

stewart-ibm, 2017-04-01:
@geissonator unless someone from the OpenBMC project is going to volunteer to write the code, it's going to be IPMI for the foreseeable future. If it was RedFish rather than something custom, then it may be justifiable to spend the effort.

Otherwise, we're spending valuable dev/test time writing to a bespoke REST API that I'm guessing isn't going to last forever (or long after RedFish support lands).

We're facing at least four BMC implementations on POWER9 systems, and it's OpenBMC where I keep getting asked to rewrite everything to be OpenBMC specific and not being able to re-use existing (working) tooling.

williamspatrick, 2017-04-01:
Keep in mind that IPMI is not required on OpenBMC. The IPMI implementation we will have will always be lacking relative to REST.

It is likely going to be a few months before we get around to implementing the aspects of IPMI that satisfy this need, if at all.

stewart-ibm, 2017-04-03:
RedFish would be better from our PoV, that seems to be going to be implemented everywhere.

williamspatrick, 2017-04-03:
Except that Redfish doesn't actually document how to do very many "actions", so it is just as filled with OEM extensions as IPMI.  I challenge you to look through the Redfish schema and figure out how to even boot a system.

rfrandse, 2018-07-18:
IPMI content has been delivered to fix this.

closed 2018-07-18 by rfrandse