openbmc/openbmc #1417: [Palmetto/Barreleye] BMC failed to boot to Standby

swe12345, 2017-03-31:
BMC failed to boot to Standby

```
root@palmetto:~# uptime
 18:36:28 up 12 min,  load average: 0.00, 0.25, 0.34
```
```
root@palmetto:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.NotReady
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.Off
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Off
```
[journal_palm309_31mar.txt](https://github.com/openbmc/openbmc/files/885352/journal_palm309_31mar.txt)

[journal_barreleye.txt](https://github.com/openbmc/openbmc/files/885381/journal_barreleye.txt)

williamspatrick, 2017-03-31:
The only thing I see useful in the logs is a lot of failures related to reset-reload handling.

gkeishin, 2017-03-31:
Yeah yesterday @JoshDKing was looking into it and as per the conversation this appears to be, it was set to BMC_NOTREADY and waited for a JobRemoved Signal which that never came and the reason to why it didn't, he is also not sure..

geissonator, 2017-03-31:
Last I heard from Josh it looked like it may be an LED service that wasn't starting.  The reset/reload one is expected to fail.  obmc-led-group-start@BmcBooted.service maybe.

vishwabmc, 2017-03-31:
That one does not even exist. Is he using a older driver ?

geissonator, 2017-03-31:
Here's the service that's not starting:
```
[barreleye]# cat obmc-led-group-start@.service
[Unit]
Description=Assert %i LED
Wants=mapper-wait@-xyz-openbmc_project-led-groups-%i.service
After=mapper-wait@-xyz-openbmc_project-led-groups-%i.service
```

If you look at https://github.com/openbmc/openbmc/blob/95b315192f5220fe0e2d08c30ee373140394ed46/meta-phosphor/common/recipes-phosphor/leds/phosphor-led-manager.bb#L20   you can see that the service is actually xyz.openbmc_project.LED.GroupManager.service.  Looks to me like we need to update the service files referencing mapper-wait@-xyz-openbmc_project-led-groups-%i.service to mapper-wait@-xyz-openbmc_project-led-groupmanager-%i.service

This seems to still be an issue even in the latest code.

williamspatrick, 2017-04-01:
@geissonator - you are misunderstanding what mapper wait does. Mapper wait waits for a dbus object and not a service. I don't see anything wrong with this Wants call.

rahulmah, 2017-04-03:
After updating our system with today's latest BMC fw(build 675/v1.99.3-196-gbce4eb8), we are not seeing this issue on both Palmetto and Barreleye system. So not a blocker for us now.

vishwabmc, 2017-04-03:
BmcBooted has gone out of the picture some time ago. So I don't see any issue. Are we just trying to investigate a code that is not present today ?

geissonator, 2017-04-03:
Hmm, I'm still missing something, in https://github.com/openbmc/phosphor-led-manager/blob/master/configure.ac#L50  I see a BUSNAME of xyz.openbmc_project.LED.GroupManager, isn't that what that mapper call needs to be waiting on?

williamspatrick, 2017-04-03:
No. Mapper wait is for dbus paths, not service / bus name. If you want to wait on a service / bus name, systemd supports that natively.

geissonator, 2017-04-03:
Ahh ok, looks good then and per Rahul all is good now, closing out.

closed 2017-04-03 by geissonator