openbmc/openbmc #1418: Sprint 17.04.05 - Committed

rfrandse, 2017-03-31:


dkodihal, 2017-04-03:
Squad 1 sprint plan

Deepak 
- New code update story.
- REST -> d-bus via introspection #1160, OR
- Settingsd design (d-bus interfaces)

Ratan
- Network Refactor:- Support for set Interface address. #1293 (unless Ratan can finish in 03.22 sprint)
- Pick new stuff post wrapping n/w story.

Jayanth
- Add Power Cap Set and Enable/Disable Sensor support #1421 
- Define DBus Interfaces for OpenBMC Dump #1377 (potential carry-over)
- Define DBus Interfaces for OpenBMC Dump #1377 (potential carry-over)
- 5 days vacation

Tom
- Some portion of Implement SEL commands #443
- Vacation (5-6 days)

Dhruv
- Assert fault LED's when association is set and cleared #941 (potential carry over)
- openpower-proc-control: Callout processors on FSI failures #1323
- Persistent storage design (need story)

Devender
- Generate bmc errors on host checkstop and host watchdog failures #1266 (only host xstop?)
- New story in n/w refactor epic (DHCP related)

Vishwa
- hostcheckstop refactor #766 (potential carry over)
- witherspoon: power button LED should blink at standby #1399
- new story for OCC pass-through talking to SBE FIFO

anoo1, 2017-04-03:
Code update sprint planning:

Deepak -
mailbox daemon story(ies) (to be created today)

Gunnar -
https://github.com/openbmc/openbmc/issues/1351
https://github.com/openbmc/openbmc/issues/1352
https://github.com/openbmc/openbmc/issues/1353

Saqib -
https://github.com/openbmc/openbmc/issues/1206
https://github.com/openbmc/openbmc/issues/1401

Adriana -
https://github.com/openbmc/openbmc/issues/1278
mailbox daemon story if needed

bradbishop, 2017-04-03:
-------------------------------------------------------------------------------------------------
***Squad 2 -***
-------------------------------------------------------------------------------------------------
-------------------------------------------------------------------------------------------------
**Matt:spinler -** 
**17.01.11:**
- #629 (integrate device tree into kernel) - waiting

**17.03.22:**
- #1380 (Create fan group definition YAML)

-------------------------------------------------------------------------------------------------
**Matt B.:msbarth -** 
- #1343 (Sensor groups) 
-------------------------------------------------------------------------------------------------
**Brad:bradbishop -**
**17.03.08:**
- #1257 (sbefifo driver base)

**17.03.22:**
- #1303 (inventory Monitoring - Determine design and yaml input layout) 

-------------------------------------------------------------------------------------------------
**Chris -**
**17.03.08:**
- #1219 (OpenFSI targeted error handling for mulitple hub links)

**17.03.22:**
- #1222 ( OpenFSI: interrupt support for hub links.)
------------------------------------------------------------------------------------------------
**Eddie:eddiejames -**
**16.12.28:**
- #624 (initial occ intf refactor) - waiting on community
- #625 (p9 occ data) - waiting on community

**17.02.08:**
- #1004 (I2C migration to official interfaces)

**17.03.22:**
- #1258 (sbefifo in kernel api)

- #1348 (OCC: force hwmon attribute numbering to conform to dts sensor definitions)

- #1175 (64 bit addressing)

-------------------------------------------------------------------------------------------------
**Brandon Wyman -**
**17.03.22:**
- #370 ( Fan control - Air cool vs Water cool policies.)

-------------------------------------------------------------------------------------------------
**Dinesh Chinari  -**
- #1315 (Ensure uboot fw_env update script uses new inventory object)

gtmills, 2017-04-05:
Squad 3 -

Adriana:anoo1 -

17.03.08:

#1297 (eSEL Missing AdditionalData [ Witherspoon ])

#1344 (Block State Transitions during Image Activation)

Gunnar:gtmills -

17.03.22:

#1331 (ImageManager: Implement interface to download image via TFTP)

Andrew:geissonator -

17.03.08:

#1274 (host state heartbeat and target work on r/r)

17.03.22:

#1310 (Power Operations on Chassis)

Josh:JoshDKing -
Attends Tuesday and Thursday

17.02.22:

#1194 (Chassis and Host state services show incorrect states when targets exit)

17.03.08:

#1286 (Incorporate new soft power off function into host state)

17.03.22:

#1163 ( Cleanup old inventory related code)

Michael:mtritz -
17.03.08:

#773 (discover_system state refactor)

17.03.22:

#995 closed will be handled in #1303
#1381 (Provide P9 CFAM resiter overrides)

Saqib: -

17.03.08:

#932 (Upgrade systemd to 231 or above.)

17.03.22:

#1356 (ItemUpdater: Validate PNOR SquashFS image)

Leonel: -
17.03.22:

#1360 (CI dependency for phosphor-logging)

#1367 (REST login intermittent fails)

closed 2017-04-27 by rfrandse