openbmc/openbmc #1420: IPL: BMC in quiesce state following power on request - filesystem error prior on romulus with 910.1714.20170327n

cdeadmin, 2017-03-31:


geissonator, 2017-03-31:
Some notes from @spinler on this one
```
On Romulus, I don't even see the fsi scan running at all:

Mar 27 22:42:24 romulus systemd[1]: Starting Start Power0...
Mar 27 22:42:24 romulus systemd[1]: Created slice system-start_host.slice.
Mar 27 22:42:24 romulus systemd[1]: Created slice system-avsbus\x2dworkaround.slice.
Mar 27 22:42:24 romulus systemd[1]: Created slice system-fsi\x2dbind.slice.
Mar 27 22:42:24 romulus systemd[1]: Created slice system-pcie\x2dslot\x2ddetect.slice.
Mar 27 22:42:24 romulus power_control.exe[765]: PowerControl: setting power up SOFTWARE_PGOOD to 1
Mar 27 22:42:24 romulus power_control.exe[765]: PowerControl: setting power up BMC_POWER_UP to 1
Mar 27 22:42:24 romulus system_manager.py[729]: Running System State: HOST_POWERING_ON
Mar 27 22:42:24 romulus timemanager[803]:  PGOOD has changed..
Mar 27 22:42:24 romulus systemd[1]: Created slice system-cpld_trigger.slice.
Mar 27 22:42:24 romulus systemd[1]: Created slice system-op\x2dwait\x2dpower\x2don.slice.
Mar 27 22:42:24 romulus systemd[1]: Starting Wait for Power0 to turn on...
Mar 27 22:42:24 romulus systemd[1]: Starting Wait for /org/openbmc/watchdog/host0...
Mar 27 22:42:24 romulus systemd[1]: Created slice system-vcs_workaround.slice.
Mar 27 22:42:24 romulus systemd[1]: Created slice system-vrm\x2dcontrol.slice.
Mar 27 22:42:24 romulus systemd[1]: Starting Wait for /xyz/openbmc_project/state/chassis0...
Mar 27 22:42:24 romulus systemd[1]: Created slice system-fsi\x2dscan.slice.
Mar 27 22:42:24 romulus systemd[1]: Created slice system-avsbus\x2denable.slice.
Mar 27 22:42:25 romulus systemd[1]: Started Start Power0.
Mar 27 22:42:25 romulus systemd[1]: Started Wait for /org/openbmc/watchdog/host0.
Mar 27 22:42:25 romulus systemd[1]: Started Wait for /xyz/openbmc_project/state/chassis0.
Mar 27 22:42:25 romulus systemd[1]: Starting Perform AVS bus workaround on VRMs...
Mar 27 22:42:25 romulus systemd[1]: Started Perform AVS bus workaround on VRMs.
Mar 27 22:42:25 romulus systemd[1]: Starting Disable the AVS bus on the VRMs...
Mar 27 22:42:26 romulus systemd[1]: Started Disable the AVS bus on the VRMs.
Mar 27 22:42:26 romulus systemd[1]: Starting Apply voltage overrides to VRMs...
Mar 27 22:42:26 romulus vrm.sh[848]: rail        set        read        current
Mar 27 22:42:26 romulus vrm.sh[848]: -------        ------- ------- -------
Mar 27 22:42:26 romulus vrm.sh[848]: vdna        0.898V        0.898V        0.250A
Mar 27 22:42:27 romulus vrm.sh[848]: vdnb        0.898V        0.004V        0.000A
Mar 27 22:42:27 romulus systemd[1]: Started Apply voltage overrides to VRMs.
Mar 27 22:42:27 romulus systemd[1]: Starting Enable the AVS bus on VRMs...
Mar 27 22:42:27 romulus systemd[1]: Started Enable the AVS bus on VRMs.
Mar 27 22:42:27 romulus systemd[1]: Starting Run VCS workaround on host0...
Mar 27 22:42:27 romulus openpower-proc-control[915]: filesystem error: directory iterator cannot open directory: No such file or directory [/sys/devices/platform/fsi-master/slave@00:00/hub@00/]

So I expect this to be a service ordering problem.  I'll ask Lei to take a look since he owns Romulus changes.
```

mine260309, 2017-04-05:
It is indeed a service ordering problem, after `vcs_workaround` has dependency on `fsi_scan`, Romulus' own vcs_workaround service does not add this dependency and thus it fails.

I will re-order the services, moving cpld_trigger before vcs_workaround to make it give PGOOD before vcs_workaround, so that fsi_scan will run.

Eventually, the power on related services are:
1. avs_workaround
2. avsbus-disable
3. vrm-control
4. avsbus-enable
5. cpld_trigger (After this, PGOOD is asserted)
6. vcs_workaround
7. start_host

mine260309, 2017-04-05:
Patch submitted: https://gerrit.openbmc-project.xyz/#/c/3514/

swanman, 2017-04-06:
910.1716.20170405t was built with the patch mentioned and this has resolved our problem.  Please push this change up ASAP so we can get it into our production builds.  Thanks.

mine260309, 2017-04-07:
@williamspatrick Please help review the patch and merge, since it blocks Romulus power on.

closed 2017-04-09 by williamspatrick