openbmc/openbmc #1422: Inventory service crashed [ Witherspoon ]

gkeishin, 2017-04-03:
```
- [x] BUILD_ID   : https://openpower.xyz/job/openbmc-build/675/
- [x] VERSION_ID : v1.99.3-196-gbce4eb8
- [x] Kernel     : Linux witherspoon 4.7.10-6786cab8778ba96d84d89c7aec34374b5f4f1968 #1 Fri Mar 31 20:33:56 UTC 2017 armv6l GNU/Linux

- [x] PNOR VERSION:
IBM-witherspoon-ibm-OP9_v1.15_1.14
        op-build-v1.15-75-g986c911-dirty
        buildroot-2017.02-4-gc28573e
        skiboot-6a3a09c
        hostboot-a58f695
        linux-4.10.6-openpower1-pef4dc3d
        petitboot-v1.4.2-pe78e1ef
        witherspoon-xml-b208e4f
        occ-4141b5f
        hostboot-binaries-82b82cd
        capp-ucode-1bb7503
```

`BMC boots good and even the Power on went good. Once we started the CT, that's when this issue started showing up.. reboot or powercycle didn't fix this.. `

```
root@witherspoon:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.Off
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Off
root@witherspoon:~# uptime
 18:28:17 up 4 min,  load average: 0.34, 0.91, 0.46
root@witherspoon:~#

/xyz/openbmc_project/
[
    "/xyz/openbmc_project/sensors",
    "/xyz/openbmc_project/software",
    "/xyz/openbmc_project/object_mapper",
    "/xyz/openbmc_project/logging"
]

```

```
● xyz.openbmc_project.Inventory.Manager.service - Phosphor Inventory Manager
   Loaded: loaded (/lib/systemd/system/xyz.openbmc_project.Inventory.Manager.service; enabled; vendor preset: enabled)
   Active: failed (Result: start-limit-hit) since Wed 2017-03-29 18:24:35 UTC; 2min 59s ago
  Process: 798 ExecStart=/usr/sbin/phosphor-inventory (code=exited, status=127)
 Main PID: 798 (code=exited, status=127)

Mar 29 18:24:35 witherspoon systemd[1]: xyz.openbmc_project.Inventory.Manager.service: Unit entered failed state.
Mar 29 18:24:35 witherspoon systemd[1]: xyz.openbmc_project.Inventory.Manager.service: Failed with result 'exit-code'.
Mar 29 18:24:35 witherspoon systemd[1]: xyz.openbmc_project.Inventory.Manager.service: Service hold-off time over, scheduling restart.
Mar 29 18:24:35 witherspoon systemd[1]: Stopped Phosphor Inventory Manager.
Mar 29 18:24:35 witherspoon systemd[1]: xyz.openbmc_project.Inventory.Manager.service: Start request repeated too quickly.
Mar 29 18:24:35 witherspoon systemd[1]: Failed to start Phosphor Inventory Manager.
Mar 29 18:24:36 witherspoon systemd[1]: xyz.openbmc_project.Inventory.Manager.service: Unit entered failed state.
Mar 29 18:24:36 witherspoon systemd[1]: xyz.openbmc_project.Inventory.Manager.service: Failed with result 'start-limit-hit'.
root@witherspoon:~#
```

- [x] Journald 
[journald_99.3-196-gbce4eb8_inventory.txt](https://github.com/openbmc/openbmc/files/889340/journald_99.3-196-gbce4eb8_inventory.txt)

gkeishin, 2017-04-03:
@bradbishop This is just weird failure.. Can you take a quick look at this and see..

dkodihal, 2017-04-03:
Mar 29 18:24:23 witherspoon phosphor-inventory[751]: /usr/sbin/phosphor-inventory: symbol lookup error: /usr/sbin/phosphor-inventory: undefined symbol: _ZTVN9sdbusplus3xyz15openbmc_project9Inventory9Decorator6server11CoolingTypeE

williamspatrick, 2017-04-03:
Are there patches on the machine?  That is what it sounds like.

gkeishin, 2017-04-03:
Well I have code updated the system both BMC and PNOR.. so if something persist post that then I'm not sure.. By the way post code update it was running good and I was able to power on as well and I didnt see the service failure then. So when I ran the CT and post CT when I check on the system I see this failure.. Post that I couldnt make it work even with power cycle..   Trying out on other system as well..

gkeishin, 2017-04-03:
This seems to be from the BMC tar ball we installed for testing.. 
```
root@witherspoon:~# ls -lart /run/initramfs/rw/cow/usr/lib/
total 0
drwxr-xr-x    3 root     root             0 Mar 29 18:24 python2.7
drwxr-xr-x    4 root     root             0 Mar 31  2017 ..
drwxr-xr-x    3 root     root             0 Mar 31  2017 .
root@witherspoon:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.Off
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Off
root@witherspoon:~# tar -xf /tmp/obmc-phosphor-debug-tarball-witherspoon.tar.xz -C /usr/
root@witherspoon:~# log
logger        logging-test  login         login.shadow  logname       logout        logoutd       logread
root@witherspoon:~# which logging-test
/usr/bin/logging-test
root@witherspoon:~# ls -lart /run/initramfs/rw/cow/usr/lib/
total 1715
-rwxr-xr-x    1 root     root       1262096 Feb 22 06:25 libstdc++.so.6.0.21
-rwxr-xr-x    1 root     root         37040 Feb 22 06:25 libflash.so.5.3.0
-rwxr-xr-x    1 root     root        128896 Feb 22 06:31 liblzma.so.5.2.2
-rwxr-xr-x    1 root     root          5032 Mar 15 17:13 libsdbusplus.so.1.0.0
-rwxr-xr-x    1 root     root        300812 Mar 28 22:18 libphosphor_dbus.so.0.0.0
-rwxr-xr-x    1 root     root         20540 Mar 28 22:20 libphosphor_logging.so.1.0.0
drwxr-xr-x    6 root     root             0 Mar 28 22:20 ..
drwxr-xr-x    3 root     root             0 Mar 28 22:22 opkg
drwxr-xr-x    3 root     root             0 Mar 29 18:24 python2.7
lrwxrwxrwx    1 root     root            19 Mar 29 18:34 libstdc++.so.6 -> libstdc++.so.6.0.21
lrwxrwxrwx    1 root     root            16 Mar 29 18:34 liblzma.so.5 -> liblzma.so.5.2.2
lrwxrwxrwx    1 root     root            17 Mar 29 18:34 libflash.so -> libflash.so.5.3.0
lrwxrwxrwx    1 root     root            28 Mar 29 18:34 libphosphor_logging.so.1 -> libphosphor_logging.so.1.0.0
lrwxrwxrwx    1 root     root            25 Mar 29 18:34 libphosphor_dbus.so.0 -> libphosphor_dbus.so.0.0.0
lrwxrwxrwx    1 root     root            21 Mar 29 18:34 libsdbusplus.so.1 -> libsdbusplus.so.1.0.0
drwxr-xr-x    4 root     root             0 Mar 29 18:34 .
root@witherspoon:~#
````
Once it takes reset. the issue is seen.

williamspatrick, 2017-04-03:
Please do not load content that way. We need to talk about what you are doing.

gkeishin, 2017-04-03:
@williamspatrick Sure .. We have not enable it yet in CT.. So we were trying out manually loading it and testing elog and callout using the binaries from the BMC tarball.

gkeishin, 2017-05-18:
This should fix this one.. https://gerrit.openbmc-project.xyz/#/c/4264/
Will close it up once merged..

closed 2017-05-22 by williamspatrick