openbmc/openbmc #1423: OpenBMCGUI: Power On shouldnt be enabled when server power state is unreachable

sivassrr, 2017-04-03:
We used the code what @iftekharuli provided for our PoC. We could do the PoC quickly. 

During that found out then when server power state is unreachable (for some reason BMC is not reachable), PowerOn button is in enabled state. 

Based on the "Server Power State" - PowerOn button should be enabled. 

Will provide screen shot to @iftekharuli

karlvochatzer, 2017-04-03:
Recommend the following behavior rule: 
When the server is in an Unreachable state (for whatever reason), display the Power On button in a disabled state. Should the server or BMC then return to a reachable state where it can now be powered back on, enable the Power On button on the page.

sivassrr, 2017-09-19:
Will double check when it comes to official driver. So closing it now.

closed 2017-09-19 by sivassrr