openbmc/openbmc #1426: HOST reaches quiesce state after host power on [Barreleye/Palmetto]

rahulmah, 2017-04-04:
HOST is reaching quiesce state after powering on host. This issue is seen with latest BMC build.

```
bash-4.1$ curl -c cjar -b cjar -k https://xx.xx.xx.xx/xyz/openbmc_project/state/host0
{
  "data": {
    "CurrentHostState": "xyz.openbmc_project.State.Host.HostState.Off",
    "RequestedHostTransition": "xyz.openbmc_project.State.Host.Transition.Off"
  },
  "message": "200 OK",
  "status": "ok"
}

bash-4.1$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -d "{\"data\": \"xyz.openbmc_project.State.Host.Transition.On\"}" -X PUT  https://9.3.3.30/xyz/openbmc_project/state/host0/attr/RequestedHostTransition
{
  "data": null,
  "message": "200 OK",
  "status": "ok"
}

bash-4.1$ curl -c cjar -b cjar -khttps://xx.xx.xx.xx/xyz/openbmc_project/state/host0                                                                          {
  "data": {
    "CurrentHostState": "xyz.openbmc_project.State.Host.HostState.Quiesced",
    "RequestedHostTransition": "xyz.openbmc_project.State.Host.Transition.On"
  },
  "message": "200 OK",
  "status": "ok"
}bash-4.1$
```

FFDC 
[journal_log_quiesce_issue.txt](https://github.com/openbmc/openbmc/files/892950/journal_log_quiesce_issue.txt)

rahulmah, 2017-04-04:
This issue may be similar to https://github.com/openbmc/openbmc/issues/1420. I have logged this as a separate issue because the "filesystem error" which is highlighted in the 1420 issue is not seen here.

geissonator, 2017-04-04:
Rahul, did you try reflashing PNOR?  The log just indicates the host is not starting and the watchdog is firing:
```
Mar 29 18:37:39 barreleye system_manager.py[754]: Running System State: HOST_BOOTING
Mar 29 18:37:39 barreleye host_watchdog.exe[679]: Setting watchdog poll interval to: 30000
Mar 29 18:37:39 barreleye host_watchdog.exe[679]: Starting watchdog with poll interval: 30000
Mar 29 18:38:09 barreleye host_watchdog.exe[679]: Polling watchdog: 0
Mar 29 18:38:09 barreleye chassis_control.py[707]: Watchdog Error, Going to quiesce
Mar 29 18:38:09 barreleye systemd[1]: Reached target Quiesce Target.
```

rahulmah, 2017-04-04:
Yes system was reflashed with latest PNOR - https://openpower.xyz/job/openpower-op-build/339/

sivassrr, 2017-04-04:
Based on today's discussion, We may need to try with the stable PNOR
https://w3.austin.ibm.com/afs/austin/projects/esw/op910.html - 0403a build's PNOR level

sivassrr, 2017-04-05:
With stable Barreleye driver, System IPL fine.

rahulmah, 2017-04-05:
I am seeing the issue with stable Barreleye pnor(see below) and latest BMC build 678/v1.99.3-203-g7a0b268. So issue is seen with stable build also. 

PNOR:
```
open-power-barreleye-v1.15-4-g333c9b6
        buildroot-2017.02-4-gc28573e
        skiboot-5.4.3
        hostboot-fe5c556
        linux-4.10.1-openpower1-p3080f75
        petitboot-v1.4.1-pb89a0fa
        barreleye-xml-253622f
        occ-f018940
        hostboot-binaries-4b44090
        capp-ucode-1bb7503
```
BMC fw
```
root@barreleye:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.3-203"
VERSION_ID="v1.99.3-203-g7a0b268"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.3-203"
BUILD_ID="v1.99.3"
```

sivassrr, 2017-04-06:
@rahulmah could you pls update which ISTEP we had passed. If ISTEP > 16 then we can request open-power /host boot team to look at it.

rahulmah, 2017-04-07:
@sivassrr : I retested after upgrading system with new latest BMC build 682/v1.99.4-2-g5e45c86 and stable barreleye image. Below is my observations

1. With first Host power on, I am able to see that system is boot to OS.
2. But with second power on onwards, I am seeing that host is going into "Quiesced" state. In serial console we dont see anything. It means none of the ISTEP is started.
FFDC:
[journal_newlogs_quiesced.txt.txt](https://github.com/openbmc/openbmc/files/904484/journal_newlogs_quiesced.txt.txt)

```
root@barreleye:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.Off
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Off
root@barreleye:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.On
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Running
root@barreleye:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.Off
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Off
root@barreleye:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.Off
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Quiesced
```

ratagupt, 2017-04-07:
Seems ib the second instance..fan control application was having failure.
```Mar 30 01:17:24 barreleye fan_control.exe[810]: Failed to GetObject: Traceback (most recent call last):
Mar 30 01:17:24 barreleye fan_control.exe[810]:   File "/usr/lib/python2.7/site-packages/dbus/service.py", line 707, in _message_cb
Mar 30 01:17:24 barreleye fan_control.exe[810]:     retval = candidate_method(self, *args, **keywords)
Mar 30 01:17:24 barreleye fan_control.exe[810]: TypeError: GetObject() takes exactly 3 arguments (2 given)
Mar 30 01:17:24 barreleye fan_control.exe[810]: fanctl: Failed to get bus name for /org/openbmc/sensors/speed/fan0
Mar 30 01:17:24 barreleye fan_control.exe[810]: fanctl: Failed to set fan[0] speed[255]
Mar 30 01:17:24 barreleye systemd[1]: Started Set Fans to Maximum.
Mar 30 01:17:25 barreleye kernel: ftgmac100 1e660000.ethernet eth0: NCSI interface down
Mar 30 01:17:25 barreleye kernel[678]: ftgmac100 1e660000.ethernet eth0: NCSI interface down
Mar 30 01:17:27 barreleye kernel: ftgmac100 1e660000.ethernet eth0: NCSI interface up
Mar 30 01:17:27 barreleye kernel[678]: ftgmac100 1e660000.ethernet eth0: NCSI interface up
Mar 30 01:17:53 barreleye systemd[1]: op-wait-power-on@0.service: Start operation timed out. Terminating.
Mar 30 01:17:53 barreleye systemd[1]: op-wait-power-on@0.service: Main process exited, code=killed, status=15/TERM
Mar 30 01:17:53 barreleye systemd[1]: Failed to start Wait for Power0 to turn on.
Mar 30 01:17:53 barreleye systemd[1]: Dependency failed for Chassis0 (Power On).
Mar 30 01:17:53 barreleye systemd[1]: Dependency failed for Chassis0 (Start).
Mar 30 01:17:53 barreleye systemd[1]: obmc-chassis-start@0.target: Job obmc-chassis-start@0.target/start failed with result 'dependency'.
Mar 30 01:17:53 barreleye systemd[1]: obmc-chassis-start@0.target: Triggering OnFailure= dependencies.
Mar 30 01:17:53 barreleye systemd[1]: obmc-power-chassis-on@0.target: Job obmc-power-chassis-on@0.target/start failed with result 'dependency'.
Mar 30 01:17:53 barreleye systemd[1]: obmc-power-chassis-on@0.target: Triggering OnFailure= dependencies.
Mar 30 01:17:53 barreleye systemd[1]: op-wait-power-on@0.service: Unit entered failed state.
Mar 30 01:17:53 barreleye systemd[1]: op-wait-power-on@0.service: Failed with result 'timeout'.
Mar 30 01:17:53 barreleye systemd[1]: Started Stop Power0.
Mar 30 01:17:53 barreleye systemd[1]: Starting Wait for Power0 to turn off...```

ratagupt, 2017-04-07:
I can see that there was some network up and down activity around that time.
```barreleye systemd[1]: op-wait-power-on@0.service: Start operation timed out. Terminating.
Mar 30 01:17:53 barreleye systemd[1]: op-wait-power-on@0.service: Main process exited, code=killed, status=15/TERM
Mar 30 01:17:53 barreleye systemd[1]: Failed to start Wait for Power0 to turn on.
 barreleye systemd[1]: obmc-power-chassis-on@0.target: Triggering OnFailure= dependencies. ```

sivassrr, 2017-04-07:
Apr 07 13:18:33 palmetto systemd[1]: dev-hugepages.mount: Couldn't determine result for ConditionVirtualization=!private-users, assuming failed: No such file or directory
Apr 07 13:18:55 palmetto kernel[678]: systemd[1]: dev-hugepages.mount: Couldn't determine result for ConditionVirtualization=!private-users, assuming failed: No such file or directory
Apr 07 13:19:21 palmetto avahi-daemon[682]: ../../avahi-0.6.32/avahi-daemon/chroot.c: open() failed: No such file or directory
Apr 07 13:20:12 palmetto systemd[1]: obmc-mgr-state.service: Unit entered failed state.
Apr 07 13:20:15 palmetto phosphor-gevent[702]: <Greenlet at 0xb657f2b0L: <bound method WSGIServer.wrap_socket_and_handle of <WSGIServer at 0xb6561270L fileno=5 address=('::14bb:894a:b02d:b9b6%3065589152', 443, 0, 3065589152L)>>(<socket at 0xb6595850L fileno=9>, ('::6039:b7b6:8e91:f402%1250468860', 52738, 0, 125)> failed with SSLEOFError
Apr 07 13:20:15 palmetto phosphor-gevent[702]: <Greenlet at 0xb657f580L: <bound method WSGIServer.wrap_socket_and_handle of <WSGIServer at 0xb6561270L fileno=5 address=('::14bb:894a:b02d:b9b6%3065589152', 443, 0, 3065589152L)>>(<socket at 0xb65957f0L fileno=10>, ('::6039:b7b6:8e91:f402%1250468860', 52740, 0, 125)> failed with SSLEOFError

sivassrr, 2017-04-07:
No output from obmc-console-client and port 2200 as well.

sivassrr, 2017-04-07:
[Dr682Palmettofailure.zip](https://github.com/openbmc/openbmc/files/905753/Dr682Palmettofailure.zip)

prkatti1, 2017-04-12:
barreleye system going into quisce state on code update.

System details
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.4-2"
VERSION_ID="v1.99.4-2-g5e45c86"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.4-2"
BUILD_ID="v1.99.4"

Linux barreleye 4.7.10-6786cab8778ba96d84d89c7aec34374b5f4f1968 #1 Thu Apr 6 21:29:46 UTC 2017 armv5tejl GNU/Linux

open-power-barreleye-v1.15-4-g333c9b6
        buildroot-2017.02-4-gc28573e
        skiboot-5.4.3
        hostboot-fe5c556
        linux-4.10.1-openpower1-p3080f75
        petitboot-v1.4.1-pb89a0fa
        barreleye-xml-253622f
        occ-f018940
        hostboot-binaries-4b44090
        capp-ucode-1bb7503

yadron, 2017-04-12:
Could this change be the reason of Barreleye/"cannot boot host and going always to quiesce" (see the new code line **print "Watchdog Error, Going to quiesce"** at the end of this code listing):

```
diff /home/build/openbmc-feb28/openbmc/build/tmp/work/all-openbmc-linux/obmc-control-chassis/1.0-r1/git/pychassisctl/chassis_control.py ./tmp/work/all-openbmc-linux/obmc-control-chassis/1.0-r1/git/pychassisctl/chassis_control.py
47,51d46
<             'identify_led': {
<                 'bus_name': 'org.openbmc.control.led',
<                 'object_name': '/org/openbmc/control/led/identify',
<                 'interface_name': 'org.openbmc.Led'
<             },
104,119d98
<     def setIdentify(self):
<         print "Turn on identify"
<         intf = self.getInterface('identify_led')
<         intf.setOn()
<         return None
< 
<     @dbus.service.method(DBUS_NAME,
<                          in_signature='', out_signature='')
<     def clearIdentify(self):
<         print "Turn on identify"
<         intf = self.getInterface('identify_led')
<         intf.setOff()
<         return None
< 
<     @dbus.service.method(DBUS_NAME,
<                          in_signature='', out_signature='')
169a149,156
>                          in_signature='', out_signature='')
>     def quiesce(self):
>         intf = self.getInterface('systemd')
>         f = getattr(intf, 'StartUnit')
>         f.call_async('obmc-quiesce-host@0.target', 'replace')
>         return None
> 
>     @dbus.service.method(DBUS_NAME,
202,204c189,190
<         print "Watchdog Error, Hard Rebooting"
<         self.Set(DBUS_NAME, "reboot", 1)
<         self.powerOff()
---
>         print "Watchdog Error, Going to quiesce"
>         self.quiesce()
```

and note that this renaming maybe does something too:

```
[build@obmc build]$ diff ~/openbmc-28feb/openbmc/build/tmp/work/armv5e-openbmc-linux-gnueabi/obmc-hostcheckstop/1.0-r1/git/hostcheckstop/host_checkstop_obj.c ./tmp/work/armv5e-openbmc-linux-gnueabi/obmc-hostcheckstop/1.0-r1/git/hostcheckstop/host_checkstop_obj.c
80c80
<         printf("Host Checkstop, rebooting host\n");
---
>         printf("Host Checkstop, going to quiesce\n");
85,87c85,87
<             "org.openbmc.control.Chassis", /* name */
<             "/org/openbmc/control/chassis0", /* object path */
<             "org.openbmc.control.Chassis", /* interface name */
---
>             "org.freedesktop.systemd1", /* name */
>             "/org/freedesktop/systemd1", /* object path */
>             "org.freedesktop.systemd1.Manager", /* interface name */
94,95c94,95
<             "reboot",
<             parm,
---
>             "StartUnit",
>             g_variant_new("(ss)", "obmc-quiesce-host@0.target", "replace"),
100a101
>         g_variant_unref(result);
203d203
<
```

and

```
[build@obmc build]$ diff ~/openbmc-28feb/openbmc/build/tmp/work/all-openbmc-linux/obmc-control-chassis/1.0-r1/git/fanctl/fan_control.c  tmp/work/all-openbmc-linux/obmc-control-chassis/1.0-r1/git/fanctl/fan_control.c 
26,28c26,28
< const char *objectmapper_service_name =  "org.openbmc.ObjectMapper";
< const char *objectmapper_object_name  =  "/org/openbmc/ObjectMapper";
< const char *objectmapper_intf_name    =  "org.openbmc.ObjectMapper";
---
> const char *objectmapper_service_name =  "xyz.openbmc_project.ObjectMapper";
> const char *objectmapper_object_name  =  "/xyz/openbmc_project/object_mapper";
> const char *objectmapper_intf_name    =  "xyz.openbmc_project.ObjectMapper";
```

and

```
diff /home/build/openbmc-feb28/openbmc/build/tmp/work/all-openbmc-linux/obmc-control-chassis/1.0-r1/git/pciedetect/pcie_slot_present_obj.c ./tmp/work/all-openbmc-linux/obmc-control-chassis/1.0-r1/git/pciedetect/pcie_slot_present_obj.c
41,43c41,43
<                       "org.openbmc.ObjectMapper", /* name */
<                       "/org/openbmc/ObjectMapper", /* object path */
<                       "org.openbmc.ObjectMapper", /* interface name */
---
>                       "xyz.openbmc_project.ObjectMapper", /* name */
>                       "/xyz/openbmc_project/object_mapper", /* object path */
>                       "xyz.openbmc_project.ObjectMapper", /* interface name */
[build@obmc build]$
```

Looking to logs, I see this "Watchdog Error, Going to quiesce":

```
root@barreleye:~# journalctl |egrep -i "quiesce"
Apr 12 10:43:21 barreleye chassis_control.py[687]: Watchdog Error, Going to quiesce
Apr 12 10:43:21 barreleye systemd[1]: Reached target Quiesce Target.
Apr 12 10:43:22 barreleye systemd[1]: Stopped target Quiesce Target.
Apr 12 10:44:13 barreleye chassis_control.py[687]: Watchdog Error, Going to quiesce
Apr 12 10:44:13 barreleye systemd[1]: Reached target Quiesce Target.
Apr 12 10:44:16 barreleye phosphor-host-state-manager[784]: Auto reboot disabled. Maintaining quiesce.
Apr 12 10:49:35 barreleye systemd[1]: Stopped target Quiesce Target.
Apr 12 10:51:13 barreleye systemd[1]: Reached target Quiesce Target.
Apr 12 10:51:13 barreleye systemd[1]: Stopped target Quiesce Target.
Apr 12 10:51:15 barreleye phosphor-host-state-manager[784]: Auto reboot disabled. Maintaining quiesce.
Apr 12 10:51:53 barreleye chassis_control.py[687]: Watchdog Error, Going to quiesce
Apr 12 10:51:54 barreleye systemd[1]: Reached target Quiesce Target.
Apr 12 10:51:54 barreleye systemd[1]: Stopped target Quiesce Target.
Apr 12 10:52:36 barreleye chassis_control.py[687]: Watchdog Error, Going to quiesce
Apr 12 10:52:36 barreleye systemd[1]: Reached target Quiesce Target.
Apr 12 10:52:36 barreleye phosphor-host-state-manager[784]: Auto reboot disabled. Maintaining quiesce.
```

causten, 2017-05-03:
Any news on this one?  This seems to hit other systems beyond Palmetto and Barreleye

gkeishin, 2017-05-04:
Closing as discussed... with @sivassrr

sivassrr, 2017-05-04:
@causten with the new PNOR what we got from Uma Yadlapatti, We could power on successfully.

mine260309, 2017-05-04:
@sivassrr could you kindly help to do more tests? e.g. do a few power cycle of host and bmc.

I tested on several Barreleye/Palmetto machines and on a Garrison these days, sometimes the boot is OK, sometimes it fails without any host console output.
I can not find stable steps to reproduce the issue though.

gkeishin, 2017-05-05:
Looks like there is still issue with this as per @rahulmah .. Re-openning. He'll update the details

sivassrr, 2017-05-05:
@mine260309 yup, yesterday we tested it as part of daily driver. For the past several weeks, we were not able to cross beyond this point. Now we could able to power on. Will try loop test.

rahulmah, 2017-05-05:
I dont see this issue seem to be resolved fully. Tested on below latest build.

Palmetto : BMC Build - 721, PNOR Build - 345 --- Hitting Quiesces state still.
Barreleye :  BMC Build - 721, PNOR Build - barreleye_03133208.pnor ---- Not hitting with these build. But let us wait and see if we are not hitting with some more future builds.

mine260309, 2017-05-05:
On a Palmetto in Shanghai lab, the PNOR is behaving abnormal, when it boots OK the PNOR is fine. When it fails to boot, reading the first few bytes from PNOR shows the first 4 bytes are corrupted:
```
root@palmetto:/tmp# hexdump -C -n 64 /dev/mtd6ro
00000000  ff ff ff ff 00 00 00 01  00 00 00 01 00 00 00 80  |................|
00000010  00 00 00 16 00 00 10 00  00 00 20 00 00 00 00 00  |.......... .....|
00000020  00 00 00 00 00 00 00 00  00 00 00 00 50 41 62 c2  |............PAb.|
00000030  70 61 72 74 00 00 00 00  00 00 00 00 00 00 00 00  |part............|
```
The expected 4 bytes are:
```
00000000  50 41 52 54 00 00 00 01  00 00 00 01 00 00 00 80  |PART............|
```
This may be a single issue on this specific machine, but it is worth to check the PNOR content when a machine fails to boot.

causten, 2017-05-08:
Great data point!  Keep up the debugging.   Having a consistent boot experience is really important

mine260309, 2017-05-09:
There is another finding about this issue: once a machine is AC power-cycled, it fails to boot host in the first few minutes.
My steps:
1. On a Palmetto with a good PNOR chip;
2. AC power cycle and check journal log `journalctl -f | tee /tmp/log&`
3. Wait for BMC_READY, after that, run `obmcutil poweron`
4. On another session, run `ssh -p 2200 root@<bmc-ip>` to check the host console;
5. Check the journal log and host console, we can see there is no host console output, and eventually it enters quiesced state
6. Keep trying poweroff/poweron, the issue reamins;
7. After a few minutes (typically 6~7 minutes after BMC_READY), it just boots to host well, and we can see host console logs;
8. After that, poweroff/poweron always work well, even BMC is rebooted.

I see similar behavior on Garrison and MSI's machine as well.

So there should be something wrong during the first few minutes after AC power cycle.

mine260309, 2017-05-09:
Attached the journal log.
[log_palmetto_boot_after_ac_cycle.txt](https://github.com/openbmc/openbmc/files/985957/log_palmetto_boot_after_ac_cycle.txt)

rahulmah, 2017-05-09:
Again seeing the issue with both Barreleye and palmetto. Tested with latest BMC build 729/v1.99.5-15-gc4c2bed build and below latest PNOR image. So this issue seem to be not resolved.

Barreleye PNOR : /gsa/ausgsa/home/u/m/umay/web/public/barreleye_03133208.pnor
Palmetto : https://openpower.xyz/job/openpower-op-build/345/distro=ubuntu,target=palmetto/artifact/images/palmetto.pnor

mdmillerii, 2017-05-09:
@rahulmah is this what you flashed or what you read?  As I don't see the corrupt signature that @mine260309 reported.

mine260309, 2017-05-11:
On the Palmetto I have one more finding:
When the issue occurs, do a read of PNOR part, and then power on, it boots fine.

So I have stable steps to reproduce the issue and a workaround:
1. AC-power cycle;
2. It fails to boot host during a few minutes;
3. During the period, run `hexdump -C -n 64 /dev/mtd6ro` to read the PNOR data, check the data should be valid, something like:
```
# hexdump -C -n 64 /dev/mtd6ro
00000000  50 41 52 54 00 00 00 01  00 00 00 01 00 00 00 80  |PART............|
00000010  00 00 00 16 00 00 10 00  00 00 20 00 00 00 00 00  |.......... .....|
00000020  00 00 00 00 00 00 00 00  00 00 00 00 50 41 62 c2  |............PAb.|
00000030  70 61 72 74 00 00 00 00  00 00 00 00 00 00 00 00  |part............|
```
4. Try power on again, it succeeds.

mine260309, 2017-05-11:
@mdmillerii @rahulmah The "corrupted" PNOR bytes issue should be a standalone issue duo to a bad flash chip. I replaced it and so far I do not see corrupted bytes anymore.

yadron, 2017-05-11:
@mine260309 ,
BTW, I've seen similar with Barreleye: if I read/write part of PNOR, then it starts booting host (actually it looks like it is frozen during SBE or other very-very first stage of booting before any output to console, so looks like SBE or other early stage booter cannot read/write PNOR).
As a guess, possibly the PNOR locked by BMC for writing (GPIO lock like GPIOFn or something) and that is the reason why SBE cannot read/write PNOR during booting and frozen.
If this guess correct, then someone need to find what changed at the end of Feb or begin or March (~ after 28 Feb), so after this date OpenBMC doesn't allow read/write PNOR during host-booting ((and both it looks for me like for old BarrelEye/Palmeto platforms due to some hardware and boot-loaders bugs some communications between boot-loaders and hardware and BMC during early stage of boots goes using reading/writing PNOR, see for example https://github.com/openbmc/openbmc/issues/1518 and https://github.com/open-power/hostboot/issues/29 ).

More details regarding my how-to here (I've been read/writing BMC_INC part of PNOR before booting host):
https://github.com/openbmc/openbmc/issues/1454

Also I've posted similar info the I cannot boot Barreleye with latest PNOR/OpenBMC yesterday here:
https://github.com/openbmc/openbmc/issues/1482

yadron, 2017-05-11:
1. I've noted that if I revert Barreleye back to stable OpenBMC 1.99.0*, then still cannot boot host until AC power reset (hardware reboot). Looks like some hardware problem exists (for example, PNOR flash becomes unreadable until hardware reset).

2. I've made Barreleye bootable/stable back (using older 1.99.0.* and stable PNOR built of 17 March 2017).

3. I've updated OpenBMC to 1.99.4-22 (known to boot host Ok after some magic with PNOR like read/write BMC_INV).

4. Cannot boot host. Checked twice (trying hardware reboot and so on).

5. I've noted that if I just run one command ("pflash -i") before command "obmcutil poweron", then host boots well without problems.

As result, it looks like some hardware problem prevents CPUs/LPC/etc reading PNOR before BMC read PNOR. In older OpenBMC (before March 2017) some read/write of PNOR existed as part of start boot-host, so it worked. In never OpenBMC you have to do any manual reading PNOR before running command "obmcutil poweron".

Could you tell how to fix hardware of Barreleye so it would start working correctly with PNOR?

geissonator, 2017-05-11:
Good data guys!  Eventually we'll get the new pnor mboxd function on P8 systems (which will fix this) but for now I'll get a new P8-specific service that does a "pflash -i".  The PNOR requires some tuning, the SBE and Hostboot do not do this tuning so the BMC needs to do it before starting the boot.  We used to have a service, inventory_upload, that did this pnor tuning under the covers but we removed that since the function wasn't needed.  Removing this is probably what caused this issue to start popping up.

mine260309, 2017-05-11:
@geissonator Although we can have a workaround fix for now, and when P8 systems are switched to mboxd, both fix the problem by reading PNOR (by BMC) before booting host.

I'm really curious why the issue occurs. Is it possible to investigate the root cause of the behavior (probably from LPC or HW perspective)?

geissonator, 2017-05-11:
My understanding is that the PNOR chip requires a certain amount of "tuning" prior to it's use.  The SBE does not have support for this tuning so it's a requirement for the BMC to do it.  As for the details on this tuning...yeah that would have to come from someone on the HW/LPC side.

closed 2017-05-11 by williamspatrick

mine260309, 2017-05-12:
Verified the `pflash -i` workaround fix on Palmetto. It now boots every time after AC power cycle.
@yadron I believe your opened issues https://github.com/openbmc/openbmc/issues/1518 and https://github.com/openbmc/openbmc/issues/1454 can be duplicated to this one, could you please re-test with this workaround?

yadron, 2017-05-12:
So workaround for Barreleye just one "pflash -i" that is similar to command "devmem 0x1e630004 w 0x203B2341" (any of these two cmds enough before "obmcutil poweron"), but in some cases for p8 platforms more registers needs to be updated manually before booting host:

```
devmem 0x1e630004 w 0x203B2341 [ -- for Barreleye it is enough to init only this one, and 3 below could be required for other P8* ]
devmem 0x1e789080 w 0x02ec0500 [ --for this cmd check if higher part of value need not to be changed, because possibly only "0500" part of value need to be rewritten]
devmem 0x1e789088 w 0x30000C00
devmem 0x1e78908c w 0xFC0003FF
```

What it does, the first devmem command similar to "pflash -i" (it just initializes SPI), second command initializes LPC-AHB bridge. And some other similar initialization. For finding it, copied values of these ASPEED2400 registers from OpenBMC "v1.99.0-429" when host was booted successfully and updated in OpenBMC "v1.99.5-28" just before running "obmcutil poweron". Probably if you have ASPEED2400 documentation, you may find more info about these registers there.

More details in #1454 

@geissonator , @williamspatrick , when you'll talk with HW/LPC guys, please ask them:

why older OpenBMC 1.99.0.* (and similar before 9 March 2017) worked well with same PNOR (without need of setting these four registers to correct values) and with later OpenBMC doesn't? Does mentioned mboxd module ( #1598, https://github.com/openbmc/mboxbridge/blob/master/Documentation/mboxd.md ) allows some communications to PNOR flash without LPC-AHB bridge or what? What is the root reason of hardware no access to PNOR flash problems? Does it mean that SBE/hostboot could not access PNOR flash before all these SPI and LPC-AHB initialized? Possibly there is one more way how SBE/hostboot could access PNOR using mboxd (or ipmid?) maybe via BMC instead of direct access, who can check that? Pls ask HW/LPC team and SBE and hostboot for checking this!

For me it looks like in older versions of OpenBMC not BMC itself initialized these four registers, but PNOR. If yes, then one more question to HW/LPC/SBE/hostboot teams: why with older OpenBMC the PNOR inited these registers itself, but with later OpenBMC the same PNOR doesn't do this anymore?

geissonator, 2017-05-12:
@yadron the theory is the earlier openbmc code was running openpower-inventory-upload/inventory_upload.py which did a pflash command under the covers.  We removed that so we lost the pnor init it was doing.