openbmc/openbmc #1429: [BarrelEye v1.99.3-48] netipmid failed reading chassis

yadron, 2017-04-04:
netipmid doesn't work if reading chassis uuid (that is similar to "obmcutil chassis" that works good):

```
journalctl |tail
Apr 04 09:46:08 barreleye netipmid[686]: Failed to get /org/openbmc/control/chassis0 bus name: Connection timed out
```
- got this after running this command at client:

`
ipmitool -vvv -I lanplus -H 192.168.39.3  -U root -P 0penBmc lan print
`

The source of problem as I could think:

`[build@obmc build]$ vi ./tmp/work/armv5e-openbmc-linux-gnueabi/phosphor-ipmi-host/1.0-r1/git/apphandler.cpp
`(because "Failed to get /org/openbmc/control/chassis0 bus name" here).

How to fix it?

tomjoseph83, 2017-04-25:
@yadron Did the ```lan print ``` command fail because of failing to read the chassis name?

yadron, 2017-04-25:
Yes, I think so.

tomjoseph83, 2017-04-26:
Even if reading the Chassis uuid fails we default to a hardcoded value. So the command would not failed because of failure to read the UUID value.

Please confirm whether the ```lan print``` fails. If it does not fail, you can go ahead and close this issue.

tomjoseph83, 2017-05-04:
@yadron Any update on this?

yadron, 2017-05-04:
I'm compiling latest OpenBMC for Barreleye. When it ready, I'll check if NET-IPMI "lan print" still fails and write here soon then.

tomjoseph83, 2017-05-11:
@yadron Can we get this closed? ```ipmitool lan print``` passes for me and doesn’t fail because of failing to read Chassis UUID.

yadron, 2017-05-18:
Yes, it doesn't fail because of Chassis UUID anymore, but still fails.
We can create separate bug, but anyway let me first print it here.

Why "RAKP 2 HMAC is invalid" happens?

It looks like it establishes IPMI v2/RMCP+ session, but cannot go on with RAKP1/RAKP2 authentication (so command "lan print" not executed).

If running ipmitool by net:

```
ipmitool -I lanplus -H 172.17.27.74 -U root -P 0penBmc lan print
> Error: no response from RAKP 1 message
Error: Received an Unexpected RAKP 2 message
Bad response length, len=52
Invalid channel: 4
```

If running at host (so using ipmitool with /dev/ipmi0), then it works:

```
ipmitool lan print
Set in Progress         : Set Complete
Auth Type Support       : MD5 
Auth Type Enable        : Callback : MD5 
                        : User     : MD5 
                        : Operator : MD5 
                        : Admin    : MD5 
                        : OEM      : MD5 
IP Address              : 172.17.27.74
Subnet Mask             : 255.255.254.0
MAC Address             : b2:53:0b:0d:18:ca
Default Gateway IP      : 172.17.26.1
Cipher Suite Priv Max   : Not Available
Bad Password Threshold  : Not Available
```

More verbose output of ipmitool by net:

```
ipmitool -vvv -I lanplus -H 172.17.27.74 -U root -P 0penBmc lan print        

>> Sending IPMI command payload
>>    netfn   : 0x06
>>    command : 0x38
>>    data    : 0x8e 0x04 

BUILDING A v1.5 COMMAND
>> IPMI Request Session Header
>>   Authtype   : NONE
>>   Sequence   : 0x00000000
>>   Session ID : 0x00000000
>> IPMI Request Message Header
>>   Rs Addr    : 20
>>   NetFn      : 06
>>   Rs LUN     : 0
>>   Rq Addr    : 81
>>   Rq Seq     : 00
>>   Rq Lun     : 0
>>   Command    : 38
<< IPMI Response Session Header
<<   Authtype                : NONE
<<   Payload type            : IPMI (0)
<<   Session ID              : 0x00000000
<<   Sequence                : 0x00000000
<<   IPMI Msg/Payload Length : 16
<< IPMI Response Message Header
<<   Rq Addr    : 81
<<   NetFn      : 07
<<   Rq LUN     : 0
<<   Rs Addr    : 20
<<   Rq Seq     : 00
<<   Rs Lun     : 0
<<   Command    : 38
<<   Compl Code : 0x00
>> SENDING AN OPEN SESSION REQUEST

<<OPEN SESSION RESPONSE
<<  Message tag                        : 0x00
<<  RMCP+ status                       : no errors
<<  Maximum privilege level            : Unknown (0x00)
<<  Console Session ID                 : 0xa0a2a3a4
<<  BMC Session ID                     : 0x2d16fc3f
<<  Negotiated authenticatin algorithm : hmac_sha1
<<  Negotiated integrity algorithm     : hmac_sha1_96
<<  Negotiated encryption algorithm    : aes_cbc_128

>> Console generated random number (16 bytes)
 fb 9d f4 21 9c cf 63 75 eb e2 38 3b ad fa 44 d6
>> SENDING A RAKP 1 MESSAGE

<<RAKP 2 MESSAGE
<<  Message tag                   : 0x00
<<  RMCP+ status                  : no errors
<<  Console Session ID            : 0xa0a2a3a4
<<  BMC random number             : 0xddd0072dcb10582be7fd1ff6546b099a
<<  BMC GUID                      : 0xd28b2a0c27be40008eb71d348c46bb73
<<  Key exchange auth code [sha1] : 0x6151311bf400a4321916fa86c30047c57e3d3efa

bmc_rand (16 bytes)
 dd d0 07 2d cb 10 58 2b e7 fd 1f f6 54 6b 09 9a
>> rakp2 mac input buffer (62 bytes)
 a4 a3 a2 a0 3f fc 16 2d fb 9d f4 21 9c cf 63 75
 eb e2 38 3b ad fa 44 d6 dd d0 07 2d cb 10 58 2b
 e7 fd 1f f6 54 6b 09 9a d2 8b 2a 0c 27 be 40 00
 8e b7 1d 34 8c 46 bb 73 14 04 72 6f 6f 74
>> rakp2 mac key (20 bytes)
 30 70 65 6e 42 6d 63 00 00 00 00 00 00 00 00 00
 00 00 00 00
>> rakp2 mac as computed by the remote console (20 bytes)
 23 82 82 1b fc f9 b5 62 e7 2b 78 62 cb d1 cc 65
 bc 6b ab eb
> RAKP 2 HMAC is invalid
Error: Unable to establish IPMI v2 / RMCP+ session
```

And I don't see any specific errors in output of BMC/journalctl.

Current version that I'm using at Barreleye:

```
cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.4-22"
VERSION_ID="v1.99.4-22-ga072ab2"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.4-22"
BUILD_ID="v1.99.4"
```

Looking to source codes, I see

```
        case PayloadType::RAKP1:
            outMessage = createResponse<PayloadType::RAKP2>(output, inMessage);
            break;
```

that is line 129 of

`build/tmp/work/armv5e-openbmc-linux-gnueabi/phosphor-ipmi-net/1.0-r1/git/message_handler.cpp
`
- does it mean that currently instead of RAKP1 it answers with RAKP2?

Who can help with fixing these issues with netipmid?

williamspatrick, 2017-05-18:
RAKP<n> is a state-machine in the IPMI authentication protocol. So we get state 1 and we respond with state 2. This looks correct. 

The verbose output is showing an HMAC failure. Tom can look into that.

yadron, 2017-05-18:
Continuing my prev msg, I've read
http://stackoverflow.com/questions/19537359/get-device-id-using-ipmitool-failed
and I can run

```
ipmitool -I lanplus -H 172.17.27.74 -U root -P 0penBmc raw 0x06 0x01
> Error: no response from RAKP 1 message
Error: Received an Unexpected RAKP 2 message
Bad response length, len=52
 00 80 01 99 02 8d 41 a7 00 4f 42 00 04 01 00
```

and one more attempt

```
ipmitool -I lanplus -H 172.17.27.74 -U root -P 0penBmc raw 0x06 0x01
> Error: no response from RAKP 1 message
Error: Received an Unexpected RAKP 2 message
Error: Received an Unexpected Open Session Response
> Error: no response from RAKP 1 message
Error: Received an Unexpected RAKP 2 message
Get Device ID command failed: 0xd4 Insufficient privilege level
Bad response length, len=15
 00
```

and once more

```
ipmitool -I lanplus -H 172.17.27.74 -U root -P 0penBmc raw 0x06 0x01
Error: Unable to establish IPMI v2 / RMCP+ session
```

, so looks like unstable and works differently.

Expected answer (if running at host):

```
ipmitool raw 0x06 0x01
 00 80 01 99 02 8d 41 a7 00 4f 42 00 04 01 00
```

I suggest debugging it first using simpler command "ipmitool -I lanplus -H 172.17.27.74 -U root -P 0penBmc raw 0x06 0x01" and when it starts working stable, then continuing with "lan print".

Log from journalctl:

```
May 18 13:16:23 barreleye netipmid[2024]: I> Channel::Read : DataIn Fd[3] Req[18] Recv[18]
May 18 13:16:23 barreleye netipmid[2024]: IPMI GROUP EXTENTIONS
May 18 13:16:23 barreleye netipmid[2024]: I> Channel::Read : DataIn Fd[3] Req[18] Recv[18]
May 18 13:16:23 barreleye netipmid[2024]: IPMI APP GET CHANNEL INFO
May 18 13:16:23 barreleye netipmid[2024]: I> Channel::Read : DataIn Fd[3] Req[18] Recv[18]
May 18 13:16:23 barreleye netipmid[2024]: IPMI APP GET CHANNEL INFO
May 18 13:16:23 barreleye netipmid[2024]: I> Channel::Read : DataIn Fd[3] Req[18] Recv[18]
May 18 13:16:23 barreleye netipmid[2024]: IPMI APP GET CHANNEL INFO
May 18 13:16:23 barreleye netipmid[2024]: I> Channel::Read : DataIn Fd[3] Req[18] Recv[18]
May 18 13:16:23 barreleye netipmid[2024]: IPMI APP GET CHANNEL INFO
May 18 13:16:23 barreleye netipmid[2024]: I> Channel::Read : DataIn Fd[3] Req[18] Recv[18]
May 18 13:16:23 barreleye netipmid[2024]: IPMI APP GET CHANNEL INFO
May 18 13:16:23 barreleye netipmid[2024]: I> Channel::Read : DataIn Fd[3] Req[18] Recv[18]
May 18 13:16:23 barreleye netipmid[2024]: IPMI APP GET CHANNEL INFO
May 18 13:16:23 barreleye netipmid[2024]: I> Channel::Read : DataIn Fd[3] Req[17] Recv[17]
May 18 13:16:23 barreleye netipmid[2024]: >> GetChannelCapabilities
May 18 13:16:23 barreleye netipmid[2024]: << GetChannelCapabilities
May 18 13:16:23 barreleye netipmid[2024]: I> Channel::Read : DataIn Fd[3] Req[30] Recv[30]
May 18 13:16:23 barreleye netipmid[2024]: >> openSession
May 18 13:16:23 barreleye netipmid[2024]: << openSession
May 18 13:16:23 barreleye netipmid[2024]: I> Channel::Read : DataIn Fd[3] Req[30] Recv[30]
May 18 13:16:23 barreleye netipmid[2024]: >> RAKP12
May 18 13:16:23 barreleye netipmid[2024]: << RAKP12
[... 229 lines cut ...]
May 18 13:20:19 barreleye ipmid[752]: IPMI GET_LAN
May 18 13:20:19 barreleye ipmid[752]: N/W data from HW IPv4:23:172.17.27.74:172.17.26.1
May 18 13:20:19 barreleye ipmid[752]: N/W data from Cache: ::
May 18 13:20:19 barreleye ipmid[752]: IPMI GET_LAN
May 18 13:20:18 barreleye kernel[662]: ftgmac100 1e660000.ethernet eth0: rx frame too long
May 18 13:20:19 barreleye ipmid[752]: N/W data from HW IPv4:23:172.17.27.74:172.17.26.1
May 18 13:20:19 barreleye ipmid[752]: N/W data from CacheUnsupported parameter 0xd
May 18 13:20:19 barreleye ipmid[752]: ERROR:[0x80] handling NetFn:[0xC], Cmd:[0x2]
May 18 13:20:19 barreleye ipmid[752]: Unsupported parameter 0xe
May 18 13:20:19 barreleye ipmid[752]: ERROR:[0x80] handling NetFn:[0xC], Cmd:[0x2]
May 18 13:20:19 barreleye ipmid[752]: Unsupported parameter 0xf
May 18 13:20:19 barreleye ipmid[752]: ERROR:[0x80] handling NetFn:[0xC], Cmd:[0x2]
May 18 13:20:19 barreleye ipmid[752]: Unsupported parameter 0x14
May 18 13:20:19 barreleye ipmid[752]: ERROR:[0x80] handling NetFn:[0xC], Cmd:[0x2]
May 18 13:20:19 barreleye ipmid[752]: Unsupported parameter 0x15
May 18 13:20:19 barreleye ipmid[752]: ERROR:[0x80] handling NetFn:[0xC], Cmd:[0x2]
May 18 13:20:19 barreleye ipmid[752]: Unsupported parameter 0x16
May 18 13:20:19 barreleye ipmid[752]: ERROR:[0x80] handling NetFn:[0xC], Cmd:[0x2]
May 18 13:20:19 barreleye ipmid[752]: Unsupported parameter 0x18
May 18 13:20:19 barreleye ipmid[752]: ERROR:[0x80] handling NetFn:[0xC], Cmd:[0x2]
May 18 13:20:19 barreleye ipmid[752]: Unsupported parameter 0x1a
May 18 13:20:19 barreleye ipmid[752]: ERROR:[0x80] handling NetFn:[0xC], Cmd:[0x2]
...

```

yongli3, 2017-07-07:
I encountered this problem too. The systemd log as below:
netipmid[1450]: Failed to get /org/openbmc/control/chassis0 bus name: Connection timed out

tomjoseph83, 2017-07-07:
@yadron @yongli3 

"ipmitool -vvv -I lanplus -H 172.17.27.74 -U root -P 0penBmc lan print"  is a wrong usage of the IPMI user account. IPMI has a nameless account that is currently supported. So no username is needed for establishing a session.

"ipmitool -vvv -I lanplus -H 172.17.27.74  -P 0penBmc lan print"  - this should be sufficient.

yongli3, 2017-07-09:
Thanks @tomjoseph83 about the "namelesss account" information

tomjoseph83, 2017-08-05:
@yadron I think we can close this issue, since we figured out it is an issue with the way ipmitool is used.

geissonator, 2017-08-07:
Please reopen if we still need something here.

closed 2017-08-07 by geissonator