openbmc/openbmc #1431: Concurrent Flash Controller Access

sjitindarsingh, 2017-04-05:
pflash has a built in flash driver which can be used to bypass the kernel driver and access the flash controller directly. If this is done while something else (like the kernel driver) is accessing the flash controller then we have a race condition and things are unlikely to go well.

Related Error Case Quoted Below:

"==== State: Open by: andrewg on 16 March 2017 07:27:52 ==== 

#=#=# 2017-03-16 07:27:50 (CDT) #=#=#
New OwnerInfo = [jmstanle@au1.ibm.com]

Joel, can you take a look?  I don't stop mboxd when I use pflash to update PNOR so I don't believe that's a requirement here when writing a PNOR partition?
#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=#=# 

==== State: Open by: micwalsh on 14 March 2017 12:23:08 ==== 

Here's the info you asked for:

910.1710.20170303a contains this:

cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.3-65"
VERSION_ID="v1.99.3-65-ga6f406c"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.3-65"
BUILD_ID="v1.99.3"

910.1710.20170306a contains this:
cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.3-39"
VERSION_ID="v1.99.3-39-g2cac4f7"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.3-39"
BUILD_ID="v1.99.3"

==== State: Open by: iawillia on 09 March 2017 21:11:55 ==== 

I do not know what these numbers mean:

910.1710.20170303a
910.1710.20170306a

Those are not valid BMC build IDs.  Please give the contents of /etc/os-release.

I suspect that you need to stop the mailbox daemon while attempting to run pflash on the latest code.  As mentioned in the error message "Is something else using the flash controller?"

systemctl stop mboxd.service 

==== State: Open by: micwalsh on 07 March 2017 15:02:49 ==== 

I successfully loaded 910.1710.20170306a and unpacked the associated tarball.

Then my command to disable manufacturing mode failed.

I tried manually unpacking the 910.1710.20170303a tarball and re-trying the failing command.  It still failed.  I then code updated back to 910.1710.20170303a and the problem did NOT recur.

The status file is /gsa/ausgsa/projects/a/autoipl/status/bard-04.170307.105903.status (i.e. https://tbaix2.rchland.ibm.com/afs/rchland.ibm.com/rel/common/non-share/fspusers/WWW/sysmgr/tailFile.php?file=/gsa/ausgsa/projects/a/autoipl/status/bard-04.170307.105903.status)

Here's an excerpt showing the failing command:

#(CST) 2017/03/07 11:29:58.196203 -    0.000000 - Issuing: { AutoPflash.pl -t beye6 -i /fspmount/csptest/bard-04/tarball/data/mnfgPolicyFlagsResetAttrOverrides.bin -p barreleye --user root --password 0penBmc --sys_user root --sys_password 0penBmc ; echo "autoscript_ret_code = ${?}" ; } > /gsa/ausgsa/projects/a/autoipl/status/bard-04.set_mfg_mode.170307.112958.status

/gsa/ausgsa/projects/a/autoipl/status/bard-04.set_mfg_mode.170307.112958.status contains the log generated by AutoPflash.pl (see https://tbaix2.rchland.ibm.com/afs/rchland.ibm.com/rel/common/non-share/fspusers/WWW/sysmgr/tailFile.php?file=/gsa/ausgsa/projects/a/autoipl/status/bard-04.set_mfg_mode.170307.112958.status).

Manually running the command on the beye6 system resulted in this:
root@barreleye:~# /usr/sbin/pflash -e -f -p /tmp/mnfgPolicyFlagsResetAttrOverrides.bin -P ATTR_TMP
FFS: Reading the flash has returned all 0xFF.
Are you reading erased flash?
Is something else using the flash controller?
Error 15 opening ffs !"

sjitindarsingh, 2017-04-05:
What is the version of pflash which was used to do this? Also can we clarify whether pflash was being run on the bmc or the host?

Can you try suspending (or stopping) the mbox daemon and seeing if it still occurs?

shenki, 2017-05-01:
Here are the build versions:

```
910.1710.20170303a contains this:

cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.3-65"
VERSION_ID="v1.99.3-65-ga6f406c"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.3-65"
BUILD_ID="v1.99.3"

910.1710.20170306a contains this:
cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.3-39"
VERSION_ID="v1.99.3-39-g2cac4f7"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.3-39"
BUILD_ID="v1.99.3"
```

amboar, 2018-04-18:
What are we actually trying to achieve with this issue?

gtmills, 2018-08-28:
@sjitindarsingh Can this be closed?

amboar, 2018-08-28:
I'm going to close it, I suspect it was user error.

closed 2018-08-28 by amboar