openbmc/openbmc #1432: dbus[812]: [system] Connection has not authenticated soon enough, closing it (auth_timeout=30000ms, elapsed: 30002ms)

johnhcwang, 2017-04-05:
Put the script to /etc/profile.d so that the system can auto reboot while boot to OS (ubuntu 16.04).
We ran this test over the weekend and got the 'Connection reset by peer' error.
```
sleep 60
/sbin/shutdown -r
```
```
Mar 24 14:28:34 zaius phosphor-host-state-manager[886]: Host State transaction request
Mar 24 14:28:34 zaius systemd[1]: Starting Stop Watchdog0...
Mar 24 14:28:34 zaius systemd[1]: Starting Disable AVSBus on VRMs...
Mar 24 14:28:34 zaius systemd[1]: Starting Turn VCS rails off...
Mar 24 14:28:34 zaius systemd[1]: Stopped target Chassis0 (Start).
Mar 24 14:28:34 zaius systemd[1]: Stopped Start Watchdog0.
Mar 24 14:28:34 zaius systemd[1]: Stopped PCIE Slot Presence Detection For Chassis0.
Mar 24 14:28:34 zaius systemd[1]: Stopped target Chassis0 (Power On).
Mar 24 14:28:34 zaius zaius_avsbus.sh[28782]: Disabling AVSBus on bus 7 VRM @0x60 rail 0x01...
Mar 24 14:28:34 zaius systemd[1]: Stopped target Host0 (Started).
Mar 24 14:28:34 zaius systemd[1]: Stopped target Start Host0.
Mar 24 14:28:35 zaius zaius_vcs.sh[28783]: Turning off VCS
Mar 24 14:28:35 zaius zaius_vcs.sh[28783]:         Setting UCD GPIO 5 to 0x3
Mar 24 14:28:35 zaius zaius_avsbus.sh[28782]: Disabling AVSBus on bus 7 VRM @0x64 rail 0x01...
Mar 24 14:28:34 zaius systemd[1]: Stopped Start host0 SBE.
Mar 24 14:28:34 zaius systemd[1]: Stopped target Start Host0 (Pre).
Mar 24 14:28:34 zaius systemd[1]: Stopped Turn VCS rails on.
Mar 24 14:28:34 zaius systemd[1]: Stopped Run VCS workaround on host0.
Mar 24 14:28:34 zaius systemd[1]: Stopped Enable AVSBus on VRMs.
Mar 24 14:28:34 zaius systemd[1]: Stopped Fix up Zaius host refclock.
Mar 24 14:28:34 zaius systemd[1]: Stopped Scan FSI devices.
Mar 24 14:28:34 zaius systemd[1]: Stopped target Power0 (On).
Mar 24 14:28:34 zaius systemd[1]: Stopped target Power0 On.
Mar 24 14:28:34 zaius systemd[1]: Stopped Wait for Power0 to turn on.
Mar 24 14:28:34 zaius systemd[1]: Stopped Start Power0.
Mar 24 14:28:34 zaius systemd[1]: Starting Wait for Power0 to turn off...
Mar 24 14:28:34 zaius ipmid[24993]: Transition request initiated successfully
Mar 24 14:28:34 zaius phosphor-host-state-manager[886]: Recieved signal that host is running
Mar 24 14:28:34 zaius phosphor-host-state-manager[886]: Change to Host State
Mar 24 14:28:34 zaius phosphor-chassis-state-manager[889]: Recieved signal that power ON is complete
Mar 24 14:28:34 zaius phosphor-chassis-state-manager[889]: Change to Chassis Power State
Mar 24 14:28:35 zaius zaius_avsbus.sh[28782]: Disabling AVSBus on bus 7 VRM @0x64 rail 0x00...
Mar 24 14:28:35 zaius zaius_avsbus.sh[28782]: Disabling AVSBus on bus 8 VRM @0x60 rail 0x01...
Mar 24 14:28:35 zaius zaius_vcs.sh[28783]:         Setting UCD GPIO 6 to 0x3
Mar 24 14:28:35 zaius zaius_avsbus.sh[28782]: Disabling AVSBus on bus 8 VRM @0x64 rail 0x01...
Mar 24 14:28:35 zaius zaius_avsbus.sh[28782]: Disabling AVSBus on bus 8 VRM @0x64 rail 0x00...
Mar 24 14:28:35 zaius systemd[1]: Started Disable AVSBus on VRMs.
Mar 24 14:28:35 zaius systemd[1]: Started Turn VCS rails off.
Mar 24 14:28:35 zaius systemd[1]: Reached target Power0 Off (Pre).
Mar 24 14:28:35 zaius systemd[1]: Started Stop Power0.
Mar 24 14:29:04 zaius dbus[812]: [system] Connection has not authenticated soon enough, closing it (auth_timeout=30000ms, elapsed: 30002ms)
Mar 24 14:29:04 zaius pgood_wait[28784]: Error obtaining host service: Connection reset by peer
Mar 24 14:29:04 zaius systemd[1]: op-wait-power-off@0.service: Main process exited, code=exited, status=1/FAILURE
Mar 24 14:29:04 zaius systemd[1]: Failed to start Wait for Power0 to turn off.
Mar 24 14:29:04 zaius systemd[1]: Dependency failed for Chassis0 (Power Off).
Mar 24 14:29:04 zaius systemd[1]: Dependency failed for Chassis0 (Stop).
Mar 24 14:29:04 zaius systemd[1]: obmc-chassis-stop@0.target: Job obmc-chassis-stop@0.target/start failed with result 'dependency'.
Mar 24 14:29:04 zaius systemd[1]: obmc-chassis-stop@0.target: Triggering OnFailure= dependencies.
Mar 24 14:29:04 zaius systemd[1]: obmc-power-chassis-off@0.target: Job obmc-power-chassis-off@0.target/start failed with result 'dependency'.
Mar 24 14:29:04 zaius systemd[1]: op-wait-power-off@0.service: Unit entered failed state.
Mar 24 14:29:04 zaius systemd[1]: op-wait-power-off@0.service: Failed with result 'exit-code'.
Mar 24 14:29:04 zaius systemd[1]: Starting Wait for Power0 to turn off...
Mar 24 14:29:05 zaius dbus[812]: [system] Connection has not authenticated soon enough, closing it (auth_timeout=30000ms, elapsed: 30033ms)
Mar 24 14:29:05 zaius mapper[28781]: Error finding '/org/openbmc/watchdog/host0' service: Connection reset by peer
Mar 24 14:29:05 zaius systemd[1]: obmc-stop-watchdog@0.service: Main process exited, code=exited, status=1/FAILURE
Mar 24 14:29:05 zaius systemd[1]: obmc-stop-watchdog@0.service: Unit entered failed state.
Mar 24 14:29:05 zaius systemd[1]: obmc-stop-watchdog@0.service: Failed with result 'exit-code'.
Mar 24 14:29:05 zaius systemd[1]: Stopped target Chassis0 (Power Off).
Mar 24 14:29:05 zaius systemd[1]: Reached target Quiesce Target.
Mar 24 14:29:05 zaius phosphor-chassis-state-manager[889]: Recieved signal that power OFF is complete
Mar 24 14:29:05 zaius phosphor-chassis-state-manager[889]: Change to Chassis Power State
Mar 24 14:29:05 zaius phosphor-host-state-manager[886]: Auto reboot enabled. Beginning reboot...
Mar 24 14:29:05 zaius phosphor-host-state-manager[886]: Host State transaction request
[... 16 lines cut ...]
Mar 24 14:29:35 zaius systemd[1]: op-wait-power-off@0.service: Unit entered failed state.
Mar 24 14:29:35 zaius systemd[1]: op-wait-power-off@0.service: Failed with result 'exit-code'.
Mar 24 14:29:35 zaius systemd[1]: Starting Wait for Power0 to turn off...
Mar 24 14:29:35 zaius dbus[812]: [system] Connection has not authenticated soon enough, closing it (auth_timeout=30000ms, elapsed: 30000ms)
Mar 24 14:29:35 zaius mapper[28836]: Error finding '/org/openbmc/watchdog/host0' service: Connection reset by peer
Mar 24 14:29:35 zaius systemd[1]: obmc-stop-watchdog@0.service: Main process exited, code=exited, status=1/FAILURE
Mar 24 14:29:35 zaius systemd[1]: obmc-stop-watchdog@0.service: Unit entered failed state.
Mar 24 14:29:35 zaius systemd[1]: obmc-stop-watchdog@0.service: Failed with result 'exit-code'.
Mar 24 14:30:05 zaius dbus[812]: [system] Connection has not authenticated soon enough, closing it (auth_timeout=30000ms, elapsed: 30029ms)
Mar 24 14:30:05 zaius pgood_wait[28841]: Error obtaining host service: Connection reset by peer
Mar 24 14:30:05 zaius systemd[1]: op-wait-power-off@0.service: Main process exited, code=exited, status=1/FAILURE
Mar 24 14:30:05 zaius systemd[1]: Failed to start Wait for Power0 to turn off.
Mar 24 14:30:05 zaius systemd[1]: Dependency failed for Chassis0 (Power Off).
Mar 24 14:30:05 zaius systemd[1]: obmc-power-chassis-off@0.target: Job obmc-power-chassis-off@0.target/start failed with result 'dependency'.
Mar 24 14:30:05 zaius systemd[1]: op-wait-power-off@0.service: Unit entered failed state.
Mar 24 14:30:05 zaius systemd[1]: op-wait-power-off@0.service: Failed with result 'exit-code'.
Mar 24 14:30:05 zaius systemd[1]: Reached target Power0 Off.
Mar 24 14:30:05 zaius systemd[1]: Starting Reload mboxd after power off...
Mar 24 14:30:05 zaius mboxd[793]: [ 1490365805.712396105] Pointing HOST LPC bus at the actual flash
Mar 24 14:30:05 zaius mboxd[793]: [ 1490365805.712557761] Assuming 64MB of flash: HOST LPC 0x0c000000
Mar 24 14:30:05 zaius mboxd[793]: [ 1490365805.712686408] Loading flash into ram at 0x72fe8000 for 0x04000000 bytes
Mar 24 14:30:16 zaius systemd[1]: Started Reload mboxd after power off.
Mar 25 01:34:46 zaius systemd[1]: Starting Cleanup of Temporary Directories...
Mar 25 01:34:46 zaius systemd[1]: Started Cleanup of Temporary Directories.
```

williamspatrick, 2017-04-05:
I don't understand the test fully.  Are you attempting to tell the BMC to reboot while it is in the middle of booting to runtime?  What is your expected behavior?

johnhcwang, 2017-04-05:
Sorry to confuse you. We put the script under the host OS /etc/profile.d/, not in BMC embedded linux.
We ran this test to make sure that the host can reboot normally and stably.

williamspatrick, 2017-04-05:
@johnhcwang - There was a leak of dbus sockets in one of the applications that was fixed with these two commits:

24f35b0 phosphor-ipmi-fru.inc: bump version
0ad5e78 phosphor-ipmi-host.bb: bump version

I think you'll need to upgrade at least to 24f35b0 if you want to do host-reboot testing.  We had similar problems in our automated reboot cycles reported and fixed with those commits.  See openbmc/openbmc#875.

johnhcwang, 2017-04-10:
Still meet the same issue after 218 host-reboot cycles. We ran the above script on two zaius systems.

The error point on first one:
- Connection has not authenticated soon enough
```
Apr 06 14:16:21 zaius systemd[1]: Reached target Power0 Off (Pre).
Apr 06 14:16:21 zaius systemd[1]: Started Stop Power0.
Apr 06 14:16:22 zaius kernel: ftgmac100 1e660000.ethernet eth0: NCSI interface down
Apr 06 14:16:22 zaius kernel[801]: ftgmac100 1e660000.ethernet eth0: NCSI interface down
Apr 06 14:16:22 zaius host_watchdog.exe[795]: Stopping watchdog
Apr 06 14:16:22 zaius systemd[1]: Started Stop Watchdog0.
ftgmac100 1e660000.ethernet eth0: NCSI interface down
Apr 06 14:16:29 zaius kernel: ftgmac100 1e660000.ethernet eth0: NCSI interface down
Apr 06 14:16:29 zaius kernel[801]: ftgmac100 1e660000.ethernet eth0: NCSI interface down
ftgmac100 1e660000.ethernet eth0: NCSI interface down
Apr 06 14:16:37 zaius kernel: ftgmac100 1e660000.ethernet eth0: NCSI interface down
Apr 06 14:16:37 zaius kernel[801]: ftgmac100 1e660000.ethernet eth0: NCSI interface down
ftgmac100 1e660000.ethernet eth0: NCSI interface down
Apr 06 14:16:44 zaius kernel: ftgmac100 1e660000.ethernet eth0: NCSI interface down
Apr 06 14:16:44 zaius kernel[801]: ftgmac100 1e660000.ethernet eth0: NCSI interface down
Apr 06 14:16:49 zaius dbus[817]: [system] Connection has not authenticated soon enough, closing it (auth_timeout=30000ms, elapsed: 30004ms)
Apr 06 14:16:49 zaius pgood_wait[6866]: Error obtaining host service: Connection reset by peer
Apr 06 14:16:49 zaius systemd[1]: op-wait-power-off@0.service: Main process exited, code=exited, status=1/FAILURE
Apr 06 14:16:49 zaius systemd[1]: Failed to start Wait for Power0 to turn off.
Apr 06 14:16:49 zaius systemd[1]: Dependency failed for Chassis0 (Power Off).
Apr 06 14:16:49 zaius systemd[1]: Dependency failed for Chassis0 (Stop).
Apr 06 14:16:49 zaius systemd[1]: obmc-chassis-stop@0.target: Job obmc-chassis-stop@0.target/start failed with result 'dependency'.
Apr 06 14:16:49 zaius systemd[1]: obmc-chassis-stop@0.target: Triggering OnFailure= dependencies.
Apr 06 14:16:49 zaius systemd[1]: obmc-power-chassis-off@0.target: Job obmc-power-chassis-off@0.target/start failed with result 'dependency'.
Apr 06 14:16:49 zaius systemd[1]: op-wait-power-off@0.service: Unit entered failed state.
Apr 06 14:16:49 zaius systemd[1]: op-wait-power-off@0.service: Failed with result 'exit-code'.
Apr 06 14:16:49 zaius systemd[1]: Starting Wait for Power0 to turn off...
Apr 06 14:16:50 zaius systemd[1]: Reached target Quiesce Target.
Apr 06 14:16:50 zaius phosphor-chassis-state-manager[889]: Recieved signal that power OFF is complete
Apr 06 14:16:50 zaius phosphor-chassis-state-manager[889]: Change to Chassis Power State
Apr 06 14:16:51 zaius phosphor-host-state-manager[888]: Auto reboot enabled. Beginning reboot...
Apr 06 14:16:51 zaius phosphor-host-state-manager[888]: Host State transaction request
Apr 06 14:16:51 zaius systemd[1]: Stopped target Quiesce Target.
Apr 06 14:16:51 zaius phosphor-host-state-manager[888]: Auto reboot enabled. Beginning reboot...
Apr 06 14:16:51 zaius phosphor-host-state-manager[888]: Host State transaction request
ftgmac100 1e660000.ethernet eth0: NCSI interface down
Apr 06 14:16:52 zaius dbus[817]: [system] Connection has not authenticated soon enough, closing it (auth_timeout=30000ms, elapsed: 30000ms)
Apr 06 14:16:52 zaius mapper[6911]: Error finding '/org/openbmc/control/power0' service: Connection reset by peer
Apr 06 14:16:52 zaius systemd[1]: op-power-stop@0.service: Main process exited, code=exited, status=1/FAILURE
```

The error point on another system:
- Connection has not authenticated soon enough
- The maximum number of active connections for UID 0 has been reached
```
Apr 06 17:39:42 zaius systemd[1]: Reached target Power0 Off (Pre).
Apr 06 17:39:42 zaius systemd[1]: Started Stop Power0.
Apr 06 17:40:11 zaius dbus[835]: [system] Connection has not authenticated soon enough, closing it (auth_timeout=30000ms, elapsed: 30001ms)
Apr 06 17:40:11 zaius mapper[7507]: Error finding '/org/openbmc/watchdog/host0' service: Connection reset by peer
Apr 06 17:40:11 zaius systemd[1]: obmc-stop-watchdog@0.service: Main process exited, code=exited, status=1/FAILURE
Apr 06 17:40:11 zaius systemd[1]: Failed to start Stop Watchdog0.
Apr 06 17:40:11 zaius systemd[1]: Dependency failed for Chassis0 (Stop).
Apr 06 17:40:11 zaius systemd[1]: obmc-chassis-stop@0.target: Job obmc-chassis-stop@0.target/start failed with result 'dependency'.
Apr 06 17:40:11 zaius systemd[1]: obmc-chassis-stop@0.target: Triggering OnFailure= dependencies.
Apr 06 17:40:11 zaius systemd[1]: obmc-stop-watchdog@0.service: Unit entered failed state.
Apr 06 17:40:11 zaius systemd[1]: obmc-stop-watchdog@0.service: Failed with result 'exit-code'.
Apr 06 17:40:11 zaius phosphor-chassis-state-manager[926]: Recieved signal that power OFF is complete
Apr 06 17:40:11 zaius phosphor-chassis-state-manager[926]: Change to Chassis Power State
Apr 06 17:40:11 zaius systemd[1]: Reached target Quiesce Target.
Apr 06 17:40:11 zaius phosphor-host-state-manager[925]: Auto reboot enabled. Beginning reboot...
Apr 06 17:40:11 zaius phosphor-host-state-manager[925]: Host State transaction request
Apr 06 17:40:12 zaius systemd[1]: Starting Stop Watchdog0...
Apr 06 17:40:12 zaius systemd[1]: Stopped target Quiesce Target.
Apr 06 17:40:12 zaius phosphor-host-state-manager[925]: Auto reboot enabled. Beginning reboot...
[... cut ...]
Apr 07 21:52:08 zaius systemd[1]: Stopped Phosphor Webserver.
Apr 07 21:52:08 zaius systemd[1]: Started Phosphor Webserver.
```
The test version is v1.99.4 including ucd workaround change.
```
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.4-1"
VERSION_ID="v1.99.4-1-gdfd88fb-dirty-john"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.4-1"
BUILD_ID="v1.99.4"
```

Is there any information I can dump for debugging?

mdmillerii, 2017-04-10:
@johnhcwang you can explore /proc/*/fd to find processes that have lots of open files files (directories with lots of entries, which are symlinks), then find the commands for those processes numbers like was done on #875 above.

mine260309, 2017-04-11:
I left a comment in #875.
This issue's root cause should be the same as #875, but we have not fixed all the issues, so it's still reproduced when we do more power cycles.

johnhcwang, 2017-04-12:
Explore /proc/*/fd for debugging
```
PID = 1 with 82 file descriptors
PID = 10 with 0 file descriptors
PID = 124 with 0 file descriptors
PID = 125 with 0 file descriptors
PID = 127 with 0 file descriptors
PID = 129 with 0 file descriptors
PID = 132 with 0 file descriptors
PID = 156 with 0 file descriptors
PID = 167 with 0 file descriptors
PID = 168 with 0 file descriptors
PID = 2 with 0 file descriptors
PID = 200 with 0 file descriptors
PID = 201 with 0 file descriptors
PID = 202 with 0 file descriptors
PID = 203 with 0 file descriptors
PID = 213 with 0 file descriptors
PID = 216 with 0 file descriptors
PID = 219 with 0 file descriptors
PID = 222 with 0 file descriptors
PID = 225 with 0 file descriptors
PID = 228 with 0 file descriptors
PID = 231 with 0 file descriptors
PID = 234 with 0 file descriptors
PID = 245 with 0 file descriptors
PID = 250 with 0 file descriptors
PID = 255 with 0 file descriptors
PID = 260 with 0 file descriptors
PID = 265 with 0 file descriptors
PID = 270 with 0 file descriptors
PID = 276 with 0 file descriptors
PID = 282 with 0 file descriptors
PID = 3 with 0 file descriptors
PID = 359 with 0 file descriptors
PID = 361 with 0 file descriptors
PID = 363 with 0 file descriptors
PID = 365 with 0 file descriptors
PID = 391 with 0 file descriptors
PID = 397 with 0 file descriptors
PID = 442 with 0 file descriptors
PID = 447 with 0 file descriptors
PID = 466 with 62 file descriptors
PID = 5 with 0 file descriptors
PID = 548 with 13 file descriptors
PID = 556 with 14 file descriptors
PID = 7 with 0 file descriptors
PID = 7221 with 0 file descriptors
PID = 7309 with 0 file descriptors
PID = 7333 with 0 file descriptors
PID = 7447 with 8 file descriptors
PID = 7448 with 6 file descriptors
PID = 7450 with 4 file descriptors
PID = 7458 with 0 file descriptors
PID = 7483 with 0 file descriptors
PID = 790 with 7 file descriptors
PID = 791 with 4 file descriptors
PID = 792 with 4 file descriptors
PID = 793 with 4 file descriptors
PID = 794 with 7 file descriptors
PID = 795 with 7 file descriptors
PID = 797 with 7 file descriptors
PID = 798 with 7 file descriptors
PID = 799 with 17 file descriptors
PID = 8 with 0 file descriptors
PID = 800 with 5 file descriptors
PID = 801 with 7 file descriptors
PID = 802 with 267 file descriptors ===> Too much
PID = 806 with 6 file descriptors
PID = 811 with 4 file descriptors
PID = 812 with 6 file descriptors
PID = 813 with 4 file descriptors
PID = 814 with 7 file descriptors
PID = 815 with 7 file descriptors
PID = 816 with 7 file descriptors
PID = 817 with 5 file descriptors
PID = 819 with 17 file descriptors
PID = 823 with 4 file descriptors
PID = 824 with 4 file descriptors
PID = 825 with 4 file descriptors
PID = 826 with 4 file descriptors
PID = 827 with 3 file descriptors
PID = 829 with 6 file descriptors
PID = 831 with 6 file descriptors
PID = 833 with 3 file descriptors
PID = 835 with 7 file descriptors
PID = 836 with 4 file descriptors
PID = 844 with 7 file descriptors
PID = 850 with 7 file descriptors
PID = 851 with 4 file descriptors
PID = 855 with 7 file descriptors
PID = 859 with 9 file descriptors
PID = 864 with 7 file descriptors
PID = 865 with 4 file descriptors
PID = 868 with 8 file descriptors
PID = 8771 with 0 file descriptors
PID = 8783 with 5 file descriptors
PID = 886 with 226 file descriptors ===> Too much
PID = 888 with 5 file descriptors
PID = 889 with 5 file descriptors
PID = 890 with 8 file descriptors
PID = 891 with 7 file descriptors
PID = 892 with 5 file descriptors
PID = 893 with 8 file descriptors
PID = 894 with 8 file descriptors
PID = 924 with 6 file descriptors
```
```
root@zaius:~# ps|grep -e 802 -e 886
  802 messageb  16:16 /usr/bin/dbus-daemon --system --address=systemd: --nofork --nopidfile --systemd-activation
  886 root       3:03 /usr/sbin/ipmid
 9099 root       0:00 grep -e 802 -e 886
```

williamspatrick, 2017-04-12:
We will treat this as the continuation of #875.

williamspatrick, 2017-04-18:
Fix available in Gerrit.  https://gerrit.openbmc-project.xyz/3746

closed 2017-04-18 by williamspatrick