openbmc/openbmc #1433: [Palmetto] BMC failed to boot to Standby

swe12345, 2017-04-05:
We are again seeing issue(#1417) of BMC not able to boot to Standby. Tested with latest build v1.99.3-208-g4f10118. As of now seeing this with Palmetto system.

```
root@palmetto:~# uptime
 18:59:29 up 35 min,  load average: 0.90, 0.92, 0.91
```
```
root@palmetto:~# cat /etc/timestamp
20170404212632
```
```
root@palmetto:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.NotReady
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.Off
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Off
```
```
root@palmetto:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.3-208"
VERSION_ID="v1.99.3-208-g4f10118"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.3-208"
BUILD_ID="v1.99.3"
root@palmetto:~#
```
[dmesg_5apr.txt](https://github.com/openbmc/openbmc/files/895431/dmesg_5apr.txt)
[journal_5apr.txt](https://github.com/openbmc/openbmc/files/895446/journal_5apr.txt)

sivassrr, 2017-04-05:
Mar 29 19:55:17 palmetto settings_manager.py[3785]: ValueError: Invalid input. Data not in allowed values
Mar 29 19:55:17 palmetto systemd[1]: org.openbmc.settings.Host.service: Main process exited, code=exited, status=1/FAILURE
Mar 29 19:55:17 palmetto systemd[1]: Failed to start Phosphor Settings Manager.
Mar 29 19:55:17 palmetto systemd[1]: org.openbmc.settings.Host.service: Unit entered failed state.
Mar 29 19:55:17 palmetto systemd[1]: org.openbmc.settings.Host.service: Failed with result 'exit-code'.
Mar 29 19:55:17 palmetto systemd[1]: org.openbmc.settings.Host.service: Service hold-off

@swe12345 could you pls collect systemctl status output as well.

rahulmah, 2017-04-05:
This problem is also seen with Barreleye. On Witherspoon, we have not seen the issue at present.

vishwabmc, 2017-04-05:
there must be a full exception .. could u post that here ?

swe12345, 2017-04-05:
[systemctl_5apr.txt](https://github.com/openbmc/openbmc/files/895695/systemctl_5apr.txt)

swe12345, 2017-04-05:
[systemctl_5apr.txt](https://github.com/openbmc/openbmc/files/895700/systemctl_5apr.txt)

rahulmah, 2017-04-05:
After downgrading Barreleye system to build 678/v1.99.3-203-g7a0b268, we are not seeing this issue. So seem to a issue introduce after this build.

williamspatrick, 2017-04-05:
@ratagupt - I think this is due to the change you did for IPMI's boot parameters.  We need to remove a file out of /var/lib/obmc/, correct?  I think it is the host0-boot_flags file needs to be deleted.

sivassrr, 2017-04-25:
@swe12345 Today's issue looks like dup of this.

williamspatrick, 2017-04-25:
This issue was fixed. Please open a different one if you are seeing a similar symptom.

closed 2017-04-25 by williamspatrick