openbmc/openbmc #1436: Build warning: The postinstall intercept hook 'update_gio_module_cache' failed (exit code: 127)!

GHF, 2017-04-05:
I got this warning from the do_rootfs task for obmc-phosphor-image:

```
NOTE: Running intercept scripts:                                                    
NOTE: > Executing update_gio_module_cache intercept ...                             
FATAL: kernel too old                                                               
WARNING: The postinstall intercept hook 'update_gio_module_cache' failed (exit code: 127)! See log for details!
WARNING: The postinstalls for the following packages will be postponed for first boot: libglib-2.0-0
NOTE:   * postponing libglib-2.0-0                                               
NOTE: /usr/local/google/home/xow/obmc/build_z/tmp/sysroots/x86_64-linux/usr/bin/rpm -q --scripts --root /usr/local/google/home/xow/obmc/build_z/tmp/work/zaius-openbmc-linux-gnueabi/obmc-phosphor-image/1.0-r0/rootfs --dbpath=/var/lib/rpm libglib-2.0-0 | sed -n -e "/^postinstall scriptlet (using .*):$/,/^.* scriptlet (using .*):$/ {/.*/p}" | sed -e "/postinstall scriptlet (using \(.*\)):$/d" -e "/^.* scriptlet (using .*):$/d" > /usr/local/google/home/xow/obmc/build_z/tmp/work/zaius-openbmc-linux-gnueabi/obmc-phosphor-image/1.0-r0/rootfs/etc/rpm-postinsts/libglib-2.0-0
```

williamspatrick, 2017-04-05:
Can you let us know which kernel your build machine is running?  I put in a change to allow building with the RHEL6 kernel. I'm surprised yours would be older than that.

GHF, 2017-04-05:
Oh I didn't even realize that it was complaining about my build machine's kernel version.

```
uname -rv
3.13.0-108-generic #155-Ubuntu SMP Wed Jan 11 16:58:52 UTC 2017
```

GHF, 2017-04-18:
My workstation is now running a kernel from Ubuntu Xenial and I still get the error. See `tmp/work/zaius-openbmc-linux-gnueabi/obmc-phosphor-image/1.0-r0/temp/log.do_rootfs`
```
xow@pewter ~/o/build_z> uname -r
4.4.0-72-generic
```
The bug doesn't seem to affect any functionality.

gtmills, 2018-05-31:
@GHF No update in a year. Can this be closed? or still an issue?

closed 2018-06-01 by GHF

hitekumar, 2018-12-17:
I still see this issue 
ERROR: xxxxxx do_rootfs: The following packages could not be configured offline and rootfs is read-only: ['libglib-2.0-0']
ERROR: xxxxxx-1.0-r0 do_rootfs: Function failed: do_rootfs

jarus0, 2020-12-09:
Getting same error

```
NOTE: Executing Tasks
ERROR: core-image-minimal-1.0-r0 do_rootfs: The postinstall intercept hook 'update_gio_module_cache' failed, details in /home/dnk025/my_dir/poky/build-microchip/tmp/work/sama5d3_xplained_sd-poky-linux-gnueabi/core-image-minimal/1.0-r0/temp/log.do_rootfs
ERROR: Logfile of failure stored in: /home/dnk025/my_dir/poky/build-microchip/tmp/work/sama5d3_xplained_sd-poky-linux-gnueabi/core-image-minimal/1.0-r0/temp/log.do_rootfs.13741
ERROR: Task (/home/dnk025/my_dir/poky/meta/recipes-core/images/core-image-minimal.bb:do_rootfs) failed with exit code '1'
NOTE: Tasks Summary: Attempted 3266 tasks of which 3249 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/dnk025/my_dir/poky/meta/recipes-core/images/core-image-minimal.bb:do_rootfs
Summary: There was 1 ERROR message shown, returning a non-zero exit code.
```

QM1412, 2023-12-04:
did you solve it?i also have this problem.could you help me?

ERROR: petalinux-user-image-1.0-r0 do_rootfs: The postinstall intercept hook 'update_gio_module_cache' failed, details in /home/qm1412/petalinux/petalinux_sys/Petalinux_qspi_flash/build/tmp/work/plnx_zynq7-xilinx-linux-gnueabi/petalinux-user-image/1.0-r0/temp/log.do_rootfs
ERROR: petalinux-user-image-1.0-r0 do_rootfs: Function failed: do_rootfs
ERROR: Logfile of failure stored in: /home/qm1412/petalinux/petalinux_sys/Petalinux_qspi_flash/build/tmp/work/plnx_zynq7-xilinx-linux-gnueabi/petalinux-user-image/1.0-r0/temp/log.do_rootfs.114374
ERROR: Task (/home/qm1412/petalinux/petalinux_sys/Petalinux_qspi_flash/project-spec/meta-plnx-generated/recipes-core/images/petalinux-user-image.bb:do_rootfs) failed with exit code '1'