openbmc/openbmc #1438: Phosphor-Dbus-Interface:- Introduce new key named creatable.

ratagupt, 2017-04-06:
Introduce the key "creatable" with value "true" in the interfaces which can be dynamically creatable.
1) EthernetInterface
2) IPaddress
3) Dump.

ratagupt, 2017-04-06:
Other interfaces could be 
4) Vlan
5) Network Bond
6) Tunnel.

bradbishop, 2017-06-29:
@ratagupt @dkodihal what exactly does this enable?  Creating an IP address seems to suggest this is required to set an IP?  I'm assuming we aren't waiting for Phase 6 for that?

ratagupt, 2017-07-06:
@bradbishop : we need to support in sdbusplus as we discussed,Any D-Bus interface having key named creatable,sdbus plus will automatically create constructor like function.
Currently we have achieved this functionality by manually creating the creatable interface in local repo.
eg is
https://github.com/openbmc/phosphor-networkd/blob/master/xyz/openbmc_project/Network/IP/Create.interface.yaml

closed 2019-10-27 by stale[bot]