openbmc/openbmc #1440: mboxd: Create pnor partitions

anoo1, 2017-04-06:


amboar, 2017-04-10:
> When copying files to memory, use mmap.

Can you expand on this? We can mmap the physical memory needed by the LPC bus into the process, and we can mmap the file from the PNOR, but we still need to copy between the two. We can't redirect LPC FW cycle accesses directly to the file on flash if that's what you were hoping for.

williamspatrick, 2017-04-20:
@amboar - Yeah the recommendation for mmap was so that we can do memcpy between the reserved memory region and the files.  Not that we were magically going to be able to mmap files into the reserved region.

Now that I think of it, this might be a good use of the 'splice' system call instead of 'mmap'.  We can probably splice the file into the reserved region and let the kernel do all the copying operation.

The reason for mmap vs read is just to let the VMM/VFS cache do their thing rather than having even more blobs of memory sitting around in the mailbox process.

amboar, 2017-04-21:
> @amboar - Yeah the recommendation for mmap was so that we can do memcpy between the reserved memory region and the files. Not that we were magically going to be able to mmap files into the reserved region.

Right, thanks for clarifying.

> Now that I think of it, this might be a good use of the 'splice' system call instead of 'mmap'. We can probably splice the file into the reserved region and let the kernel do all the copying operation.

We would need to hook up support in the aspeed-lpc-ctrl driver. Currently it doesn't populate the `splice_read()` or `splice_write()` callbacks, and neither does `drivers/mtd/mtdchar.c` (I think that's the right place?). There are generic implementations of the callbacks that will probably work, but this needs more investigation. We also still have the issue of doing this in "small" chunks so as not to stall all other processes. But I agree in general, I think it's feasible to reimplement `copy_flash()` and `write_flash()` in terms of `splice()`. It would also require some tinkering with the test cases to implement the libc wrapper for `splice()`.

> The reason for mmap vs read is just to let the VMM/VFS cache do their thing rather than having even more blobs of memory sitting around in the mailbox process.

Yep.

williamspatrick, 2017-04-21:
Memcpy via mmap solves the small chunk problem, right?  Maybe we just stick with that over splice for now. 

Why do we have this performance issue?  Do we not have full preemption enabled?

rfrandse, 2017-05-05:
https://gerrit.openbmc-project.xyz/#/c/4087/

mdmillerii, 2017-05-05:
I started this reply some time ago:

> Why do we have this performance issue? Do we not have full preemption enabled?

The mtd layer is implemented via synchronous calls, except for erase.  The driver has a yield in the wait loop but fundamentally will only issue one thread at a time.  This also prevents the SPI layer from interrupting a write to perform a read (erase is also not interruptible in the current code) even though the flash chips support it.  The mtd layer actually has better support for directly addressed windows than asynchronous transfers (eg the offload engine).

amboar, 2017-05-08:
Sorry, had been meaning to leave a reply along the lines of @mdmillerii's, but never got around to it.

closed 2017-06-29 by williamspatrick