openbmc/openbmc #1441: [Barreleye latest] Failed building obmc-phosphor-image due to phosphor-ipmi-fru compiling fails

yadron, 2017-04-06:
For me it looks like gcc updated from 5.3.0 to 6.2.0 and as result failed after that:

```
ERROR: phosphor-ipmi-fru-1.0-r1 do_compile: oe_runmake failed
ERROR: phosphor-ipmi-fru-1.0-r1 do_compile: Function failed: do_compile (log file is located at /home/build/openbmc3/openbmc/build/tmp/work/armv5e-openbmc-linux-gnueabi/phosphor-ipmi-fru/1.0-r1/temp/log.do_compile.24703)
ERROR: Logfile of failure stored in: /home/build/openbmc3/openbmc/build/tmp/work/armv5e-openbmc-linux-gnueabi/phosphor-ipmi-fru/1.0-r1/temp/log.do_compile.24703
```

Looks like something fails during do_configure checking:

cat temp/log.do_configure.12795
```
checking whether the C compiler works... no
configure: error: in `/home/build/openbmc3/openbmc/build/tmp/work/armv5e-openbmc-linux-gnueabi/phosphor-ipmi-bt/1.0-r1/build':
configure: error: C compiler cannot create executables
```

_(see openbmc/build/tmp/work/armv5e-openbmc-linux-gnueabi/phosphor-ipmi-fru/1.0-r1/build during building)_

Probably this is simple to check, so I'll look more into later and write here then.

mine260309, 2017-04-25:
MSI building P8 machine hit the same issue, and I confirm it can be workaround by increasing the RAM, either physical RAM or swap space would help.
On MSI's case, it is consuming more than 20GiB RAM building it.

So this **is** the same issue caused by GCC6: https://gcc.gnu.org/bugzilla/show_bug.cgi?id=80290

rfrandse, 2017-06-02:
<https://gerrit.openbmc-project.xyz/4544> fru_gen: reduce map in generated data
Resolves  #1441 [Barreleye latest] Failed building obmc-phosphor-image due to phosphor-ipmi-fru compiling fails

closed 2017-06-02 by williamspatrick