openbmc/openbmc #1443: [BarrelEye v1.99.3-110] Updating at BMC console BMC flash fails during "Verifying kb"

yadron, 2017-04-07:
Updating at BMC console BMC flash fails during "Verifying kb", probably because of CPU-busy.

Looks like:

1. Upload image-bmc to BMC /run/initramfs/

2. Umount /dev/mtdblock4, /dev/mtdblock5

3. Run ./update

4. Probably you'll get "Verifying kb: 31xxx/32768 (xx%) failed" (at least my system uptime == 3 days and it is so).

5. Try more attempts of ./update, but the same problem.

6. I know that before it happened if CPU busy. So killing something CPU-hungry (in my case killed hwmon.py with commands: "top; cp /usr/sbin/hwmon.py /usr/sbin/hwmon.py_; rm /usr/sbin/hwmon.py; kill 782).

7. Run ./update again and this time successful.

I've already written regarding this issue before (see https://github.com/openbmc/openbmc/issues/1262 with the same and https://github.com/openbmc/openbmc/issues/1287 with other "problem reading flash"), and now I see that it is still actual.

yadron, 2017-04-07:
The version is: VERSION_ID="v1.99.3-208-g4f10118".

First I booted BMC, but cannot boot HOST. I've updated .pnor to latest
https://openpower.xyz/job/openpower-op-build/lastStableBuild/distro=ubuntu,target=barreleye/
, but this didn't help.

Then I've noted that for some reason, after installation of latest BMC image (v1.99.3-208), the "/usr/sbin/phosphor-mapper" missed. Errors looks like these:

```
journalctl |tail -n 50
Apr 07 11:52:27 barreleye systemd[1]: Starting Phoshpor System Manager...
Apr 07 11:52:28 barreleye settings_manager.py[1316]: Traceback (most recent call last):
Apr 07 11:52:28 barreleye settings_manager.py[1316]:   File "/usr/sbin/settings_manager.py", line 251, in <module>
Apr 07 11:52:28 barreleye settings_manager.py[1316]:     allobjects = create_object(s.SETTINGS)
Apr 07 11:52:28 barreleye settings_manager.py[1316]:   File "/usr/sbin/settings_manager.py", line 45, in create_object
Apr 07 11:52:28 barreleye settings_manager.py[1316]:     mapper = obmc.mapper.Mapper(bus)
Apr 07 11:52:28 barreleye settings_manager.py[1316]:   File "/usr/lib/python2.7/site-packages/obmc/mapper/bindings.py", line 32, in __init__
Apr 07 11:52:28 barreleye settings_manager.py[1316]:     obj = bus.get_object(MAPPER_NAME, MAPPER_PATH, introspect=False)
Apr 07 11:52:28 barreleye settings_manager.py[1316]:   File "/usr/lib/python2.7/site-packages/dbus/bus.py", line 241, in get_object
Apr 07 11:52:28 barreleye settings_manager.py[1316]:     follow_name_owner_changes=follow_name_owner_changes)
Apr 07 11:52:28 barreleye settings_manager.py[1316]:   File "/usr/lib/python2.7/site-packages/dbus/proxies.py", line 248, in __init__
Apr 07 11:52:28 barreleye settings_manager.py[1316]:     self._named_service = conn.activate_name_owner(bus_name)
Apr 07 11:52:28 barreleye settings_manager.py[1316]:   File "/usr/lib/python2.7/site-packages/dbus/bus.py", line 180, in activate_name_owner
Apr 07 11:52:28 barreleye settings_manager.py[1316]:     self.start_service_by_name(bus_name)
Apr 07 11:52:28 barreleye settings_manager.py[1316]:   File "/usr/lib/python2.7/site-packages/dbus/bus.py", line 278, in start_service_by_name
Apr 07 11:52:28 barreleye settings_manager.py[1316]:     'su', (bus_name, flags)))
Apr 07 11:52:28 barreleye settings_manager.py[1316]:   File "/usr/lib/python2.7/site-packages/dbus/connection.py", line 651, in call_blocking
Apr 07 11:52:28 barreleye settings_manager.py[1316]:     message, timeout)
Apr 07 11:52:28 barreleye settings_manager.py[1316]: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name xyz.openbmc_project.ObjectMapper was not provided by any .service files
Apr 07 11:52:28 barreleye systemd[1]: org.openbmc.settings.Host.service: Main process exited, code=exited, status=1/FAILURE
Apr 07 11:52:28 barreleye systemd[1]: Failed to start Phosphor Settings Manager.
Apr 07 11:52:28 barreleye systemd[1]: org.openbmc.settings.Host.service: Unit entered failed state.
Apr 07 11:52:28 barreleye systemd[1]: org.openbmc.settings.Host.service: Failed with result 'exit-code'.
Apr 07 11:52:29 barreleye mapper[1329]: Error starting event loop: No route to host
Apr 07 11:52:29 barreleye systemd[1]: Starting Wait for /org/openbmc/settings/host0...
Apr 07 11:52:29 barreleye systemd[1]: mapper-wait@-org-openbmc-settings-host0.service: Main process exited, code=exited, status=1/FAILURE
Apr 07 11:52:29 barreleye systemd[1]: Failed to start Wait for /org/openbmc/settings/host0.
Apr 07 11:52:29 barreleye systemd[1]: mapper-wait@-org-openbmc-settings-host0.service: Unit entered failed state.
Apr 07 11:52:29 barreleye systemd[1]: mapper-wait@-org-openbmc-settings-host0.service: Failed with result 'exit-code'.

```

I've added it manually:
scp ./openbmc/build/tmp/work/barreleye-openbmc-linux-gnueabi/obmc-phosphor-image/1.0-r0/rootfs/usr/sbin/phosphor-mapper root@172.17.27.113:/usr/sbin/phosphor-mapper

And more errors like these are:

```
Apr 07 12:12:29 barreleye systemd[1]: obmc-read-eeprom@system-chassis-io_board.service: Failed with result 'core-dump'.

Apr 07 12:12:34 barreleye sync_inventory_items.py[1110]: Traceback (most recent call last):
Apr 07 12:12:34 barreleye sync_inventory_items.py[1110]:   File "/usr/sbin/sync_inventory_items.py", line 145, in <module>
Apr 07 12:12:34 barreleye sync_inventory_items.py[1110]:     inv_value = get_inv_value(inv_obj, prop_name)
Apr 07 12:12:34 barreleye sync_inventory_items.py[1110]:   File "/usr/sbin/sync_inventory_items.py", line 56, in get_inv_value
Apr 07 12:12:34 barreleye sync_inventory_items.py[1110]:     value = dbus_method(INV_INTF_NAME, prop_name)
Apr 07 12:12:34 barreleye sync_inventory_items.py[1110]:   File "/usr/lib/python2.7/site-packages/dbus/proxies.py", line 70, in __call__
Apr 07 12:12:34 barreleye sync_inventory_items.py[1110]:     return self._proxy_method(*args, **keywords)
Apr 07 12:12:34 barreleye sync_inventory_items.py[1110]:   File "/usr/lib/python2.7/site-packages/dbus/proxies.py", line 145, in __call__
Apr 07 12:12:34 barreleye sync_inventory_items.py[1110]:     **keywords)
Apr 07 12:12:34 barreleye sync_inventory_items.py[1110]:   File "/usr/lib/python2.7/site-packages/dbus/connection.py", line 651, in call_blocking
Apr 07 12:12:34 barreleye sync_inventory_items.py[1110]:     message, timeout)
Apr 07 12:12:34 barreleye sync_inventory_items.py[1110]: dbus.exceptions.DBusException: org.freedesktop.DBus.Python.dbus.exceptions.DBusException: org.freedesktop.UnknownProperty: Custom Field 1
```

[... cut ...]
.[0;1;31m●.[0m .[0;1;31mobmc-sync-inventory-item@system-chassis-io_board.service .[0m loaded .[0;1;31mfailed failed   .[0m Sync system/chass
is/io_board
.[0;1;31m●.[0m .[0;1;31mobmc-sync-inventory-item@system-chassis-motherboard.service.[0m loaded .[0;1;31mfailed failed   .[0m Sync system/cha
ssis/motherboard

```
Since the OpenBMC from my point of view such a broken from the beginning of March, maybe it is better taking the sources of 28 Feb and adding one-by-one changes for understanding what was the reasons of all these problems?

williamspatrick, 2017-04-18:
If you are manually unmounting filesystems, this is not the correct update procedure.

closed 2017-04-18 by williamspatrick