openbmc/openbmc #1444: Create inotify daemon

dkodihal, 2017-04-07:
Create inotify daemon to notify about file additions.

One known use is the notifying the code update image managers about an image being downloaded.

dkodihal, 2017-04-07:
Assuming a generic inotify daemon.

- [ ] Write xyz.openbmc_project.Common.FileAdded interface. This should emit a d-bus signal to indicate FileAdded.
- [ ] inotify daemon : have the app accept via command-line a list of pathnames to monitor.
- [ ] inotify daemon : implement xyz.openbmc_project.Common.FileAdded interface

vishwabmc, 2017-04-07:
@dkodihal Are we planning to provide all the granularity that the inotify(7) gives ?. I imply that by 'generic' inotify daemon.

dkodihal, 2017-04-07:
- [ ] add inotify watch on a specified location to know about software images being downloaded.
- [ ] verify the image is a tar archive and has a valid MANIFEST file.

amboar, 2017-04-10:
Is this because we have two routes for uploading an image? REST and TFTP? Why can't the respective code issue its own DBUS signals in this regard? Are we planning on having N codebases handling uploads, where N is some significant number (i.e. more than 2)? Also, if we have an inotify daemon, why should it be validating the uploaded files? Isn't that best left to code that actually consumes the artifacts?

dkodihal, 2017-04-10:
Is this because we have two routes for uploading an image? REST and TFTP?
- Yes.

Why can't the respective code issue its own DBUS signals in this regard?
- Not sure I followed. I will be placing the inotify daemon in the phosphor code-update repo.

Are we planning on having N codebases handling uploads, where N is some significant number (i.e. more than 2)?
- As of now, it's the code-update codebase.

Also, if we have an inotify daemon, why should it be validating the uploaded files? Isn't that best left to code that actually consumes the artifacts?
- Right, that's how it's going to be. My tasks were sort of WIP since we were discussing whether the code that consumes this is also being done via this story. Now I know that there's another story for that, I have the tasks updated.

amboar, 2017-04-10:
@dkodihal As an aside, you can "quote" things with the ">" (like email does for replies).

> > Why can't the respective code issue its own DBUS signals in this regard?

> Not sure I followed. I will be placing the inotify daemon in the phosphor code-update repo.

Is it the case that the inotify daemon will notify the code update mechanism via d-bus? If so, I imagine it would do so via a d-bus signal and if it is done that way, is there any reason we can't have the REST and TFTP servers issue the signal, and eliminate the inotify daemon?

Maybe the notification isn't done via dbus, but the approach isn't clear to me.

dkodihal, 2017-04-10:
> Is it the case that the inotify daemon will notify the code update mechanism via d-bus?

The inotify daemon is within the same phosphor application that's handling code updates (wasn't intended to be a generic inotify daemon for the BMC), so it's just going to be a method call.

dkodihal, 2017-04-10:
> @dkodihal As an aside, you can "quote" things with the ">" (like email does for replies).

Thanks! That helps.

amboar, 2017-04-10:
> I will be placing the inotify daemon in the phosphor code-update repo.

By this do you mean https://github.com/openbmc/phosphor-bmc-code-mgmt/ ?

> The inotify daemon is within the same phosphor application that's handling code updates (wasn't intended to be a generic inotify daemon for the BMC), so it's just going to be a method call.

Oh, so this isn't actually a daemon? Just implementing inotify support in phosphor-bmc-code-mgmt?

dkodihal, 2017-04-10:
> > I will be placing the inotify daemon in the phosphor code-update repo.

> By this do you mean https://github.com/openbmc/phosphor-bmc-code-mgmt/ ?

Yep.

> > The inotify daemon is within the same phosphor application that's handling code updates (wasn't intended to be a generic inotify daemon for the BMC), so it's just going to be a method call.

> Oh, so this isn't actually a daemon? Just implementing inotify support in phosphor-bmc-code-mgmt?

Right, this will be within the  phosphor-bmc-code-mgmt process that handles image updates.

amboar, 2017-04-11:
I think we should change the title of the issue in that case to avoid confusion.

closed 2017-04-19 by williamspatrick