openbmc/openbmc #1447: [Enhancement] wget to access https site

swe12345, 2017-04-10:
Referring to https://github.com/openbmc/openbmc-test-automation#467

@mdmillerii  - Based on this link http://askubuntu.com/questions/338869/error-while-downloading-https-files-using-wget adding --secure-protocol option might help https to work

```
root@palmetto:~# wget --secure-protocol=protocol https://openpower.xyz
wget: unrecognized option '--secure-protocol=protocol'
BusyBox v1.24.1 (2017-04-05 12:03:34 CDT) multi-call binary.

Usage: wget [-c|--continue] [-s|--spider] [-q|--quiet] [-O|--output-document FILE]
        [--header 'header: value'] [-Y|--proxy on/off] [-P DIR]
        [-U|--user-agent AGENT] [-T SEC] URL...
```

williamspatrick, 2017-04-10:
Busybox wget does not have that option and the "real" wget is much bigger. What is it you are trying to accomplish?

sivassrr, 2017-04-10:
@williamspatrick we would like to get the BMC / PNOR image directly from https://openpower.xyz/ and onto BMC

rfrandse, 2017-04-14:
Contact Deepak, he is implementing rest interface to upload image to bmc.

sivassrr, 2017-04-17:
@rfrandse, @dkodihal is using rest interface to upload the image to BMC. This enhancement is to get image using wget from external world.

dkodihal, 2017-04-17:
Why does the external world have to login to the BMC and use wget, when REST is their interface?

rfrandse, 2017-04-26:
Use rest api

closed 2017-04-26 by rfrandse

williamspatrick, 2017-04-26:
We have no plans to use anything other than busybox wget and will not be enhancing busybox wget to support a "real" wget parameter like this.  Please find some other mechanism to get images onto the BMC in your environment.