openbmc/openbmc #1448: [Palmetto/Barreleye] BMC is going into NotReady State after code update

swe12345, 2017-04-10:
BMC is going into NotReady State after code update
with the latest build. Seeing this issue on Palmetto and Barreleye.
```
root@palmetto:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.4-14"
VERSION_ID="v1.99.4-14-ga450877"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.4-14"
BUILD_ID="v1.99.4"
```
```
root@palmetto:~# cat /etc/timestamp
20170409122441
```
```
root@palmetto:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.NotReady
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.Off
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Off
```
[journal_10apr.txt](https://github.com/openbmc/openbmc/files/909619/journal_10apr.txt)
[dmesg_10apr.txt](https://github.com/openbmc/openbmc/files/909622/dmesg_10apr.txt)

ratagupt, 2017-04-10:
Mar 30 00:56:56 palmetto systemd[1]: op-reset-pgood-check@0.service: Main process exited, code=exited, status=1/FAILURE
Mar 30 00:56:56 palmetto systemd[1]: Failed to start Check Chassis0 pgood on reset.
Mar 30 00:56:56 palmetto systemd[1]: Dependency failed for Chassis0 (Reset Check).
Mar 30 00:56:56 palmetto systemd[1]: obmc-chassis-reset@0.target: Job obmc-chassis-reset@0.target/start failed with result 'dependency'.
Mar 30 00:56:56 palmetto systemd[1]: Dependency failed for Create file to indicate chassis0 pgood is on.
Mar 30 00:56:56 palmetto systemd[1]: Dependency failed for Start chassis0 on after BMC reset.
Mar 30 00:56:56 palmetto systemd[1]: op-reset-chassis-on@0.service: Job op-reset-chassis-on@0.service/start failed with result 'dependency'.
Mar 30 00:56:56 palmetto systemd[1]: op-reset-set-power-on@0.service: Job op-reset-set-power-on@0.service/start failed with result 'dependency'.
Mar 30 00:56:56 palmetto systemd[1]: op-reset-pgood-check@0.service: Unit entered failed state.
Mar 30 00:56:56 palmetto systemd[1]: op-reset-pgood-check@0.service: Failed with result 'exit-code'.
Mar 30 00:56:56 palmetto systemd[1]: Reached target Chassis0 power on after reset.

seems op-reset-pgood-check service is faling.
@dhruvibm :- Can you please take a look?

geissonator, 2017-04-10:
The failing of the op-reset-pgood-check service is expected so that's not the issue here.  It just means pgood was not on prior to the BMC reset and the service should not continue.

geissonator, 2017-04-10:
Usually in these situations you can do a "ps -ef | grep mapper" and see what service is hung up.  Otherwise you need to walk through the services for obmc-standby and do "systemctl status <service>" on them to determine which did not start.

swe12345, 2017-04-10:
[systemctl_10apr.txt](https://github.com/openbmc/openbmc/files/910126/systemctl_10apr.txt)

williamspatrick, 2017-04-10:
I prefer to use `systemctl list-running-jobs` to determine what is blocking state.

swe12345, 2017-04-11:
[systemctl_listjobs.txt](https://github.com/openbmc/openbmc/files/912231/systemctl_listjobs.txt)

williamspatrick, 2017-04-11:
`120 mapper-wait@-xyz-openbmc_project-control-host0.service start running`

@geissonator - I don't think this was ever a dbus object. Was this a copy paste error somewhere?

geissonator, 2017-04-11:
Ooops, that's my new dbus object coming in with a different patch set (that's not merged yet).  I was thinking not running this target would be fine until it was available but guess it's messing with obmc-standby.

geissonator, 2017-04-11:
https://gerrit.openbmc-project.xyz/#/c/3579/

closed 2017-04-11 by williamspatrick