openbmc/openbmc #1449: REST commands getting 415 (HTTP_UNSUPPORTED_MEDIA_TYPE) vs 200 (HTTP_OK)

micwalsh, 2017-04-10:
On system w51, REST commands are getting 415 (HTTP_UNSUPPORTED_MEDIA_TYPE) instead of the usual 200 (HTTP_OK).

williamspatrick, 2017-04-10:
What tool are you using?  How are you using it?  There really isn't enough information to go off of here.

micwalsh, 2017-04-10:
Andrew asked me to look at this failure: http://tulpok01p4.aus.stglabs.ibm.com:8080/jenkins/job/OpenBMC/job/automated-test/job/helper-jobs/job/AES-Standby/2079/robot/report/log.html#s1-t1

Drilling down to the actual error, the following robot keyword failed:
OpenBMC Get Request from lib/rest_client.robot

It failed because the resp.status_code should be 200 rather than 415.

I ran some REST commands manually on the system and got the same failure.

Adriana Kobylak had a look and said the following:

We have all the GET cmds but there’s no trace of what the response is sent:

Mar 29 19:33:30 witherspoon phosphor-gevent[905]: ::4039:bc76:8e91:f402%1300540188 - - [2017-03-29 19:33:30] "GET /xyz/openbmc_project/state/bmc0/ HTTP/1.1" 200 167 0.045611
Mar 29 19:33:34 witherspoon phosphor-gevent[905]: ::4039:bc76:8e91:f402%1300540188 - - [2017-03-29 19:33:34] "GET /xyz/openbmc_project/state/bmc0 HTTP/1.1" 200 321 0.058787
Mar 29 19:33:39 witherspoon phosphor-gevent[905]: ::4039:bc76:8e91:f402%1300540188 - - [2017-03-29 19:33:39] "GET /xyz/openbmc_project/state/bmc0/CurrentBMCState HTTP/1.1" 404 324 0.031872
Mar 29 19:33:42 witherspoon phosphor-gevent[905]: ::4039:bc76:8e91:f402%1300540188 - - [2017-03-29 19:33:42] "GET /xyz/openbmc_project/state/bmc0/CurrentBMCState/ HTTP/1.1" 404 324 0.030922

geissonator, 2017-04-21:
Hit this on a CI job today:
```
Apr 21 01:42:28 witherspoon phosphor-gevent[793]: ::60a9:a576:8e91:f402%1285925660 - - [2017-04-21 01:42:28] "POST /login HTTP/1.1" 200 320 0.033844
Apr 21 01:42:28 witherspoon phosphor-gevent[793]: ::60a9:a576:8e91:f402%1285925660 - - [2017-04-21 01:42:28] "POST /login HTTP/1.1" 200 320 0.034275
Apr 21 01:42:28 witherspoon phosphor-gevent[793]: ::60a9:a576:8e91:f402%1285925660 - - [2017-04-21 01:42:28] "GET /xyz/openbmc_project/state/bmc0//attr/CurrentBMCState HTTP/1.1" 415 285 0.055219
Apr 21 01:42:28 witherspoon phosphor-gevent[793]: ::60a9:a576:8e91:f402%1285925660 - - [2017-04-21 01:42:28] "POST /login HTTP/1.1" 200 320 0.033475
Apr 21 01:42:28 witherspoon phosphor-gevent[793]: ::60a9:a576:8e91:f402%1285925660 - - [2017-04-21 01:42:28] "GET /xyz/openbmc_project/state/bmc0//attr/CurrentBMCState HTTP/1.1" 415 285 0.060613
Apr 21 01:42:29 witherspoon dropbear[878]: Exit (root): Exited normally
```

```
Get To Stable State :: BMC cleanup drive to stable state              URI:https://IP:443/xyz/openbmc_project/state/bmc0//attr/CurrentBMCState, method:Get, args:{}
Response code:415, Content:{
  "data": {
    "description": "Expecting content type 'application/json', got ''"
  }, 
  "message": "415 Unsupported Media Type", 
  "status": "error"
}
URI:https://IP:443/xyz/openbmc_project/state/bmc0//attr/CurrentBMCState, method:Get, args:{}
Response code:415, Content:{
  "data": {
    "description": "Expecting content type 'application/json', got ''"
  }, 
  "message": "415 Unsupported Media Type", 
  "status": "error"
}
| FAIL |
415 != 200
```

closed 2017-10-26 by micwalsh