openbmc/openbmc #1450: openpower-occ-control systemd service fails to start

dkodihal, 2017-04-11:
Mar 30 00:57:24 witherspoon systemd[1]: [[0;1;39morg.open_power.OCC.PassThrough.service: Main process exited, code=exited, status=255/n/a[[0m
Mar 30 00:57:24 witherspoon systemd[1]: [[0;1;31mFailed to start OpenPOWER OCC controller.[[0m
Mar 30 00:57:24 witherspoon systemd[1]: [[0;1;39morg.open_power.OCC.PassThrough.service: Unit entered failed state.[[0m

dkodihal, 2017-04-17:
The way we're fixing this now is to create OCC objects when CPU inventory objects get created (which will be notified via d-bus signals).

rfrandse, 2017-06-12:
<https://gerrit.openbmc-project.xyz/4127> Create OCC pass-through objects
Resolves  #1450 openpower-occ-control systemd service fails to start

closed 2017-06-14 by williamspatrick