openbmc/openbmc #1452: [BarrelEye v1.99.4-14] obmcutil state works incorrectly

yadron, 2017-04-11:
The command "obmcutil state" works strange. First:

```
root@barreleye:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.NotReady
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.Off
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Off
```

If "obmcutil poweron" command, then it still shows the same.

If restarting services *state* like

```
root@barreleye:~# ps|grep state
  762 root       0:00 /usr/sbin/netman_watch_dns /run/systemd/netif/state
  887 root       0:00 /usr/sbin/phosphor-host-state-manager
  888 root       0:00 /usr/sbin/phosphor-chassis-state-manager
  889 root       0:00 /usr/sbin/phosphor-bmc-state-manager
  982 root       0:00 grep state
root@barreleye:~# kill 887
root@barreleye:~# kill 888
root@barreleye:~# kill 889
```

, then it shows updated states:

```
root@barreleye:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.NotReady
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.On
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Running
```

- but still incorrectly, because host not booted.

I cannot boot host still (there were no transition to "power ON is complete" and  no call of org.openbmc.control.Host->boot() (so no output "Booting host, Using primary side of the bios flash" of /usr/sbin/control_host.exe)

I'm debugging what is the difference of latest version (that works strange) and older "28 feb: v1.99.3" that works Ok (and for older host always being booted without problems).

For older (when booting host successfully):

```
Apr 10 14:28:15 barreleye systemd[1]: Stopped target Chassis0 (Stop).
Apr 10 14:28:22 barreleye timemanager[872]: Changing time settings is *deferred* now
Apr 10 14:28:22 barreleye timemanager[872]: Changing time settings is *deferred* now
Apr 10 14:28:22 barreleye timemanager[872]: Changing time settings is *deferred* now
Apr 10 14:28:22 barreleye kernel: ftgmac100 1e660000.ethernet eth0: rx crc err
Apr 10 14:28:22 barreleye kernel: ftgmac100 1e660000.ethernet eth0: rx frame too long
Apr 10 14:28:22 barreleye kernel: ftgmac100 1e660000.ethernet eth0: rx crc err
Apr 10 14:28:22 barreleye kernel[691]: ftgmac100 1e660000.ethernet eth0: rx crc err
Apr 10 14:28:22 barreleye timemanager[872]:  User Settings have changed..
Apr 10 14:28:22 barreleye kernel[691]: ftgmac100 1e660000.ethernet eth0: rx frame too long
Apr 10 14:28:22 barreleye kernel[691]: ftgmac100 1e660000.ethernet eth0: rx crc err
Apr 10 14:28:22 barreleye phosphor-chassis-state-manager[814]: Recieved signal that power ON is complete
Apr 10 14:28:22 barreleye phosphor-chassis-state-manager[814]: Change to Chassis Power State
Apr 10 14:28:23 barreleye system_manager.py[746]: Running System State: HOST_POWERED_ON
Apr 10 14:28:24 barreleye systemd[1]: Started Turn On identify LED.
Apr 10 14:28:24 barreleye systemd[1]: Reached target Power0 On.
Apr 10 14:28:24 barreleye systemd[1]: Reached target Power0 (On).
Apr 10 14:28:24 barreleye systemd[1]: Starting PCIE Slot Presence Detection For Chassis0...
Apr 10 14:28:25 barreleye systemd[1]: Started PCIE Slot Presence Detection For Chassis0.
```

For new (cannot boot host):

```
root@barreleye:~# journalctl |grep POWER
Apr 10 13:50:20 barreleye systemd[1]: Starting OpenPOWER Host0 Control...
Apr 10 13:50:23 barreleye system_manager.py[746]: Power GPIO config: ['PGOOD', '', [('POWER_PIN', False)], [('USB_RESET', False)], [], 'FSI_DATA', 'FSI_CLK', 'FSI_ENABLE', 'CRONUS_SEL', [('BMC_THROTTLE', True), ('IDBTN', False)]]
[... 55 lines cut ...]
---
>         printf("Host Checkstop, going to quiesce\n");
85,87c85,87
<             "org.openbmc.control.Chassis", /* name */
<             "/org/openbmc/control/chassis0", /* object path */
<             "org.openbmc.control.Chassis", /* interface name */
---
>             "org.freedesktop.systemd1", /* name */
>             "/org/freedesktop/systemd1", /* object path */
>             "org.freedesktop.systemd1.Manager", /* interface name */
94,95c94,95
<             "reboot",
<             parm,
---
>             "StartUnit",
>             g_variant_new("(ss)", "obmc-quiesce-host@0.target", "replace"),
100a101
>         g_variant_unref(result);
203d203
<
```

- could it be changing names of dbus objects _(here org.openbmc.control.Chassis -> org.freedesktop.systemd1)_ be the reason of problems?

I checked carefully if something changed in power_control_obj.c, but looks like there were no significant changes of power-pin control itself.

geissonator, 2017-04-11:
#1448 ?  Could you test the latest release with that fix?

yadron, 2017-04-11:
I'll try latest release and tell here then (probably tomorrow).

yadron, 2017-04-12:
Ok, the state works now,

```
root@barreleye:~# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.4-15"
VERSION_ID="v1.99.4-15-g4105c81"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.4-15"
BUILD_ID="v1.99.4"
root@barreleye:~# 
root@barreleye:~# 
root@barreleye:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.On
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Off
root@barreleye:~# obmcutil power
pgood = 1
state = 1
pgood_timeout = 10
root@barreleye:~# 
root@barreleye:~# 
root@barreleye:~# obmcutil poweroff
root@barreleye:~# obmcutil power   
pgood = 0
state = 0
pgood_timeout = 10
root@barreleye:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.Off
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Off
root@barreleye:~# 
root@barreleye:~# 
root@barreleye:~# obmcutil poweron
root@barreleye:~# obmcutil state  
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.On
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Running
```

,but still cannot boot host:

```
root@barreleye:~# journalctl |egrep -i "POWER"
Apr 12 09:57:13 barreleye systemd[1]: Created slice system-org.openbmc.control.Power.slice.
Apr 12 09:57:13 barreleye systemd[1]: Created slice system-op\x2dreset\x2dset\x2dpower\x2don.slice.
Apr 12 09:57:13 barreleye systemd[1]: Created slice system-org.openbmc.buttons.Power.slice.
Apr 12 09:57:32 barreleye kernel[693]: systemd[1]: Created slice system-org.openbmc.control.Power.slice.
Apr 12 09:57:32 barreleye kernel[693]: systemd[1]: Created slice system-op\x2dreset\x2dset\x2dpower\x2don.slice.
Apr 12 09:57:32 barreleye kernel[693]: systemd[1]: Created slice system-org.openbmc.buttons.Power.slice.
Apr 12 09:58:22 barreleye systemd[1]: Starting Wait for /org/openbmc/control/power0...
Apr 12 09:59:23 barreleye systemd[1]: Starting OpenPOWER Host0 Control...
Apr 12 09:59:23 barreleye systemd[1]: Starting Phosphor Power0 Control...
Apr 12 09:59:23 barreleye systemd[1]: Starting Phosphor Power Button0...
Apr 12 09:59:26 barreleye system_manager.py[749]: New object: /org/openbmc/control/power0 (:1.71)
Apr 12 09:59:26 barreleye system_manager.py[749]: Power GPIO config: ['PGOOD', '', [('POWER_PIN', False)], [('USB_RESET', False)], [], 'FSI_DATA', 'FSI_CLK', 'FSI_ENABLE', 'CRONUS_SEL', [('BMC_THROTTLE', True), ('IDBTN', False)]]
Apr 12 09:59:26 barreleye power_control.exe[785]: Power GPIO latch output:
Apr 12 09:59:26 barreleye power_control.exe[785]: Power GPIO power good input: PGOOD
Apr 12 09:59:26 barreleye power_control.exe[785]: Power GPIO 1 power_up outputs
Apr 12 09:59:26 barreleye power_control.exe[785]: Power GPIO power_up[0] = POWER_PIN active LOW
Apr 12 09:59:26 barreleye power_control.exe[785]: Power GPIO 1 reset outputs
Apr 12 09:59:26 barreleye power_control.exe[785]: Power GPIO reset[0] = USB_RESET active LOW
Apr 12 09:59:26 barreleye power_control.exe[785]: Power GPIO 0 pci reset outputs
Apr 12 09:59:26 barreleye power_control.exe[785]: FSI DATA GPIO: FSI_DATA
Apr 12 09:59:26 barreleye power_control.exe[785]: FSI CLK GPIO: FSI_CLK
Apr 12 09:59:26 barreleye power_control.exe[785]: FSI ENABLE GPIO: FSI_ENABLE
Apr 12 09:59:26 barreleye power_control.exe[785]: CRONUS SEL GPIO: CRONUS_SEL
Apr 12 09:59:26 barreleye power_control.exe[785]: Hostctl GPIO optionals: 2
Apr 12 09:59:26 barreleye power_control.exe[785]: Hostctl optional GPIO[0] = BMC_THROTTLE active HIGH
Apr 12 09:59:26 barreleye power_control.exe[785]: Hostctl optional GPIO[1] = IDBTN active LOW
[... 82 lines cut ...]
Apr 12 10:43:24 barreleye systemd[1]: Reached target Power0 Off.
Apr 12 10:43:24 barreleye systemd[1]: Reached target Chassis0 (Power Off).
Apr 12 10:43:24 barreleye phosphor-chassis-state-manager[791]: Recieved signal that power OFF is complete
Apr 12 10:43:24 barreleye phosphor-chassis-state-manager[791]: Change to Chassis Power State
Apr 12 10:43:24 barreleye systemd[1]: Starting Start Power0...
Apr 12 10:43:24 barreleye systemd[1]: Stopped target Chassis0 (Power Off).
Apr 12 10:43:24 barreleye systemd[1]: Stopped target Power0 Off.
Apr 12 10:43:24 barreleye systemd[1]: Stopped Stop Power0.
Apr 12 10:43:24 barreleye systemd[1]: Stopped Wait for Power0 to turn off.
Apr 12 10:43:24 barreleye systemd[1]: Starting Wait for Power0 to turn on...
Apr 12 10:43:24 barreleye phosphor-chassis-state-manager[791]: Recieved signal that power OFF is complete
Apr 12 10:43:24 barreleye phosphor-chassis-state-manager[791]: Change to Chassis Power State
Apr 12 10:43:27 barreleye power_control.exe[785]: PowerControl: setting power up POWER_PIN to 0
Apr 12 10:43:27 barreleye systemd[1]: Started Start Power0.
Apr 12 10:43:27 barreleye system_manager.py[749]: Running System State: HOST_POWERING_ON
Apr 12 10:43:29 barreleye power_control.exe[785]: PowerControl: pgood: 1, setting reset USB_RESET to 1
Apr 12 10:43:29 barreleye system_manager.py[749]: Running System State: HOST_POWERED_ON
Apr 12 10:43:29 barreleye systemd[1]: Started Wait for Power0 to turn on.
Apr 12 10:43:29 barreleye systemd[1]: Reached target Chassis0 (Power On).
Apr 12 10:43:29 barreleye systemd[1]: Reached target Power0 On.
Apr 12 10:43:29 barreleye systemd[1]: Reached target Power0 (On).
Apr 12 10:43:29 barreleye phosphor-chassis-state-manager[791]: Recieved signal that power ON is complete
Apr 12 10:43:29 barreleye phosphor-chassis-state-manager[791]: Change to Chassis Power State
root@barreleye:~#
```

geissonator, 2017-04-12:
Need more from the journal to debug, but looks like you hit a host watchdog (i.e. host didn't start).  Have you tried putting a known good pnor image on?  There's a bug in master PNOR where it's taking longer then the allotted 30s to boot.  A workaround for that is "systemctl disable obmc-start-watchdog@0.service" to disable the watchdog.

yadron, 2017-04-12:
@geissonator , thank you. I've tried turning off watchdog, but it didn't help. The updating of complete .pnor image doesn't help too. The 30s timeout of watchdog is not the reason too.

I've found the reason _(why cannot boot host for Barreleye after ~ 9 march that is 1.99.3+ version of OpenBMC)_  and it is related to content of PNOR, but I don't understand how. Looks like some-kind of bug in PNOR itself, so update of PNOR part (0x01e75000..0x01e76d38) required after BMC booted and before booting host.

More details in https://github.com/openbmc/openbmc/issues/1454

BTW, why nobody found it before I did, even this problem existed for more than a month for Barreleye?

agangidi53, 2017-04-18:
@yadron  I'm Adi from Rackspace. We didn't find this issue because we are working off 'stable' branch. Master branch is moving fast with all the commits related to P9 and hence is tricky to deploy hardware with, at scale.

yadron, 2017-04-18:
@agangidi53 , thank you for the comment. How to fix this issue that Barreleye cannot boot host if OpenBMC updated to master? If you find that it is actual, could you please create a new similar issue at open-power/hostboot?

Regarding the topic of the current issue "_obmcutil state works incorrectly_", it can be closed (so I'm closing it), because fixed.

Regarding second topic "_why cannot boot host for Barreleye after ~ 9 march that is 1.99.3+ version of OpenBMC_", it is moved to #1454 .

closed 2017-04-18 by yadron

yadron, 2017-04-18:
@agangidi53 , I've just checked with fresh Barreleye (other box than one I used before) that if updating from 1.99.0-429 to v1.99.4-34, then the same story (host doesn't work and needs manual upload-inventory first).