openbmc/openbmc #1453: OpenBMC GUI: Enhancement: No. of users logged in should be displayed in the main panel.

sivassrr, 2017-04-12:
Currently we could able to login multiple instance of a root user in a given system. 
Using GUI also able to login with same user id (root), multiple instance. 

So would like to know how many instance of root user had logged-in and who are all users currently logged in. Because when multiple user logged in, It is better to know who are all logged in. Just to ensure there is any parallel operation going on etc..

williamspatrick, 2017-04-12:
This should not be tracked for now as openbmc issues.  We do not even have code in public for the webUI.  Please work with the IBM design team directly on your ideas of what would be "useful" so that they can do the user research and prioritization by their process.

closed 2017-04-12 by williamspatrick

sivassrr, 2017-04-12:
Sure @williamspatrick  got it. Will do that way.