openbmc/openbmc #1454: [BarrelEye v1.99.4-15] Cannot boot host error, and found the reason that is missed op-inventory-upload service, so workaround how to fix here

yadron, 2017-04-12:
Remember, I've already written regarding this problem in all these issues

https://github.com/openbmc/openbmc/issues/1287 - my issue (and some architectural concerns here)
https://github.com/openbmc/openbmc/issues/1262 - other similar issue regarding Palmetto, and my debugging logs here (so this issue is a part of 1287 for me).
https://github.com/openbmc/openbmc/issues/832 - the issue that is being fixed right now (by GitHub:Geissonator) and possibly will help with 1287.
https://github.com/openbmc/openbmc/issues/1324 - the minor issue that is a part of https://github.com/openbmc/openbmc/issues/1287 and https://github.com/openbmc/openbmc/issues/1262 (like I've already written it in https://github.com/openbmc/openbmc/issues/1262 , but again for highlighting it)
https://github.com/openbmc/openbmc/issues/1358 - /usr/sbin/ipmid broken (error of loading libapphandler.so.0), fixed and closed.
https://github.com/openbmc/openbmc/issues/1443 - Updating at BMC console BMC flash fails during "Verifying kb".
https://github.com/openbmc/openbmc/issues/1452 - "obmcutil state" didn't work for v1.99.4-14 (and fixed with v1.99.4-15, but still cannot boot host)

and I've found today what is the main reason.

After long and hard debugging _(after few days debugging)_, I've found that the only difference for Barreleye v1.99.3-48 regarding attempt of host boot comparing to v1.99.4-15 is op-inventory-upload@.service (means it worked in 1.99.3 because /usr/sbin/inventory_upload.py executes just before booting the host and in v1.99.4-15 this op-inventory-upload@.service missed), this is the output from journalctl of 1.99.3 when it is Ok:

```
Apr 12 16:53:06 barreleye inventory_upload.py[955]: WARNING: Erase region adjusted to 0x01e75000..0x01e77000
Apr 12 16:53:06 barreleye inventory_upload.py[955]: About to erase 0x01e75000..0x01e77000 !
Apr 12 16:53:06 barreleye inventory_upload.py[955]: Erasing...
Apr 12 16:53:06 barreleye inventory_upload.py[955]: [171B blob data]
Apr 12 16:53:06 barreleye inventory_upload.py[955]: About to program "/tmp/tmpzKPWpS" at 0x01e75000..0x01e76d38 !
Apr 12 16:53:06 barreleye inventory_upload.py[955]: Programming & Verifying...
Apr 12 16:53:07 barreleye inventory_upload.py[955]: [114B blob data]
Apr 12 16:53:07 barreleye inventory_upload.py[955]: Updating actual size in partition header...
```

, what it does: "Uploading inventory to PNOR in dtb format..." (just running one command "pflash', '-f', '-e', '-p', tmpfile, '-P', 'BMC_INV" once (and the tmp file prepared by "'dtc', '-O', 'dtb'"):

```
root@barreleye:~# cat /lib/systemd/system/op-inventory-upload@.service
[Unit]
Description=Inventory Upload
Wants=obmc-host-start-pre@%i.target
Before=obmc-host-start-pre@%i.target
Conflicts=obmc-chassis-stop@%i.target

[Service]
Restart=no
Type=oneshot
RemainAfterExit=yes
ExecStart=/usr/sbin/inventory_upload.py
Environment="PYTHONUNBUFFERED=1"

[Install]
WantedBy=obmc-chassis-start@%i.target
```

What I did, I've taken latest v1.99.4-15 and reverted back all changes in power-management programs back to 28 Feb version (but it didn't help), and then noted that there is also this missed op-inventory-upload service during host boot (other worked the same), and then uploaded manually both dtc program (cmd "scp /home/build/openbmc-feb28/openbmc/build/tmp/work/s188-openbmc-linux-gnueabi/obmc-phosphor-image/1.0-r0/rootfs/usr/bin/dtc root@172.17.27.137:/usr/bin/") and /usr/sbin/inventory_upload.py and ran once manually at BMC (after BMC started and just before "obmcutil poweroff; obmcutil poweron" commands) this command:

`/usr/sbin/inventory_upload.py`

and after that host booted without problems.

The hack program for reverting v1.99.4-15 to 1.99.3 looks like (because I'm not sure if it will work with new management-power programs, like it could be other bugs also - I'll check that tomorrow and tell here then):

Added 

```
do_compile_prepend() {
  bbwarn "Reverting:"

  bbwarn `cd /home/build/openbmc4/openbmc/build;/bin/bash ~/orig/revert.sh`
}
```

to files

```
  328  vi ./meta-phosphor/common/recipes-core/systemd/obmc-targets.bb
  331  vi ./meta-phosphor/common/recipes-core/systemd/obmc-targets.bb
  486  vi ../meta-phosphor/common/recipes-phosphor/chassis/obmc-op-control-power.bb 
[... 211 lines cut ...]
cp ~/openbmc-feb28/openbmc/build/tmp/work/all-openbmc-linux/obmc-mgr-state/1.0-r1/git/fanctl/fan_control.c  tmp/work/all-openbmc-linux/obmc-mgr-state/1.0-r1/git/fanctl/fan_control.c 
cp ~/openbmc-feb28/openbmc/build/tmp/work/all-openbmc-linux/obmc-mgr-state/1.0-r1/git/hostcheckstop/host_checkstop_obj.c tmp/work/all-openbmc-linux/obmc-mgr-state/1.0-r1/git/hostcheckstop/host_checkstop_obj.c 
cp ~/openbmc-feb28/openbmc/build/tmp/work/all-openbmc-linux/obmc-mgr-state/1.0-r1/git/op-pwrctl/power_control_obj.c ./tmp/work/all-openbmc-linux/obmc-mgr-state/1.0-r1/git/op-pwrctl/power_control_obj.c
cp ~/openbmc-feb28/openbmc/build/tmp/work/all-openbmc-linux/obmc-mgr-state/1.0-r1/git/pciedetect/pcie_slot_present_obj.c ./tmp/work/all-openbmc-linux/obmc-mgr-state/1.0-r1/git/pciedetect/pcie_slot_present_obj.c
cp ~/openbmc-feb28/openbmc/build/tmp/work/all-openbmc-linux/obmc-mgr-state/1.0-r1/git/pychassisctl/chassis_control.py ./tmp/work/all-openbmc-linux/obmc-mgr-state/1.0-r1/git/pychassisctl/chassis_control.py
cp ~/openbmc-feb28/openbmc/build/tmp/work/all-openbmc-linux/obmc-mgr-state/1.0-r1/git/pystatemgr/discover_system_state.py ./tmp/work/all-openbmc-linux/obmc-mgr-state/1.0-r1/git/pystatemgr/discover_system_state.py
cp ~/openbmc-feb28/openbmc/build/tmp/work/all-openbmc-linux/obmc-mgr-state/1.0-r1/git/pytools/obmcutil ./tmp/work/all-openbmc-linux/obmc-mgr-state/1.0-r1/git/pytools/obmcutil 

cp ~/openbmc-feb28/openbmc/build/tmp/work/all-openbmc-linux/obmc-flash-bmc/1.0-r1/git/fanctl/fan_control.c  tmp/work/all-openbmc-linux/obmc-flash-bmc/1.0-r1/git/fanctl/fan_control.c 
cp ~/openbmc-feb28/openbmc/build/tmp/work/all-openbmc-linux/obmc-flash-bmc/1.0-r1/git/hostcheckstop/host_checkstop_obj.c tmp/work/all-openbmc-linux/obmc-flash-bmc/1.0-r1/git/hostcheckstop/host_checkstop_obj.c 
cp ~/openbmc-feb28/openbmc/build/tmp/work/all-openbmc-linux/obmc-flash-bmc/1.0-r1/git/op-pwrctl/power_control_obj.c ./tmp/work/all-openbmc-linux/obmc-flash-bmc/1.0-r1/git/op-pwrctl/power_control_obj.c
cp ~/openbmc-feb28/openbmc/build/tmp/work/all-openbmc-linux/obmc-flash-bmc/1.0-r1/git/pciedetect/pcie_slot_present_obj.c ./tmp/work/all-openbmc-linux/obmc-flash-bmc/1.0-r1/git/pciedetect/pcie_slot_present_obj.c
cp ~/openbmc-feb28/openbmc/build/tmp/work/all-openbmc-linux/obmc-flash-bmc/1.0-r1/git/pychassisctl/chassis_control.py ./tmp/work/all-openbmc-linux/obmc-flash-bmc/1.0-r1/git/pychassisctl/chassis_control.py
cp ~/openbmc-feb28/openbmc/build/tmp/work/all-openbmc-linux/obmc-flash-bmc/1.0-r1/git/pystatemgr/discover_system_state.py ./tmp/work/all-openbmc-linux/obmc-flash-bmc/1.0-r1/git/pystatemgr/discover_system_state.py
cp ~/openbmc-feb28/openbmc/build/tmp/work/all-openbmc-linux/obmc-flash-bmc/1.0-r1/git/pytools/obmcutil ./tmp/work/all-openbmc-linux/obmc-flash-bmc/1.0-r1/git/pytools/obmcutil 

cp ~/openbmc-feb28/openbmc/build/tmp/work/all-openbmc-linux/obmc-mgr-download/1.0-r1/git/fanctl/fan_control.c  tmp/work/all-openbmc-linux/obmc-mgr-download/1.0-r1/git/fanctl/fan_control.c 
cp ~/openbmc-feb28/openbmc/build/tmp/work/all-openbmc-linux/obmc-mgr-download/1.0-r1/git/hostcheckstop/host_checkstop_obj.c tmp/work/all-openbmc-linux/obmc-mgr-download/1.0-r1/git/hostcheckstop/host_checkstop_obj.c 
cp ~/openbmc-feb28/openbmc/build/tmp/work/all-openbmc-linux/obmc-mgr-download/1.0-r1/git/op-pwrctl/power_control_obj.c ./tmp/work/all-openbmc-linux/obmc-mgr-download/1.0-r1/git/op-pwrctl/power_control_obj.c
cp ~/openbmc-feb28/openbmc/build/tmp/work/all-openbmc-linux/obmc-mgr-download/1.0-r1/git/pciedetect/pcie_slot_present_obj.c ./tmp/work/all-openbmc-linux/obmc-mgr-download/1.0-r1/git/pciedetect/pcie_slot_present_obj.c
cp ~/openbmc-feb28/openbmc/build/tmp/work/all-openbmc-linux/obmc-mgr-download/1.0-r1/git/pychassisctl/chassis_control.py ./tmp/work/all-openbmc-linux/obmc-mgr-download/1.0-r1/git/pychassisctl/chassis_control.py
cp ~/openbmc-feb28/openbmc/build/tmp/work/all-openbmc-linux/obmc-mgr-download/1.0-r1/git/pystatemgr/discover_system_state.py ./tmp/work/all-openbmc-linux/obmc-mgr-download/1.0-r1/git/pystatemgr/discover_system_state.py
cp ~/openbmc-feb28/openbmc/build/tmp/work/all-openbmc-linux/obmc-mgr-download/1.0-r1/git/pytools/obmcutil ./tmp/work/all-openbmc-linux/obmc-mgr-download/1.0-r1/git/pytools/obmcutil 

```

yadron, 2017-04-13:
Trying the same workaround for native OpenBMC at Barreleye (this time without prior reverting power-management code), and it works:

1. Taken last OpenBMC as is (without any modifications):

```
root@barreleye:~# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.4-15"
VERSION_ID="v1.99.4-15-g4105c81"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.4-15"
BUILD_ID="v1.99.4"
```

2. Upload "dtc" program to BMC (if I got it right, this program written by IBM and prepares some PCI/CPU/DIMM/etc hardware info for uploading to PNOR addresses 0x01e75000..0x01e76d38):

`[build@obmc ~]$ scp /home/build/openbmc-feb28/openbmc/build/tmp/work/barreleye-openbmc-linux-gnueabi/obmc-phosphor-image/1.0-r0/rootfs/usr/bin/dtc root@172.17.27.138:/usr/bin/
`

3. Upload wrapper program inventory_upload.py to BMC (that calls "dtc" and then "pflash'" for saving to PNOR):

`[build@obmc ~]$ scp /home/build/openbmc-feb28/openbmc/build/tmp/work/barreleye-openbmc-linux-gnueabi/obmc-phosphor-image/1.0-r0/rootfs/usr/sbin/inventory_upload.py root@172.17.27.138:/usr/sbin/`

4. Check power status:

```
root@barreleye:~# obmcutil power
pgood = 0
state = 0
pgood_timeout = 10
```

5. Execute:

```
root@barreleye:~# /usr/sbin/inventory_upload.py 
Uploading inventory to PNOR in dtb format...
WARNING: Erase region adjusted to 0x01e75000..0x01e77000
About to erase 0x01e75000..0x01e77000 !
Erasing...
[==================================================] 100%
About to program "/tmp/tmpqOK_cS" at 0x01e75000..0x01e76ed4 !
Programming & Verifying...
[==================================================] 100%
Updating actual size in partition header...
```

6. Run at the second console:

`root@barreleye:~# obmc-console-client
`

7. Boot host:

`root@barreleye:~# obmcutil poweron
`

8. After ~10 seconds, I see that host booting (output of "obmc-console-client" at the second console):

```
  3.93570|ISTEP  6. 3
  4.30889|ISTEP  6. 4
  4.30948|ISTEP  6. 5
...
```

williamspatrick, 2017-04-14:
This makes it seem like you are running a pnor image from Foxconn and not the op-build latest?  We consciously removed support for this inventory upload and did not see any problems in our testing because the real op-build firmware has never had this partition to begin with.  Are you able to upgrade the pnor to an open source version?

yadron, 2017-04-14:
What is Foxconn?

I use op-build open source PNOR version (taken on 13 March 2017 from https://openpower.xyz/job/openpower-op-build/lastStableBuild/distro=ubuntu,target=barreleye/ ), and the instruction how I updated were published https://github.com/openbmc/openbmc/issues/1287 and looks like:

1. Upgraded OpenBMC.

2. root@barreleye:~# pflash -E -f -i -p /tmp/barreleye.pnor
(the file barreleye.pnor from https://openpower.xyz/job/openpower-op-build/lastStableBuild/distro=ubuntu,target=barreleye/).

3. reboot

4. BMC rebooted Ok, but host not booted yet

 ).

I'll try updating .pnor to latest 12 Apr build, but I don't think it changed much from 13 March.

williamspatrick, 2017-04-14:
@yadron - Foxconn is the manufacturer of Barreleye.  I did not know where you got your machine but they have their own original BMC and PNOR versions.

What the `inventory_upload.py` does is attempts to take the BMC's view of the system inventory and push it into a PNOR partition so that the host firmware can use it.  This was, effectively, a hack for the original Barreleye machine development because there was a tool that one customer wanted to use and we did not have time to modify the host firmware properly for this support.  Since we have changed the BMC inventory collection software, this `inventory_upload.py` actually no longer functioned anyhow.  We removed it so that it was no longer needlessly doing work of erasing the PNOR partition on each boot.

I spoke with one of the owners of the Hostboot code base today and he informed me of two things:
1. They did add this partition to the open source firmware PNOR description.  I previously was under the impression that this was only added by Foxconn directly in their custom PNOR images.
2. He did not know of any reason why this would fail to boot the system if the BMC did not populate it.  This person wrote the original code in the host firmware to use this partition and it was purposefully written so that an empty / invalid table should cause no effect.

He did suggest you could modify the PNOR layout in the op-build image to remove this partition.  Otherwise, it sounds like this is a bug that should be opened against open-power/hostboot instead of openbmc/openbmc.

yadron, 2017-04-17:
@williamspatrick , I don't know how to compile an op-build image.

I'll keep using default PNOR (https://openpower.xyz/job/openpower-op-build/lastStableBuild/distro=ubuntu,target=barreleye/) and custom OpenBMC (compiling based on https://openpowerblog.wordpress.com/2016/04/23/how-to-build-openbmc-binary-for-barreleye/ instructions).

If I see again this Barreleye error _"need uploading system inventory manually by "inventory_upload.py" after BMC updated"_, then I'll create similar bug in "open-power/hostboot".

For me at looks like at least this bug already known/actual: "If updated OpenBMC to latest at Barreleye, then needs both updating .pnor and after that manually running inventory_upload.py as written above" _(at least needs running it manually once, but I'll write here later if not only once)_.

williamspatrick, 2017-04-17:
If you have to run inventory_upload.py manually at all this seems like a bug in hostboot.

How to build PNOR images: https://github.com/open-power/op-build#building-an-image

yadron, 2017-05-12:
Don't close this ticket yet. I'm testing latest "v1.99.5-28" right now both with Barreleye and with Dragon platforms.

I'm still having some problems with booting host apart from the need of reading PNOR before poweron.

Quickly (what happens):

The latest OpenBMC (v1.99.5-28) suppose to be working at p8 both booting host Ok, but, unfortunately, I cannot boot host if using latest OpenBMC + Dragon-configs (similar to Barreleye) on platform similar to BarrelEye (let's call this platform Dragon for now). This Dragon is similar to Barreleye (the same P8 and many chips are similar), but have possibility of inserting more hardware to motherboard than native Barreleye.

I'll check today if similar problem exists on native BarrelEye platform (probably not that I've already checked with v1.99.4-22, but maybe v1.99.5-28 differs that I'm going to check at Barreleye).

If debugging this issue at Dragon platform, the state and power status looks Ok and BMC thinks that host booted, but actually host non-booted and no console output at all (so SBE or hostboot frozen before "STEP 6.3"), and at BMC watchdog timeout happens:

```
May 11 16:17:11 dragon system_manager.py[764]: Running System State: HOST_BOOTING
May 11 16:17:11 dragon host_watchdog.exe[722]: Setting watchdog poll interval to: 30000
May 11 16:17:11 dragon host_watchdog.exe[722]: Starting watchdog with poll interval: 30000; May 11 16:17:42 dragon host_watchdog.exe[722]: Polling watchdog: 0
...
May 11 16:17:42 dragon chassis_control.py[718]: Watchdog Error, Going to quiesce
```

If I increase watchdog timeout, it doesn't help. If any services depends problems or quiesce problem, I fixed it (the BMC doesn't go to quiesce anymore after I increased watchdogs timeouts, because it just waits for host booted and no fail, so no Quiesce:

```
root@dragon:~# obmcutil power
pgood = 1
state = 1
pgood_timeout = 10
root@dragon:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.On
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Running
```
, but host not booted and no output at all from obmc-console-client
).

Looks like SBE or other early stage boot-loader works incorrectly with latest OpenBMC at this Dragon platform (and I'm using the same PNOR that worked well booting Dragon host with older OpenBMC v1.99.0.-429, but for some reason it doesn't work anymore if updating to OpenBMC to v1.99.5-28).

Note that for Barreleye similar issue #1426 is going to be fixed by #1548 (is it related to https://github.com/openbmc/mboxbridge/blob/master/Documentation/mboxd.md module? Where this module existed before and where it exists now and where it will exist in future? Do you think that if we add mboxd module to latest versions of PNOR/OPenBMC for my Dragon platform, then it can help?)

Currently I'm trying copying parts of working OpenBMC "v1.99.0-429" and it's PNOR for overwriting these parts in non-working latest OpenBMC for seeing if maybe it helps with host-booting (that are files "dtc, inventory_upload.py, /tmp/ipmifru01,  ipmifru05,  ipmifru07,  ipmifru09,  ipmifru0b, pnor/
host-ipmid/*,  ipmid,                ipmifru02,  ipmifru06, ipmifru08,  ipmifru0a,  ipmifru0c" and parts of pnor/ : ATTR_PERM  ATTR_TMP  BMC_INV  CVPD  HBB  HBD  HBI  IMA_CATALOG MVPD  NVRAM  OCC  PAYLOAD  SBE  SECBOOT  TEST).

Also just checked SPI registers with devmem for working version of Dragon (host booted with older OpenBMC) and non-working (cannot boot host with latest OpenBMC):

1. The values of SPI differs before reading PNOR and after. Before reading PNOR:

```
root@dragon:~# devmem 0x1e630000
0x00000003
root@dragon:~# devmem 0x1e630004
0x00000300
root@dragon:~# devmem 0x1e630008
0x00000000
root@dragon:~# devmem 0x1e630010
0x00000000
root@dragon:~# devmem 0x1e630014
0x00000000
root@dragon:~#
```

These values changes after running "pflash -i" cmd (so something of SPI initialized regarding PNOR by this cmd):

`0x00000003->0x00000003 0x00000300->0x203B2341 0x00000000->0x00000000 ...
`

yadron, 2017-05-12:
Continuing my previous message, just have found workaround for fixing boot-host of Dragon (that is similar to Barreleye #1426 and #1598 issues, but apart from "pflash -i" also required updating some other registers of AST2400 (like enabling SPI and LPC-AHB bridge that all related to hardware level accessing PNOR flash):

```
devmem 0x1e630004 w 0x203B2341
devmem 0x1e789080 w 0x02ec0500
devmem 0x1e789088 w 0x30000C00
devmem 0x1e78908c w 0xFC0003FF
```
What it does, the first devmem command similar to "pflash -i" (it just initializes SPI), second command initializes LPC-AHB bridge. And some other similar initialization related to flashes. Actually you may read document ast2400v07.pdf for more info. What we did for finding it, copied values of these ASPEED2400 registers from OpenBMC "v1.99.0-429" when host was booted successfully and updated in OpenBMC "v1.99.5-28" just before running "obmcutil poweron".

I don't think that we can close this issue now, because we need to test if new PNOR will work without this workaround first. Possibly we'll need more LPC-AHB initialization in SBE/hostboot boot-loaders than https://github.com/openbmc/openbmc/issues/1598

Sharing complete list of commands as is "how to boot host of Dragon using OpenBMC v1.99.5-28" (both showing that "pflash -i" does the same that is one command "devmem 0x1e630004"):

```
root@dragon:~# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.5-28"
VERSION_ID="v1.99.5-28-g60b4452"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.5-28"
BUILD_ID="v1.99.5"

root@dragon:~# obmcutil power
pgood = 0
state = 0
pgood_timeout = 10
root@dragon:~# devmem 0x1e630004
0x00000300
root@dragon:~# devmem 0x1e789080
0x02EC0400
root@dragon:~# devmem 0x1e789088
0x3000FFF8
root@dragon:~# devmem 0x1e78908c
0xFFF8F007

root@dragon:~# pflash -i
Flash info:
-----------
Name          = /dev/mtd6
Total size    = 64MB 
Erase granule = 64KB 

TOC@0x00000000 Partitions:
...
ID=27      OTHER_SIDE 02000000..02008000 (actual=00000000) 

TOC@0x02000000 Partitions:
-----------
ID=00            part 02000000..02001000 (actual=00001000) 
...
ID=25      OTHER_SIDE 00000000..00008000 (actual=00000000) 

root@dragon:~# devmem 0x1e630004
0x203B2341  [-- note that this value changed after pflash -i]
root@dragon:~# devmem 0x1e789080
0x02EC0400  [-- this the same]
root@dragon:~# devmem 0x1e789088
0x3000FFF8   [-- this the same]
root@dragon:~# devmem 0x1e78908c
0xFFF8F007   [-- this the same]

root@dragon:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.Off
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Off

root@dragon:~# devmem 0x1e630004 w 0x203B2341
root@dragon:~# devmem 0x1e789080 w 0x02ec0500
root@dragon:~# devmem 0x1e789088 w 0x30000C00
root@dragon:~# devmem 0x1e78908c w 0xFC0003FF

root@dragon:~# devmem 0x1e630004
0x203B2341
root@dragon:~# devmem 0x1e789080
0x02EC0500    [-- updated manually]
root@dragon:~# devmem 0x1e789088
0x30000C00     [-- updated manually]
root@dragon:~# devmem 0x1e78908c
0xFC0003FF    [-- updated manually]

root@dragon:~# obmcutil poweron
root@dragon:~# obmc-console-client 
  6.84568|IBM-INFO| Detecting CVPD Present, Target=0x40000, cvpd_present=0x1
...
```

I believe that this is final workaround instead of all these:

https://github.com/openbmc/openbmc/issues/1287 - my issue (and some architectural concerns here)
https://github.com/openbmc/openbmc/issues/1262 - other similar issue regarding Palmetto, and my debugging logs here (so this issue is a part of 1287 for me).
https://github.com/openbmc/openbmc/issues/832 - the issue that is being fixed right now (by GitHub:Geissonator) and possibly will help with 1287.
https://github.com/openbmc/openbmc/issues/1324 - the minor issue that is a part of https://github.com/openbmc/openbmc/issues/1287 and https://github.com/openbmc/openbmc/issues/1262 (like I've already written it in https://github.com/openbmc/openbmc/issues/1262 , but again for highlighting it)
https://github.com/openbmc/openbmc/issues/1452 - "obmcutil state" didn't work for v1.99.4-14 (and fixed with v1.99.4-15, but still cannot boot host)
https://github.com/openbmc/openbmc/issues/1454 - BMC_INV → PNOR; found workaround for this bug "cannot boot Barreleye host" (all these issues starting from 1287 were regarding this bug and this bug existed in OpenBmc after ~ 9 March) and workaround at last in current #1454.

Still the question why older OpenBMC 1.99.0.* (and similar before 9 March 2017) worked well with the same PNOR (without need of setting these four registers to correct values) and with later OpenBMC doesn't. One guess is that mentioned mboxd module ( https://github.com/openbmc/mboxbridge/blob/master/Documentation/mboxd.md ) allows some communications to PNOR flash without LPC-AHB bridge, but didn't investigate yet what is mboxd and if it helps with these LPC-AHB hardware no access to PNOR flash problem (so it looks like the root of problems both for Barreleye and Dragon that SBE/hostboot could not access PNOR flash before all these SPI and LPC-AHB initialized, but possibly there is one more way how SBE/hostboot could access PNOR using mboxd maybe, who can check that? Pls ask HW team and SBE and hostboot for checking this! ).

williamspatrick, 2017-05-12:
@yadron - Your initial values of some of these registers do not match the values being programmed by the kernel on barreleye.  `arch/arm/mach-aspeed/aspeed.c` would run `do_barreleye_setup` on Barreleye, which would set up some of these registers.  If you have modified the device tree so that on your system it is no longer `rackspace,barreleye-bmc` it will no longer run that function.  Have you added a 'do_dragon_setup' or something equivalent?

gtmills, 2018-05-31:
@yadron No update in over a year. Can this be closed or still a problem? Thanks!

gtmills, 2018-07-05:
Closing. no update in a year+, this is an old build and barreleye is no longer supported. Please reopen if still an issue. thanks!

closed 2018-07-06 by gtmills