openbmc/openbmc #1457: Sprint 17.04.19 - Committed

rfrandse, 2017-04-13:


dkodihal, 2017-04-17:
Squad 1

Deepak
- #1439 mboxd: Create the toc in virtual pnor
- #1440 mboxd: Create pnor partitions ?

Ratan
mboxd story?
code-update refactor story?
Get Devender on remaining n/w refactoring?

Jayanth
- #1377 Define DBus Interfaces for OpenBMC Dump
- New investigation story to finalize dump content and open more stories.

Tom
- IPMI SEL or DCMI

Dhruv
- #941 Assert fault LED's when association is set and cleared?
- #1323 openpower-proc-control: Callout processors on FSI failures

Vishwa
- Follow-up on OCC pass-through or new OCC story?

Devender
- Enable openpower-dbus error interfaces?
- Enable app-local error yamls?

Persistent storage?
Host watchdog refactor?

anoo1, 2017-04-17:
# Squad 3 -

 **Adriana:anoo1 -**

- code update

---

**Gunnar:gtmills -**

- code update

---

**Andrew:geissonator -**

- 

---

**Josh:JoshDKing -**
Attends Tuesday and Thursday

Carry over?
- #1286 
- #1163 

---

**Michael:mtritz -

- No new stories

---

**Saqib: -**

- Code update? Automation?

---

**Leonel: -

- Automation

bradbishop, 2017-04-17:
Squad 2

spinler:
 - next fan control story

barth:
 - next fan control story

closed 2017-05-11 by rfrandse